verilator/verilator #5742: Verilator closes ungracefully upon $error without calling vpiEndOfSimulation, causing Cocotb to terminate simulation abruptly

s3rg1o7, 2025-01-21:
> Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contaned so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs
The issue is fully described (including reproducible code snippets) [here](https://github.com/cocotb/cocotb/issues/4378#issuecomment-2602904818)

> What 'verilator' command line do we use to run your example?
`make sim SIM=verilator` to start the cocotb simulation

> What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.030 2024-10-27 rev v5.030-45-g2cb1a8de7

> What OS and distribution are you using?
Ubuntu 24.04.1 LTS

> May we assist you in trying to fix this in Verilator yourself?
Probably not, the analysis from the [Cocotb's team](https://github.com/cocotb/cocotb/issues/4378#issuecomment-2602904818) is a missing call to the `vpiEndOfSimulation` when user calls `$error` 
(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

closed 2025-01-21 by wsnyder

s3rg1o7, 2025-01-25:
> See [#4980](https://github.com/verilator/verilator/issues/4980) and [#5135](https://github.com/verilator/verilator/issues/5135).

> (from #5135) All are historical simplifications from when there was no event loop in Verilator.

Am I to understand that my error is caused by what's described in #4980 and that's a known and open problem that no one has worked on, yet?

wsnyder, 2025-01-25:
Correct.