verilator/verilator #5745: Blocking assignment signal change not showing in waveform

kenobi314159, 2025-01-22:
Hello,

I am running a very simple Verilog testbench using `Verilator 5.006 2023-01-22 rev (Debian 5.006-3)` and for some reason the resulting waveform doesn't show all changes of one of the signals, even though when I print the value before every change, changing values are printed.

The testbench code looks as below. The DUT can be found [here](https://github.com/GabbedT/Arithmetic-Circuits/blob/main/Integer/Dividers/non_restoring_divider.sv).
```SystemVerilog
module tb;

    localparam DATA_WIDTH = 8;

    logic                    clk_i;
    logic                    clk_en_i;
    logic                    rst_n_i;
    logic [DATA_WIDTH - 1:0] dividend_i;
    logic [DATA_WIDTH - 1:0] divisor_i;
    logic                    data_valid_i;

    logic [DATA_WIDTH - 1:0] quotient_o;
    logic [DATA_WIDTH - 1:0] remainder_o;
    logic                    divide_by_zero_o;
    logic                    data_valid_o;
    logic                    idle_o;

    always #10 clk_i = ~clk_i;

    logic [DATA_WIDTH - 1:0] dividend_reg;

non_restoring_divider #(
    .DATA_WIDTH (DATA_WIDTH)
) dut (
    .clk_i             (clk_i),
    .clk_en_i          (clk_en_i),
    .rst_n_i           (rst_n_i),
    .dividend_i        (dividend_i),
    .divisor_i         (divisor_i),
    .data_valid_i      (data_valid_i),

    .quotient_o        (quotient_o),
    .remainder_o       (remainder_o),
    .divide_by_zero_o  (divide_by_zero_o),
    .data_valid_o      (data_valid_o),
    .idle_o            (idle_o)
);

initial begin

    clk_i = 0;
    clk_en_i = 1;
    rst_n_i = 0;
    data_valid_i = 0;
    @(posedge clk_i);
    @(posedge clk_i);
    rst_n_i = 1;
    @(posedge clk_i);

    // stimulus
    for (int i = 0; i < 10; i++) begin
        while (idle_o == 0) begin
            $display("%t ns: waiting dividend_i %x divisor_i %x", $time, dividend_i, divisor_i);
            dividend_i = $random;
            divisor_i = divisor_i + 1;
            @(posedge clk_i);
        end
        $display("%t ns: Iteration %0d", $time, i);

        dividend_i = $random;
        divisor_i = $random;
        data_valid_i = 1;
        @(posedge clk_i);
        data_valid_i = 0;
    end

    #200 $finish;
end

  // Debug signal
always_ff @(posedge clk_i) begin
    dividend_reg <= dividend_i;
end

  // Simulator dependent system tasks that can be used to
  // dump simulation waves.
initial begin
    $dumpfile("dump.fst");
    $dumpvars;
end
endmodule
```

I verilate the code using
`verilator --Wno-fatal --binary --main --trace-fst tb.sv non_restoring_divider.sv`
then run the testbench, which outputs this:
```
./obj_dir/Vtb
                  50 ns: Iteration 0
                  70 ns: waiting dividend_i c4 divisor_i 9c
                  90 ns: waiting dividend_i 02 divisor_i 9d
                 110 ns: waiting dividend_i e4 divisor_i 9e
                 130 ns: waiting dividend_i 78 divisor_i 9f
                 150 ns: waiting dividend_i bc divisor_i a0
                 170 ns: waiting dividend_i b6 divisor_i a1
                 190 ns: waiting dividend_i e4 divisor_i a2
                 210 ns: waiting dividend_i b7 divisor_i a3
                 230 ns: Iteration 1
                 250 ns: Iteration 2
                 270 ns: waiting dividend_i c5 divisor_i 82
                 290 ns: waiting dividend_i 62 divisor_i 83
                 310 ns: waiting dividend_i 35 divisor_i 84
                 330 ns: waiting dividend_i 12 divisor_i 85
                 350 ns: waiting dividend_i 56 divisor_i 86
                 370 ns: waiting dividend_i fd divisor_i 87
                 390 ns: waiting dividend_i f4 divisor_i 88
                 410 ns: waiting dividend_i 3b divisor_i 89
                 430 ns: Iteration 3
                 450 ns: Iteration 4
                 470 ns: waiting dividend_i 1e divisor_i 9e
                 490 ns: waiting dividend_i 9e divisor_i 9f
                 510 ns: waiting dividend_i ee divisor_i a0
                 530 ns: waiting dividend_i 10 divisor_i a1
                 550 ns: waiting dividend_i 76 divisor_i a2
                 570 ns: waiting dividend_i 63 divisor_i a3
                 590 ns: waiting dividend_i c0 divisor_i a4
                 610 ns: waiting dividend_i 1d divisor_i a5
                 630 ns: Iteration 5
                 650 ns: Iteration 6
                 670 ns: waiting dividend_i 63 divisor_i 78
                 690 ns: waiting dividend_i 6f divisor_i 79
                 710 ns: waiting dividend_i ce divisor_i 7a
                 730 ns: waiting dividend_i f7 divisor_i 7b
                 750 ns: waiting dividend_i d1 divisor_i 7c
                 770 ns: waiting dividend_i 9e divisor_i 7d
                 790 ns: waiting dividend_i 77 divisor_i 7e
                 810 ns: waiting dividend_i 6e divisor_i 7f
                 830 ns: Iteration 7
                 850 ns: Iteration 8
                 870 ns: waiting dividend_i 94 divisor_i e2
                 890 ns: waiting dividend_i 66 divisor_i e3
                 910 ns: waiting dividend_i 1f divisor_i e4
                 930 ns: waiting dividend_i 09 divisor_i e5
                 950 ns: waiting dividend_i e3 divisor_i e6
                 970 ns: waiting dividend_i 1f divisor_i e7
                 990 ns: waiting dividend_i 84 divisor_i e8
                1010 ns: waiting dividend_i c5 divisor_i e9
                1030 ns: Iteration 9
- tb.sv:74: Verilog $finish
```
As you can see between each "Iteration" there is several random values of "dividend_i" and "divisor_i" as expected.

However, when I open the resulting wave file in gtkwave, it shows this:
[image]

The "dividend_i" and "divisor_i" signals only change around valid cycles and the inner loop doesn't seem to take effect. What's even stranger, the debug signal "dividend_reg" which is just a delayed copy of "dividend_i" is changing in every cycle (and isn't delayed).

This is really my first time trying out the Verilator so it might be some kind of a stupid mistake on my side, but I couldn't find any existing article on how to solve it. I'd be happy if you could tell me what I'm doing wrong.

wsnyder, 2025-01-22:
Please use the recent version and see if the issue persists.

kenobi314159, 2025-01-24:
With latest version 5.032 the issue no longer appears. Thanks. 👍

closed 2025-01-24 by kenobi314159