verilator/verilator #5746: Error creating Verilated binary on final step (/usr/bin/ld)

jordankrim, 2025-01-23:
I built Verilator 5.021 on Red Hat Enterprise Linux release 8.10 (Ootpa) using clang++.  When I attempt to verilate code to create a binary it gives a error message:

 /usr/bin/ld: $WORK_AREA/verif/build/../../build/vobj_my_work/vobj_my_work_secondary/verilated_vcd_c.o: undefined reference to symbol 'pthread_join@@GLIBC_2.2.5'
//usr/lib64/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile_tile:116: run] Error 1

Is this a Verilator issue because it is built incorrectly or something else?

wsnyder, 2025-01-23:
Note RedHat 8 is past its 10-year end-of-life, your company needs to be using something more modern. Also would suggest use the most recent Verilator version you'll likely have a better experience.

Alternatively googling the error message gives some suggestions on what to try, if you solve it please post back here: https://github.com/Microsoft/cpprestsdk/issues/927

closed 2025-01-23 by wsnyder

wsnyder, 2025-01-23:
The Verilator binary can be compiled with a different gcc. What matters is the ./configure before it is the same GCC so that the verilated.mk has the proper commands.  Realistically, once configured it makes sense to build verilator and so its version would match.

include/verilated.mk indicates the version:
`CFG_CXX_VERSION = "g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0"`

`--std` not being honored can be because of certain distros forcing certain flags (usually MacOS), or because Verilator uses a newer standard to get coroutines, there's no current way to turn that off, but those flags are only used when you verilate with `--timing` or `--binary`.

Your error suggests the problem is related to the systemc library GCC version not matching what you are using, as opposed to Verilator itself.