verilator/verilator #5747: `function automatic` local variables behave as if they had `static` duration

augfab, 2025-01-23:
Verilator shows a strange behavior related to local variables with automatic storage duration. I'm probably overlooking something trivial or misinterpreting the spec, please do correct any misunderstanding.

I expect that a local variable with automatic storage duration is freshly allocated when entering the function (_IEEE 1800-2012 LRM_, Section 13.4.2 _Static and Automatic functions_, p. 292). I observe local variables with `automatic` storage duration behave as if they had `static` duration.

I am using Verilator 5.032.
```shell
$ verilator --version
Verilator 5.032 2025-01-01 rev v5.032
```
Also tested with today's master (0507fb46558b1f0826e3feea2b3b5113db6a2d89,  2025-01-23), with the exact same result.

My OS/distribution/machine is GNU/Linux, Ubuntu 20.04, `x86_64`:
```
$ uname -a
Linux ******* 5.15.0-89-generic #99~20.04.1-Ubuntu SMP Thu Nov 2 15:16:47 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
```

Here's a patch with a reproducer in Verilator `test_regress` format showing the problem:
```patch
From 0783a3901d6e6f70eca76dfa0ad0f5c815c0ceb9 Mon Sep 17 00:00:00 2001
From: Augustin Fabre <augustin.fabre@arm.com>
Date: Thu, 23 Jan 2025 21:04:30 +0100
Subject: [PATCH] Tests: add t_func_automatic

Test automatic function variables lifetime.

Make sure that a variable with automatic storage (the default for free
functions marked `automatic') is freshly allocated when entering the
function (IEEE 1800-2012 LRM, Section 13.4.2 Static and Automatic
functions, p. 292).
---
 test_regress/t/t_func_automatic.py | 18 +++++++
 test_regress/t/t_func_automatic.v  | 83 ++++++++++++++++++++++++++++++
 2 files changed, 101 insertions(+)
 create mode 100755 test_regress/t/t_func_automatic.py
 create mode 100644 test_regress/t/t_func_automatic.v

diff --git a/test_regress/t/t_func_automatic.py b/test_regress/t/t_func_automatic.py
new file mode 100755
index 000000000..250edefdc
--- /dev/null
+++ b/test_regress/t/t_func_automatic.py
@@ -0,0 +1,18 @@
+#!/usr/bin/env python3
+# DESCRIPTION: Verilator: Test automatic function variables lifetime
+#
+# Copyright 2025 by Wilson Snyder. This program is free software; you
+# can redistribute it and/or modify it under the terms of either the GNU
+# Lesser General Public License Version 3 or the Perl Artistic License
+# Version 2.0.
+# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0
+
+import vltest_bootstrap
+
+test.scenarios('simulator')
+
+test.compile()
+
+test.execute()
+
+test.passes()
diff --git a/test_regress/t/t_func_automatic.v b/test_regress/t/t_func_automatic.v
new file mode 100644
index 000000000..7114937b8
[... 145 lines cut ...]
```
I expect to see:
```
'{'h0}
'{'h1}
'{'h2}
...
'{'h9}
```

But with Verilator I see:
```
'{'h0}
'{'h0, 'h1}
'{'h0, 'h1, 'h2}
...
'{'h0, 'h1, 'h2, 'h3, 'h4, 'h5, 'h6, 'h7, 'h8, 'h9}
```

I ran the same testcase with QuestaSim, VCS and Xcelium, they all behave in the way I expected (the local variable with automatic storage duration is freshly allocated upon entering the function or the scoped block).

I can't look into fixing this myself. I had a look at the C++ code generated by Verilator, it seems like after inlining and loop unrolling the scoping of each function's local variable is lost, and they become global variables.

Thanks for your time, and for this great tool!
-- Augustin

wsnyder, 2025-01-23:
Thanks for the good bug report. Seems a bug with the function inliner optimization, `-fno-inline-funcs` makes it pass.  I'll take a look.

augfab, 2025-01-24:
Thanks!

For completeness, here's another reproducer for the same behavior with `automatic` variables in a local scope (`begin ... end`).

```patch
From 1236bb3611ca422f0eb337b08b3cc30ef96ae153 Mon Sep 17 00:00:00 2001
From: Augustin Fabre <augustin.fabre@arm.com>
Date: Fri, 24 Jan 2025 08:53:27 +0100
Subject: [PATCH] Tests: add t_func_automatic_scope

Refinement on t_func_automatic. Automatic variables declared in a
local scope should have automatic duration. Entering the scope should
freshly allocate a new variable.
---
 test_regress/t/t_func_automatic_scope.py | 18 +++++++
 test_regress/t/t_func_automatic_scope.v  | 62 ++++++++++++++++++++++++
 2 files changed, 80 insertions(+)
 create mode 100755 test_regress/t/t_func_automatic_scope.py
 create mode 100644 test_regress/t/t_func_automatic_scope.v

diff --git a/test_regress/t/t_func_automatic_scope.py b/test_regress/t/t_func_automatic_scope.py
new file mode 100755
index 000000000..250edefdc
--- /dev/null
+++ b/test_regress/t/t_func_automatic_scope.py
@@ -0,0 +1,18 @@
+#!/usr/bin/env python3
+# DESCRIPTION: Verilator: Test automatic function variables lifetime
+#
+# Copyright 2025 by Wilson Snyder. This program is free software; you
+# can redistribute it and/or modify it under the terms of either the GNU
+# Lesser General Public License Version 3 or the Perl Artistic License
+# Version 2.0.
+# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0
+
+import vltest_bootstrap
+
+test.scenarios('simulator')
+
+test.compile()
+
+test.execute()
+
+test.passes()
diff --git a/test_regress/t/t_func_automatic_scope.v b/test_regress/t/t_func_automatic_scope.v
new file mode 100644
index 000000000..8d936cf11
--- /dev/null
+++ b/test_regress/t/t_func_automatic_scope.v
@@ -0,0 +1,62 @@
+// DESCRIPTION: Verilator: Test automatic function variables lifetime
+//
+// Make sure that a variable with automatic storage (the default for
+// free functions marked `automatic') is freshly allocated when
+// entering the scope.
+//
+// This file ONLY is placed under the Creative Commons Public Domain, for
+// any use, without warranty, 2025 by Augustin Fabre.
+// SPDX-License-Identifier: CC0-1.0
+
+`define stop $stop
+`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d: got=%0x exp=%0x (%s !== %s)\n", `__FILE__,`__LINE__, (gotv), (expv), `"gotv`", `"expv`"); `stop; end while(0);
+
+module t();
+  typedef struct {
+    int y;
+  } Y;
+
+  class Z;
+    int z;
+  endclass
+
+  function automatic void main();
+    $display("queue");
+    for (int i = 0; i < 3; ++i) begin
+      int q[$];
+      q.push_back(0);
+      $display("  %p", q);
+      `checkh(q.size(), 1);
+    end
+
+    $display("scalar");
+    for (int i = 0; i < 3; ++i) begin
+      int x;
+      ++x;
+      $display("  %0d", x);
+      `checkh(x, 1);
+    end
+
+    $display("struct");
+    for (int i = 0; i < 3; ++i) begin
+      Y y;
+      ++y.y;
+      $display("  %p", y);
+      `checkh(y.y, 1);
+    end
+
+    $display("class");
+    for (int i = 0; i < 3; ++i) begin
+      Z z = new();
+      ++z.z;
+      $display("  %p", z);
+      `checkh(z.z, 1);
+    end
+  endfunction
+
+  initial begin
+    main();
+    $write("*-* All Finished *-*\n");
+    $finish;
+  end
+endmodule
-- 
2.25.1

```

Verilator behaves unexpectedly:
```
driver: Leaving directory '/home/xxxxxxxx/src/verilator/test_regress/obj_vlt/t_func_automatic_scope'
	export GCOV_PREFIX_STRIP='99'
	export GCOV_PREFIX='/home/xxxxxxxx/src/verilator/test_regress/obj_dist/gcov_1'
	obj_vlt/t_func_automatic_scope/Vt_func_automatic_scope    > obj_vlt/t_func_automatic_scope/vlt_sim.log
queue
  '{'h0} 
  '{'h0, 'h0} 
%Error: t/t_func_automatic_scope.v:29: got=2 exp=1 (q.size() !== 1)
%Error: t/t_func_automatic_scope.v:29: Verilog $stop
Aborting...
Aborted (core dumped)
%Warning: vlt/t_func_automatic_scope: Exec of obj_vlt/t_func_automatic_scope/Vt_func_automatic_scope failed: %Error: t/t_func_automatic_scope.v:29: got=2 exp=1 (q.size() !== 1)
vlt/t_func_automatic_scope: %Error: Exec of obj_vlt/t_func_automatic_scope/Vt_func_automatic_scope failed: %Error: t/t_func_automatic_scope.v:29: got=2 exp=1 (q.size() !== 1)
vlt/t_func_automatic_scope: FAILED: Exec of obj_vlt/t_func_automatic_scope/Vt_func_automatic_scope failed: %Error: t/t_func_automatic_scope.v:29: got=2 exp=1 (q.size() !== 1)
==SUMMARY: Passed 0  Failed 1  Time 0:05

======================================================================
#vlt/t_func_automatic_scope: %Error: Exec of obj_vlt/t_func_automatic_scope/Vt_func_automatic_scope failed: %Error: t/t_func_automatic_scope.v:29: got=2 exp=1 (q.size() !== 1)
		make&& test_regress/t/t_func_automatic_scope.py  --vlt
==TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:05
```

QuestaSim, VCS and Xcelium behave in the expected way:
```
queue
  '{0}
  '{0}
  '{0}
scalar
  1
  1
  1
struct
  '{y:1}
  '{y:1}
  '{y:1}
class
  '{z:1}
  '{z:1}
  '{z:1}
*-* All Finished *-*
```

wsnyder, 2025-02-25:
Problem is understood; each entry to a function must reset all automatics to initial value.  However the normal constructor (at module level) must not in case such settings have side effects.

wsnyder, 2025-02-26:
Fixed.  A warning that given the complexity I won't be surprised if this causes some other things to break.

closed 2025-02-26 by wsnyder