verilator/verilator #5753: Optimize clearFired

wsnyder, 2025-01-25:
@kbieganski, t_timing_fork_join.py makes this code:

```
void Vt_timing_fork_join___024root___eval_triggers__act(Vt_timing_fork_join___024root* vlSelf) {
    ...
    if (vlSelfRef.t__DOT__event1.isFired()) {
        vlSelfRef.t__DOT__event1.clearFired();
    }
    if (vlSelfRef.t__DOT__event2.isFired()) {
        vlSelfRef.t__DOT__event2.clearFired();
    }
    if (vlSelfRef.t__DOT__event3.isFired()) {
        vlSelfRef.t__DOT__event3.clearFired();
    }
```

It looks like isFired is testing m_fired, then cleared fired just clears m_Fired. I'd think it would be more efficient to just always clear m_Fired, avoiding the branches; the only downside will be write cache pressure when the variable is already clear.  (Unless it's somehow a hot variable between threads... which seems unlikely)

This appears due to this: https://github.com/verilator/verilator/blob/05c7a3b90b7607ef4ae522b5ccec23d6ad308966/src/V3SenExprBuilder.h#L185

Removing the if, tests still pass.

@kbieganski (I think your code) can this be optimized to just skip the if() or am I missing something?  Thanks

kbieganski, 2025-01-29:
I think you're correct, although @gezalore is the real author of this code (I just moved it).

gezalore, 2025-01-29:
I agree, that change seems ok

closed 2025-01-30 by wsnyder