verilator/verilator #5755: How does verilator convert internal signals into class attributes ?

lovisXII, 2025-01-26:
I'm using verilator version 4.028 since I'm on wsl and v5 is not available, I would need to compile it from source, I tried in the past and I got many issues with libraries and compiler version.

Since only the documentation for v5 is available it's kinda tough to find any good information. My question is the following :

When you write a system verilog module verilator convert it to c++ and the internal signals are translated into some kind of class attribues. For example let's say I have this :

top module : core
inside it I got a module named exe instanciated a u_exe
Inside exe I got a signal name branch_v From the c++ side it will create : core__DOT__u_exe__DOT__branch_v
The thing is that some signal are not present inside the Vcore.h files, in my case I actually cannot find the branch_v signal. I guess that verilator is doing some optimization but I would like to be able to force it to generate some signal.

I've read about those stuff : /*verilator inline_module*/ but I'm not sure what it's suppose to do

wsnyder, 2025-01-26:
WSL running Ubuntu 22.04 LTS is known to work fine on git master.

I assume you read through https://veripool.org/guide/latest/connecting.html

probably it needs to be clearer:  Internal signals are not to be accessed, only the top level .h signals are, so the names shouldn't be relevant.  If you use /*verilator public*/ you can access the signals, and the DOTs will go away. Generally it's better to avoid public and be more portable across simulators using the DPI/VPI.

closed 2025-01-26 by wsnyder

lovisXII, 2025-01-26:
Well at least it's not available in ubuntu package, it seems only v4 is.
Yes I did, I finally managed to find a workaround /*verilator public_flat*/ which seems to work fine.

And maybe it's not done to be accessed but it is possible to do so, see : https://github.com/openhwgroup/cva6/blob/master/corev_apu/tb/ariane_tb.cpp

It's pretty conveniant to capture some internal information and create some performance events or anything related to the micro-architecture

wsnyder, 2025-01-26:
Glad it's fixed.  It looks like that Ariane example could have used DPI and would have been portable, which is my point. ;)  But if it's working, then fine.

Yes, 20.04 is 5 years old now - you realize it goes unsupported later this year (unless you pay)?  Note packages always lag the main repositories by years, so better to always build from scratch.

lovisXII, 2025-01-26:
Alright, I'll have a look to a DPI version, thanks for you help !

wsnyder, 2025-01-26:
BTW that code might have been written before Verilator's DPI supported memories, given it had Verilator V4 mentions.  Note it you do use the DPI it would work in the "other direction", that is the Verilog should call the DPI function that calls into C, passing the memory as an argument, to performs the memory ops in C.