verilator/verilator #5758: No VPI access to interface memebers when using --public-depth instead of --public-flat-rw

paul-demo, 2025-01-28:
I am using Verilator 5.024 with VPI / cocotb. I found that using `--public-depth 1` instead of `--public-flat-rw` speeds up simulations by a factor of 2.3x, a gigantic improvement if your python testbench does not need to reach deep into the DUT hierarchy to peek/poke signals and can just access the top level ports and signals.

However, for interfaces, I am finding that signals become inaccessible, which is a problem since I want to hook up a python AXI-stream (or similar) agent which connects to the interfaces of the DUT. This all works fine but with `--public-flat-rw` I get the error `AttributeError: s_axis_if contains no object named tdata`.

I looked into the verilated files and observed that with `--public-depth 1` the signals which are part of the interface do not exist in the public section of the generated C++ class (which is defined in `Vtop_AXIS_IF__U20_I1_DB1_S0.h` in this case), and with `--public-depth 2` they signals that are members of the interface do exist. This is as I would expect, since the interfaces are part of the DUT top level and thus the signals in that interface would be 2 levels deep. This is shown in the screenshot diff below:

[image]

There must be some additional step where these signals become accessible over VPI, and for whatever reason, the appropriate setting of `--public-depth N` works for regular verilog signals, but does not work for verilog signals that are embedded in an interface, even if I make N sufficiently large.

I attempted to tag some signals with `/* verilator public */` and similar source code directives, but this did not change the result. Is there a place in the Verilator code where signals of an interface are treated differently, with regard to the VPI interface, when one or the other of `--public-depth` and `--public-flat-rw` is used?

paul-demo, 2025-01-28:
Comparing `Vtop__Syms.cpp`, I can see where some `__Vscope_dut__s_axis_if.varInsert()` calls differ between the two. The parameters of the interface are in-tact in both cases because I've used `--public-params`. However, only the `--public-flat-rw` version shows the lines like this

```c++
__Vscope_dut__s_axis_if.varInsert(__Vfinal,"tdata", &(TOP__dut__DOT__s_axis_if.tdata), false, VLVT_WDATA,VLVD_NODIR|VLVF_PUB_RW,1 ,127,0);
```

Does this help narrow it down and could we possibly make this work for interfaces like it does for regular module/signal hierarchy?

paul-demo, 2025-01-28:
Actually, looking closer at the first big screenshot from my first post in this thread, it's very weird that `tuser` shows up in both cases but the other 8 signals in the interface do not.  That actually seems like a bug: `CData`, `SData`, and `VlWide` types are not included when using `--public-depth`, but for some reason the one `IData` field is included in both cases.

paul-demo, 2025-01-29:
Could it be that `level()` is not initialized and thus this `level() - 1` math wraps around here?

https://github.com/verilator/verilator/blob/a379382d32b1b52f8490c4d0ddf0d186f8c4d1e8/src/V3LinkParse.cpp#L348-L354

wsnyder, 2025-01-29:
Thanks for debugging, f you run with --debug in the .tree file you can see the levels. (L and number)

```
    1: MODULE 0x555557065680 <e1836> {f45ai}  __024root  L1 [P] [1ps]
    1: MODULE 0x55555712efc0 <e1838> {f45ai}  t  L0 [1ps]
    3:1: MODULE 0x555557010000 <e6> {a0aa}  @CONST-POOL@  L0 [NONE]
```

You can see the level numbers start at zero, so that subtract is suspicious.

paul-demo, 2025-01-29:
It looks like it's using `L9` despite the interface being in the top level, and despite `--public-depth 1` vs `--public-depth 2` having an observable effect on what is in the __syms class that gets generated.

```bash
grep -r -F "s_axis_if -> IFACE" ./build
```

```
1:2: CELL 0x600003800c00 <e428480> {bq20af}  __PVT__dut__DOT__s_axis_if -> IFACE 0x60000381cf00 <e138286> {d9al}  AXIS_IF__U20_I1_DB1_S0  L9 [LIB] [1ns]
```

Two other thoughts I had on this:
1) The subtraction `level() - 1` is probably fine because `m_level` is an `int` and so it could just go negative if it was left at zero for some reason.

2) Maybe for some reason the interface's signals are getting pruned by a bug in V3Dead.cpp. I noticed there is some perhaps relevant function `preserveTopIfaces` but haven't dug into what could be going on there.
https://github.com/verilator/verilator/blob/a379382d32b1b52f8490c4d0ddf0d186f8c4d1e8/src/V3Dead.cpp#L476

paul-demo, 2025-01-29:
For a different / smaller example which is basically this:

```systemverilog
module tb;

  logic clk;
  logic reset;

  AXIS_IF #(
      .DATA_W(32),
      .USER_W(8),
      .ID_W  (2),
      .DEST_W(2),
      .STRB_W(1)
  ) s_axis_if (
      .aclk   (clk),
      .aresetn(~reset)
  );

  AXIS_IF #(
      .DATA_W(32),
      .USER_W(8),
      .ID_W  (2),
      .DEST_W(2),
      .STRB_W(1)
  ) m_axis_if (
      .aclk   (clk),
      .aresetn(~reset)
  );

  dut dut_i (
      .clk(clk),
      .reset(reset),
      .s_axis_if(s_axis_if),
      .m_axis_if(m_axis_if)
  );

endmodule
```

Verilator `--debug` shows `L5` for the interfaces, which is yet another depth than `L9` from the more complicated dut example (but which had the same structure).
```
    1:2:1: IFACEREFDTYPE 0x1359074a0 <e6777#> {i12af} @dt=0x0@ cell=s_axis_if if=AXIS_IF -> CELL 0x60000273c200 <e6619#> {i12af}  s_axis_if -> IFACE 0x60000273c000 <e6996#> {d9al}  AXIS_IF  L5 [LIB] [1ns]
...
    1:2:1: IFACEREFDTYPE 0x135907780 <e6781#> {i23af} @dt=0x0@ cell=m_axis_if if=AXIS_IF -> CELL 0x60000273c300 <e6685#> {i23af}  m_axis_if -> IFACE 0x60000273c000 <e6996#> {d9al}  AXIS_IF  L5 [LIB] [1ns]
```

wsnyder, 2025-01-30:
Looking at #5708, perhaps that is related, the level is calculated based on recursion need not what the user observes.  Also if an interface is used at several different levels, at top and lower down, it seems likewise wrong to trace it in both places; ideally we need to calculate the level based on number of instantiations (AstCells) deep.

paul-demo, 2025-01-30:
I don't see any obvious issues, but also am not very familiar with all the steps that take place. It appears that V3LinkLevel generates the level just using the rank of the nodes here:

https://github.com/verilator/verilator/blob/e0e164cea216e09a3de9c331b0b3c4180e5fd6ee/src/V3LinkCells.cpp#L177

This would suggest that the graph itself is incorrect if those nodes are coming out as L5/L9 instead of something lower based on where the parent interface is declared.

wsnyder, 2025-02-01:
For similar in test_regress format:

<details>

```
interface AXIS_IF (
                   input logic aclk
                   );
   logic [127:0] tdata;
   logic [ 31:0] tuser;
   logic tvalid, tready;
   modport master(input aclk, output tdata, tuser, tvalid, input tready);
   modport slave(input aclk, input tdata, tuser, tvalid, output tready);
endinterface : AXIS_IF

module dut (
            input clk,
            AXIS_IF.slave s_axis_if
            );
   assign s_axis_if.tready = s_axis_if.tdata[0];
endmodule

module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;
   AXIS_IF s_axis_if (.aclk(clk));
   dut u_dut (.clk, .s_axis_if(s_axis_if));
   initial begin
      $c("Verilated::scopesDump();");
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

</details>

I get

```
$ test_regress/t/t_0.py --public-depth 0
  scopesDump:

$ test_regress/t/t_0.py --public-depth 1
  scopesDump:
    SCOPE 0x5630771bf640: top.TOP
    SCOPE 0x5630771bf680: top.t

$ test_regress/t/t_0.py --public-depth 2
  scopesDump:
    SCOPE 0x560f3add5640: top.TOP
    SCOPE 0x560f3add5680: top.t
    SCOPE 0x560f3add56c0: top.t.u_dut

$ test_regress/t/t_0.py --public-depth 3
  scopesDump:
    SCOPE 0x55ec8cbd0640: top.TOP
    SCOPE 0x55ec8cbd0680: top.t
    SCOPE 0x55ec8cbd06c0: top.t.s_axis_if
    SCOPE 0x55ec8cbd0700: top.t.u_dut
```

This is consistent with what I suggested above (but was perhaps not clear enough on), levels() is the maximum depth that interface is used, which is underneath "sub".  What is wanted in your case is for it to be the same depth as "sub", so a new accessor needs to be used instead of levels()  (e.g. levelMin()).

paul-demo, 2025-02-01:
Sounds good. It seems like we are in agreement of the symptom / request, that interfaces' signals become public according to their depth of declaration rather than [deepest] depth of usage. This would make them act more like signals which already obey this property, I think.

Perhaps my example would be more compelling if we nest a few more layers of DUTs. The deeper it goes before one of the interface signals is consumed, the more `--public-depth` levels will need to be specified, even though the interface declaration is still at the same level in `tb`.

```systemverilog
module dut (
    AXIS_IF.slave s_axis_if
);
  sub_dut u_sub_dut (.s_axis_if(s_axis_if));
endmodule

module sub_dut (
    AXIS_IF.slave s_axis_if
);
  sub_sub_dut u_sub_sub_dut(.s_axis_if(s_axis_if));
endmodule

module sub_sub_dut (
    AXIS_IF.slave s_axis_if
);
  assign s_axis_if.tready = s_axis_if.tdata[0];
endmodule
```

wsnyder, 2025-09-24:
Implemented as described in comments.