verilator/verilator #5760: Verilator failed to detect write assignment in two synchronous blocks.

navinreddy23, 2025-02-03:
Thanks for taking the time to report this.

What would you like added/supported?
I accidentally assigned a value to an output logic `adc_clk` in two synchronous blocks. It took a while to debug on the real hardware, as I couldn't get this bit to toggle. Even Vivado failed to report this issue. It would be a great feature if this was detected at build time and reported as an error. This will also help beginners like me to figure out issues easily and sooner.

Attached is an example of a simple, reproducible module that can be compiled with Verilator. 

[adc_clk_gen.cpp.txt](https://github.com/user-attachments/files/18639834/adc_clk_gen.cpp.txt)
[adc_clk_gen.sv.txt](https://github.com/user-attachments/files/18639835/adc_clk_gen.sv.txt)

Build flags
```
verilator -Wall --trace --trace-max-array 1024 --trace-max-width 32 --exe --build -cc -j 0 $(CPPFILE) $(SVFILE)  -CFLAGS "-g -O0"
```

What 'verilator --version' are you using?  Did you try it with the git master version?
```
> verilator --version
Verilator 5.032 2025-01-01 rev v5.032
```

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this in Verilator yourself?
I have fixed the issue by simply removing the assignment to `adc_clk`, but it would be great if this feature is implemented

wsnyder, 2025-02-15:
Pull welcome.  Note this should be a synthesis-only warning, so presumably off by default.

wsnyder, 2025-02-26:
Note #5052 is a slightly different case, but likely should be covered by the same new checker.

wsnyder, 2026-09-13:
Fixed with #7968 above.

closed 2026-09-13 by wsnyder