verilator/verilator #5764: Managing large intermediate files

alexforencich, 2025-02-04:
Some very large files get written out during the process of running a simulation with verilator.  Specifically, I am seeing pairs of GCH files get written out which can add up to 100 MB or more for each run, but only for some testbenches.  As well as potentially hundreds of generated C++ files with a massive amount of redundancy.  With lots of tests (e.g. in a CI flow), this can eat up a lot of disk space and cause trouble (github actions has a storage limit of only 14 GB per runner).  Is there a way to automatically clean up all of the intermediate files during the build process such that only the executable needed to actually run the simulation is retained?

Also, I am using verilator with cocotb, so I'm not sure if this is an issue with verilator generally, or if it's something inherent to cocotb's process for building and running verilator.

On a related note, are there particular HDL constructs that result in Verilator has trouble with, resulting in mountains of generated code?  One relatively simple module that I ran under verilator (10G Ethernet MAC+PHY) produced about 180 generated C++ files, about 500 MB of temporary files in total and an output executable of 33 MB.  And it looks like the build was not hierarchical as all of the signal references are absolute, which is not a great sign as presumably multiple instances of the same core would result in many more copies of effectively the same intermediate code.  As the core in question here could be step-and-repeated in a design many times (order 10-100x) and combined with a lot of additional logic, this level of temporary file production for a single instance is worrying.

alexforencich, 2025-02-04:
I suspect part of the problem may be that verilator is not really doing compile-time constant elaboration.  Looks like it maybe it isn't evaluating the output of an LFSR/CRC matrix generator function (which has constant inputs and hence should reduce to a constant output), and as a result is generating a massive amount of code that runs at t=0 instead of doing that beforehand and simply baking in the resulting values.  I will be reworking that module relatively soon.

alexforencich, 2025-02-04:
Reworked that module to use a "proper" constant function and it is a definite improvement; verilator can certainly perform constant propagation when it wants to.  Down to 2 intermediate CPP files down to 2.  But, out of 8 tests, two of them dumped out two 70 MB GCH files, the other 6 did not dump out any GCH files.  Why is it that sometimes the GCH files are produced and sometimes they are not?  And when there would be, is there any kind of automated way to clean that up as soon as they are no longer necessary?

paul-demo, 2025-02-06:
I believe you can delete all intermediate files as soon as the simulation binary is compiled, so you could easily do that after the 'make' step that invokes gcc/clang to compile the verilated files, and just keep the (I think one) file that is required for running the test alongside the python code.

Cocotb's runner.py has build() and test() phases. We remove the `make` step from the build step and using the Verilator `--build` flag to let Verilator invoke the c++ compiler itself. After that gcc/clang compilation finishes, you could delete every intermediate file except the resulting binary, I think. I believe .gch files are intended to spend disk space in order to save time.

alexforencich, 2025-02-06:
So, the question is how to delete all of the intermediates in an automated manner so that the github runners don't run out of disk space.  I cannot access the runners directly, so manually deleting stuff is not possible.  In my case, I don't care if the tests take an extra 5 seconds to compile and/or run, but it's a problem if the whole shebang crashes due to running out of disk space.  The runners already have to download verilator and compile it from source anyway, which takes 5 pin per runner, as the version in the ubuntu repos is much too old.  In my case everything is run under tox/pytest with cocotb-test.  Is this something that should be handled in the upstream verilator makefiles?  Or perhaps the cocotb makefiles?  Or maybe even in cocotb-test?  Not sure what makes the most sense.

The repo in question where these tests are being implemented is here: https://github.com/fpganinja/taxi .  It's currently at around 800 tests (where each test is a completely separate verilator run) that get distributed across 20 runners, but the number of tests will be increasing significantly as more code is added, probably by at least a factor of two and potentially an order of magnitude.  If disk space is a problem, I can potentially add more runners as a workaround, but github only allows 20 to run at any one time.

paul-demo, 2025-02-06:
It looks like you're using [cocotb-test](https://github.com/themperek/cocotb-test/blob/master/cocotb_test/simulator.py#L1118), whose source code seems very similar (if not directly the inspiration for) cocotb's [runner.py](https://github.com/cocotb/cocotb/blob/master/src/cocotb_tools/runner.py).

You could edit either of those two runners at runtime from the python from which you invoke them. You would just patch the command list that gets produced at the link below. In cocotb-test it looks like there are 3 steps in the cmd[] list: perl, make, run-binary. You would just insert a fourth one: perl, make, cleanup, run-binary.
https://github.com/themperek/cocotb-test/blob/master/cocotb_test/simulator.py#L1118

This patch can be done at runtime because it's python. You just get whatever runner object you're using and make a reference to the get-build-command function, wrap it with a new function, and reassign the wrapped command-getter in its place.

In the longer term, you could probably mix that functionality into either of those projects or into Verilator. For example, maybe you add a `clean_everything_but_the_binary` makefile target to Cocotb here:
https://github.com/cocotb/cocotb/blob/f27ec7a23a07818019da23697dfc44e62c12905f/src/cocotb_tools/makefiles/simulators/Makefile.verilator#L82-L85
And then optionally invoke that make target from whichever Cocotb runner you happen to be using.

Hope this helps a little bit.

paul-demo, 2025-02-06:
I'm sure you can come up with some makefile magic too. Just suggested `rm {sim_build}.gch` might get you 90% of the way there. See also this makefile, which the generated Vtop.mk imports:

https://github.com/verilator/verilator/blob/master/include/verilated.mk.in

alexforencich, 2025-02-06:
I added

    .INTERMEDIATE: $(VK_PCH_H).fast.gch $(VK_PCH_H).slow.gch

to the generated Vtop.mk, and then make automatically deleted the gch files at the end of the build.

I guess the question is, what files should be defined as intermediate, and what makefile should that go in.  For something like verilator, I think it makes sense to define pretty much everything as intermediate, as when you run verilator it writes out all of the cpp and header files, so all of the compilation steps will have to be re-run anyway.

paul-demo, 2025-02-06:
@wsnyder probably has some thoughts on that question. Maybe there should be a 3-level switch and a new command line argument to control it:

- keep every build artifact (current behavior)
- keep everything that is human readable; delete intermediate .o and .gch files which are binary
- delete everything except the final binary (good when you know you won't need incremental recompilation because, as you point out, re-verilating a design will tend to touch all the files anyway and so Make won't be able to do any incremental optimizations anyway)

wsnyder, 2025-02-07:
Running Verilator with e.g. identical Verilog code and a user C++ file that changes will only re-make that C++ file (it will not regenerate C++). This will require the intermediate files still exist to be fast, so my preference is not to change the build process to remove intermediates.

alexforencich, 2025-02-07:
So your recommendation is to patch verilated.mk as part of the CI flow before running the tests, then?

wsnyder, 2025-02-07:
Yes, I'd suggest the patch.  We can consider a patch to add an remove-intermediate build option once you've proven it works well, if you need it.

wsnyder, 2025-02-07:
BTW if you aren't already doing so, you can also use an action to build verilator into a artifact and restore it, so you don't need to rebuild every time.  Alternatively, restore from a docker container.  (Maybe cocotb already has a container ready?)

alexforencich, 2025-02-07:
I could get more creative with build artifacts, but TBH I don't really care how long it takes to run so long as it's not taking many hours.  Creating an artifact during one run isn't going to save any time, in fact it will probably take longer as it will take time to wrangle the artifact, and it will introduce more points of failure.  Using an artifact form a previous run would save time, but potentially introduces more ways for things to break.  I'm not familiar with docker either, nor do I care to rope in some additional 3rd party image.

If this can be addressed upstream, great.  Otherwise, I'll just patch the verilator install directly before running the tests.

Also IIRC there is a .a file produced during the build process.  Is that .a file just the verilated DUT code?  It seems like saving that .a file would be sufficient even when changing a C++ testbench, everything used to produce the .a file could be deleted without causing any problems.

paul-demo, 2025-02-07:
Maybe a more elegant solution for you would be to use the `-run` capabilities of the CICD framework you're using. I'm not familiar with that but it appears from [their example](https://github.com/V0XNIHILI/install-verilator-action) you can just run this after installing verilator: `- run: verilator --version`

Doesn't this suggest that you could just append your `. INTERMEDIATE` stuff onto the bottom of the file like this? I think this might be a 1-liner that you add to your project's `regression-tests.yml` file.

Maybe like this?
```bash
echo ".INTERMEDIATE blah blah balh" >> $(verilator --getenv VERILATOR_ROOT)/include/verilated.mk
```

alexforencich, 2025-02-07:
Yeah, that's pretty much exactly what I had in mind, although it might need a sudo in there somewhere.  If it doesn't go in the upstream verilator repo, then I'll just add a line in the github actions yml file to modify the installation directly.

wsnyder, 2025-02-15:
Sounds like there's no Verilator change needed here in the end - Unless I hear otherwise I'll close next week.

alexforencich, 2025-02-15:
No, the most straightforward way to solve this is to patch the Verilator makefiles.  If the verilated design is built into a .a file or similar, then the "user" C++ portion can be built without needing the PCH files, correct?

alexforencich, 2025-02-15:
Also, it turns out that this seems to be a more annoying problem locally than on github actions, at least for the time being.  With all of the tests distributed across 20 runners (about 40 tests per runner), running `df` before and after `tox` doesn't show a significant increase in disk usage.  But locally, running all of the tests eats up 6 GB, and 4.6 GB of that are the GCH files.

    $ du -hs tb/
    5.9G    tb/
    $ find . -name "*.gch" | wc -l
    66
    $ find . -name "*.gch" -print0 | xargs -0 du -chs
    #snip
    4.6G    total

I also don't know why only some tests generate GCH files...there are 855 tests, but only 66 GCH files across all of them.  The tests that do generate GCH files seem to generate a pair of ~70MB files (Vtop__pch.h.slow.gch and Vtop__pch.h.fast.gch).  If every test generated a pair of 70 MB GCH files, that would be about 120 GB of disk space.  So this has the potential to be a rather serious problem if it isn't mitigated, and it seems to me that the proper place to mitigate it is a minor adjustment to the Verilator makefiles.

wsnyder, 2025-02-15:
When you say test, you mean your tests, not Verilator's test_regress?

What is the size of the sources?  If << the GCH size, perhaps this is a gcc bug - what sizes do you get from clang?

wsnyder, 2025-02-15:
Also compare to the .o sizes.  So for one large test what's the comparison of sizes of the .cpp, .gch and .o?

wsnyder, 2025-02-15:
And note if you have e.g. -ggdb it will be much larger - avoid that ;)

alexforencich, 2025-02-15:
So in this case there are "inner tests" and "outer tests".  The "outer tests" are handled by pytest and the "inner tests" are handled by cocotb.  Each "outer test" is a separate verilator build, with different RTL and/or top-level parameters.  So 855 tests = 855 separate runs of verilator + GCC.  

Here is a complete list of every output file for one of the tests:

    $ ls -alh
    total 143M
    drwxr-xr-x 2 alex alex 4.0K Feb 13 22:47 .
    drwxr-xr-x 3 alex alex 4.0K Feb 13 22:47 ..
    -rw------- 1 alex alex  694 Feb 13 22:47 55issxgl_results.xml
    -rwxr-xr-x 1 alex alex 391K Feb 13 22:47 test_taxi_axis_baser_rx_64
    -rw-r--r-- 1 alex alex  598 Feb 13 22:47 verilated.d
    -rw-r--r-- 1 alex alex  598 Feb 13 22:47 verilated_dpi.d
    -rw-r--r-- 1 alex alex  30K Feb 13 22:47 verilated_dpi.o
    -rw-r--r-- 1 alex alex 214K Feb 13 22:47 verilated.o
    -rw-r--r-- 1 alex alex  365 Feb 13 22:47 verilated_threads.d
    -rw-r--r-- 1 alex alex  30K Feb 13 22:47 verilated_threads.o
    -rw-r--r-- 1 alex alex  657 Feb 13 22:47 verilated_vpi.d
    -rw-r--r-- 1 alex alex 169K Feb 13 22:47 verilated_vpi.o
    -rw-r--r-- 1 alex alex  719 Feb 13 22:47 verilator.d
    -rw-r--r-- 1 alex alex 9.7K Feb 13 22:47 verilator.o
    -rw-r--r-- 1 alex alex 289K Feb 13 22:47 Vtop___024root__DepSet_h84412442__0.cpp
    -rw-r--r-- 1 alex alex  127 Feb 13 22:47 Vtop___024root__DepSet_h84412442__0.d
    -rw-r--r-- 1 alex alex  19K Feb 13 22:47 Vtop___024root__DepSet_h84412442__0.o
    -rw-r--r-- 1 alex alex 659K Feb 13 22:47 Vtop___024root__DepSet_h84412442__0__Slow.cpp
    -rw-r--r-- 1 alex alex  139 Feb 13 22:47 Vtop___024root__DepSet_h84412442__0__Slow.d
    -rw-r--r-- 1 alex alex 112K Feb 13 22:47 Vtop___024root__DepSet_h84412442__0__Slow.o
    -rw-r--r-- 1 alex alex  896 Feb 13 22:47 Vtop___024root__DepSet_h84412442__1__Slow.cpp
    -rw-r--r-- 1 alex alex  139 Feb 13 22:47 Vtop___024root__DepSet_h84412442__1__Slow.d
    -rw-r--r-- 1 alex alex 5.8K Feb 13 22:47 Vtop___024root__DepSet_h84412442__1__Slow.o
    -rw-r--r-- 1 alex alex 6.6K Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0.cpp
    -rw-r--r-- 1 alex alex  111 Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0.d
    -rw-r--r-- 1 alex alex 3.7K Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0.o
    -rw-r--r-- 1 alex alex  18K Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0__Slow.cpp
    -rw-r--r-- 1 alex alex  126 Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0__Slow.d
    -rw-r--r-- 1 alex alex  16K Feb 13 22:47 Vtop___024root__DepSet_heccd7ead__0__Slow.o
    -rw-r--r-- 1 alex alex  15K Feb 13 22:47 Vtop___024root.h
    -rw-r--r-- 1 alex alex 7.2K Feb 13 22:47 Vtop___024root__Slow.cpp
    -rw-r--r-- 1 alex alex   94 Feb 13 22:47 Vtop___024root__Slow.d
    -rw-r--r-- 1 alex alex  30K Feb 13 22:47 Vtop___024root__Slow.o
    -rw-r--r-- 1 alex alex 300K Feb 13 22:47 Vtop__ALL.a
    -rw-r--r-- 1 alex alex 1.9K Feb 13 22:47 Vtop_classes.mk
    -rw-r--r-- 1 alex alex 3.0K Feb 13 22:47 Vtop.cpp
    -rw-r--r-- 1 alex alex   29 Feb 13 22:47 Vtop.d
    -rw-r--r-- 1 alex alex  669 Feb 13 22:47 Vtop__Dpi.cpp
    -rw-r--r-- 1 alex alex   93 Feb 13 22:47 Vtop__Dpi.d
    -rw-r--r-- 1 alex alex  520 Feb 13 22:47 Vtop__Dpi.h
    -rw-r--r-- 1 alex alex  936 Feb 13 22:47 Vtop__Dpi.o
    -rw-r--r-- 1 alex alex 3.5K Feb 13 22:47 Vtop.h
    -rw-r--r-- 1 alex alex 2.4K Feb 13 22:47 Vtop.mk
    -rw-r--r-- 1 alex alex 8.6K Feb 13 22:47 Vtop.o
    -rw-r--r-- 1 alex alex  773 Feb 13 22:47 Vtop__pch.h
    -rw-r--r-- 1 alex alex  541 Feb 13 22:47 Vtop__pch.h.fast.d
    -rw-r--r-- 1 alex alex  70M Feb 13 22:47 Vtop__pch.h.fast.gch
    -rw-r--r-- 1 alex alex  541 Feb 13 22:47 Vtop__pch.h.slow.d
    -rw-r--r-- 1 alex alex  70M Feb 13 22:47 Vtop__pch.h.slow.gch
    -rw-r--r-- 1 alex alex  39K Feb 13 22:47 Vtop__Syms.cpp
    -rw-r--r-- 1 alex alex  101 Feb 13 22:47 Vtop__Syms.d
    -rw-r--r-- 1 alex alex 1.4K Feb 13 22:47 Vtop__Syms.h
    -rw-r--r-- 1 alex alex  72K Feb 13 22:47 Vtop__Syms.o
    -rw-r--r-- 1 alex alex  423 Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0.cpp
    -rw-r--r-- 1 alex alex  162 Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0.d
    -rw-r--r-- 1 alex alex 2.3K Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0.o
    -rw-r--r-- 1 alex alex  960 Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0__Slow.cpp
    -rw-r--r-- 1 alex alex  174 Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0__Slow.d
    -rw-r--r-- 1 alex alex 6.0K Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__DepSet_h7c64a03f__0__Slow.o
    -rw-r--r-- 1 alex alex 1.6K Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61.h
    -rw-r--r-- 1 alex alex 1.5K Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__Slow.cpp
    -rw-r--r-- 1 alex alex  145 Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__Slow.d
    -rw-r--r-- 1 alex alex 6.1K Feb 13 22:47 Vtop_taxi_axis_if__D40_U1_UB61__Slow.o
    -rw-r--r-- 1 alex alex 3.0K Feb 13 22:47 Vtop__ver.d
    -rw-r--r-- 1 alex alex 6.0K Feb 13 22:47 Vtop__verFiles.dat

Everything is relatively small except for the two massive GCH files, which correspond to 140MB of the total 143MB.

I'm thinking the only intermediate that really "needs" to be saved here is Vtop__ALL.a, just about everything else can presumably get deleted once Vtop__ALL.a is generated.

wsnyder, 2025-02-15:
What are the GCH sizes using clang?

alexforencich, 2025-02-15:
Dunno, how do I get it to build with clang instead of GCC?

wsnyder, 2025-02-15:
./configure CXX=clang++
then rebuild verilator.

I don't know if cocotb needs something else.

wsnyder, 2025-02-15:
Another thing to try, edit verilated.mk:

```
$(VK_OBJS_FAST): %.o: %.cpp $(VK_PCH_H).fast.gch
        $(OBJCACHE) $(CXX) $(OPT_FAST) $(CXXFLAGS) $(CPPFLAGS) $(VK_PCH_I_FAST) -c -o $@ $<

$(VK_OBJS_SLOW): %.o: %.cpp $(VK_PCH_H).slow.gch
        $(OBJCACHE) $(CXX) $(OPT_SLOW) $(CXXFLAGS) $(CPPFLAGS) $(VK_PCH_I_SLOW) -c -o $@ $<
```

To remove the PCH references, e.g. 

```
$(VK_OBJS_FAST): %.o: %.cpp
        $(OBJCACHE) $(CXX) $(OPT_FAST) $(CXXFLAGS) $(CPPFLAGS) -c -o $@ $<

$(VK_OBJS_SLOW): %.o: %.cpp
        $(OBJCACHE) $(CXX) $(OPT_SLOW) $(CXXFLAGS) $(CPPFLAGS) -c -o $@ $<
```

And report the build times with and without PCH.  I wonder if you're better just not using PCH at all, versus bothering to build it that hugely and remove it.

alexforencich, 2025-02-15:
I see, so I need to rebuild verilator from source instead of using the system package.  I'll look into that, but that definitely complicates things a bit, and if it's a GCC bug it will probably be quite some time before the fix is available in LTS distros.  So, it's still worth taking a quick look at the build process to see what files are actually necessary to keep things snappy, and what can be removed.

My understanding is that the verilated executable consists of three parts: the verilated design itself (`Vtop*`), standard verilator library components (`verilated*`), and "user" C++ top-level/testbench code (in my case I think that would be `verilator.o`).  Naturally you want to make the rebuild process as fast as possible.  When the HDL is changed, there isn't much that can be done - all the verilated C++ gets regenerated, so that part has to be fully recompiled.  But when the "user" part is changed, the verilated C++ doesn't have to get recompiled so long as certain intermediate files are kept around.  What at the GCH files need for, exactly?  It seems like they must be used as part of building the verilated code, not for the "user" code, so as long as the verilated code gets compiled down to a .o or .a file, then the GCH files can be removed without any build time penalties.  And it seems like `make` can automatically remove stuff by explicitly marking it as `.INTERMEDIATE`.

alexforencich, 2025-02-15:
Some quick build time tests:

Full build:

    real	0m17.043s
    user	0m15.929s
    sys	0m1.072s

Delete verilator.o and rerun:

    real	0m0.968s
    user	0m0.908s
    sys	0m0.057s

With your suggested modification, I get:

    real	0m19.836s
    user	0m18.927s
    sys	0m0.859s

So it looks like building the GCH files definitely help with the overall build time, we just need to figure out how to adjust the makefiles to not try to build them when not necessary.

If I add the line `.INTERMEDIATE: $(VK_PCH_H).fast.gch $(VK_PCH_H).slow.gch` to verilated.mk, I get the following for a full build:

    real	0m16.965s
    user	0m15.845s
    sys	0m1.086s

(And make automatically deletes the PCH files)

And for a rebuild I get:

    real	0m0.945s
    user	0m0.877s
    sys	0m0.066s

wsnyder, 2025-02-15:
Can you repeat with a "ccache -C" between each?  I would expect if only the main.cpp file changes (not verilator generated .cpp's) and if the .gch is kept around the build will be faster.

alexforencich, 2025-02-15:
Full build with clean cache:

    real	0m16.951s
    user	0m15.845s
    sys	0m1.072s

Delete verilator.o (cocotb toplevel) and rebuild:

    real	0m0.956s
    user	0m0.892s
    sys	0m0.062s

Full build with INTERMEDIATE line and clean cache:

    real	0m16.942s
    user	0m15.782s
    sys	0m1.125s

Delete verilator.o (cocotb toplevel) and rebuild:

    real	0m0.943s
    user	0m0.887s
    sys	0m0.054s

So cleaning the cache makes no appreciable difference, and deleting the GCH files makes no appreciable difference.

alexforencich, 2025-02-15:
Also, when I do the rebuild I get output like so:

    [alex@frodo test_taxi_axis_baser_rx_64]$ rm verilator.o 
    [alex@frodo test_taxi_axis_baser_rx_64]$ time make -f Vtop.mk 
    g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -Os  -c -o verilator.o /home/alex/Projects/taxi-clean/.tox/py3/lib/python3.13/site-packages/cocotb/share/lib/verilator/verilator.cpp
    g++    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_threads.o Vtop__ALL.a   -Wl,-rpath,/home/alex/Projects/taxi-clean/.tox/py3/lib/python3.13/site-packages/cocotb/libs -L/home/alex/Projects/taxi-clean/.tox/py3/lib/python3.13/site-packages/cocotb/libs -lcocotbvpi_verilator  -pthread -lpthread -latomic   -o test_taxi_axis_baser_rx_64
    
    real	0m0.943s
    user	0m0.887s
    sys	0m0.054s

So, the `verilator.o` file is built from cocotb's standard top-level C++ file, then `Vtop__ALL.a` is roped in along with the other `verilated*.o`files to produce the final executable.   Neither step seems to need the GCH files.  Definitely looks like the GCH files could be removed, along with any other intermediate file needed to produce `Vtop__ALL.a`.

wsnyder, 2025-02-16:
Did you clear the cache between trials (ccache -C)?

alexforencich, 2025-02-16:
Yes, I cleaned the cache before both of those full runs.  I did not clean it for the rebuild test after deleting verilator.o.

wsnyder, 2025-02-16:
Can you try that?  If the ccache hits the pch will certainly not be helpful.

I'm also still curious about clang.

I think based on this I'll accept the intermediate patch, I'm just not sure if it should be always there, or optional default on, or optional default off.