verilator/verilator #5766: Fix Unexpected Waveform when using always_comb and always_ff

YilouWang, 2025-02-05:
Hi, 

I encountered an unexpected waveform difference when running the same testbench in Verilator and QuestaSim. The design includes both combinational logic (always_comb) and sequential logic (always_ff).

```
always_comb begin
  data_n = data_q;  // Latch previous value
  out_valid = 0;
  if (in_valid) begin
    data_n = in_data; // Update immediately when valid
    out_valid = 1;
  end
end

always_ff @(posedge clk or negedge rst_n) begin
  if (~rst_n) begin
    data_q <= '0;
  end else begin
    data_q <= data_n; // Sequential update
  end
end

assign out_data = data_q;
assign out_data_prev = data_n; // Expected to reflect previous data_q
```

it seems that Verilator and QuestaSim handle this differently.

Verilator Output (GTKWave): 

[image]

QuestaSim Output(Vsim GUI):

[image]

Expected Behavior:
out_data_prev should hold the previous out_data value and not update immediately. The behavior in QuestaSim is expected, while Verilator’s behavior appears incorrect.

Possible Issue with Combinational + Sequential Logic

- The signal out_data_prev is assigned from data_n, which is updated inside always_comb.
- In QuestaSim, data_n correctly holds the previous data_q value until in_valid is asserted.
- In Verilator, data_n seems to update in a way that does not match traditional event-driven simulation, likely due to different scheduling in combinational logic.

I'd like to try to help with this, but I don't know where to start. 

So any insights/help would be appreciated!

YilouWang, 2025-02-05:
I put the complete test here.

```
module dut (
    input  logic        clk,
    input  logic        rst_n,
    input  logic        in_valid,
    input  logic [31:0] in_data,
    output logic [31:0] out_data,
    output logic [31:0] out_data_prev,
    output logic        out_valid
);

  logic [31:0] data_n, data_q;

  always_comb begin
    data_n = data_q;
    out_valid = 0;
    if (in_valid) begin
      data_n = in_data;
      out_valid = 1;
    end
  end

  always_ff @(posedge clk or negedge rst_n) begin
    if (~rst_n) begin
      data_q <= '0;
    end else begin
      data_q <= data_n;
    end
  end

  assign out_data = data_q;
  assign out_data_prev = data_n;
endmodule

module t_racing_issue;

  parameter CLK_PERIOD = 10;
  parameter NUM_ENTRIES = 10; // Number of test entries

  logic clk, rst_n;
  logic in_valid, out_valid;
  logic [31:0] in_data;
  logic [31:0] out_data, out_data_prev;

  dut uut (
    .clk(clk),
    .rst_n(rst_n),
    .in_valid(in_valid),
    .in_data(in_data),
    .out_valid(out_valid),
    .out_data(out_data),
    .out_data_prev(out_data_prev)
  );

  always # (CLK_PERIOD / 2) clk = ~clk;

  initial begin
    // Initialize signals
    clk = 0;
    rst_n = 0;
    in_valid = 0;
    in_data = '0;

    // Reset
    # (2 * CLK_PERIOD);
    rst_n = 1;

    // Apply test stimulus
    for (int i = 0; i < NUM_ENTRIES; i++) begin
      in_data = $random;
      in_valid = 1;
      # (CLK_PERIOD);
      in_valid = 0;
    end

    // Successful execution marker
    $write("*-* All Finished *-*");
    $finish;
  end

  // Dump waveforms
  initial begin
    $dumpfile("t_racing_issue.vcd");
    $dumpvars(0, t_racing_issue);
  end

endmodule
```

paul-demo, 2025-02-06:
Verilator definitely implements basic sequantial logic correctly. If you're just getting started I'd recommend you use some framework like Cocotb (which will dump sensible waves for you) or one of Dan's C++ testbench frameworks that you can find on his github or [here](https://zipcpu.com/blog/2017/06/21/looking-at-verilator.html).

In your case, you just have a nonfunctional testbench. This loop is going to set in_valid to 0, and then in the same timestep, set it back to 1, because there's no delay or awaited event between `in_valid = 0` and `in_valid = 1` in the next iteration.
```systemverilog
    // Apply test stimulus
    for (int i = 0; i < NUM_ENTRIES; i++) begin
      in_data = $random;
      in_valid = 1;
      # (CLK_PERIOD);
      in_valid = 0;
    end

```

closed 2025-02-07 by wsnyder

YilouWang, 2025-02-07:
@paul-demo 

Thank you very much for your response. I agree that in_valid keeps holding 1 in the testbench. However, the problem stays—I am still wondering why the waveform in Verilator looks different from VSim. Even if in_valid holds 1, there should still be a one-cycle delay.

Additionally, my other question is: if I want to fix this, where should I start?

wsnyder, 2025-02-07:
IEEE process ordering rules say a simulator can decide when to interleave statements between processes, and can compress out events, that is either answer is IEEE legal.  Verilator makes different and more aggressive choices than other simulators intentionally for speed.  This isn't something we will change.