verilator/verilator #5769: /*verilator lint_save/restore*/ breaks VFileContent reference counting

dsargeantbrcm, 2025-02-09:
We have been chasing random crashes when running verilator for a while. Anecdotally they seem to be related to the positioning of lint_save/restore comments. We have seen SEGVs and linked list corruption errors on 5.028 through to cc1133c0 on master. We have also seen some really inexplicable parsing syntax errors.

Building with "CXX_FLAGS="-fsanitize=address -static-libasan" gave a clue as to what might be going on. For example:
```
verilator --lint-only -debugi 5 test_regress/t/t_lint_restore_bad.v
...
--accepting rule at line 443 ("// verilator lint_restore")
=================================================================
==2678198==ERROR: AddressSanitizer: heap-use-after-free on address
```

Investigating further it looks like the VFileContent reference counting doesn't take into account the m_contentp copy in verilatorCmtLintSave(). When verilatorCmtLintRestore() is called the VFileContent can be deleted prematurely. I think the implicit FileLine copy constructor is currently used. Adding an explicit copy constructor to FileLine that also calls m_contentp->refInc() like FileLine(FileLine* fromp) seems to fix it. e.g:

```
diff --git a/src/V3FileLine.h b/src/V3FileLine.h
index e885cc7cd..13bbb8472 100644
--- a/src/V3FileLine.h
+++ b/src/V3FileLine.h
@@ -183,6 +183,19 @@ public:
         , m_filenameno{singleton().nameToNumber(filename)}
         , m_waive{false}
         , m_contentLineno{0} {}
+    explicit FileLine(const FileLine& from)
+        : m_msgEnIdx{from.m_msgEnIdx}
+        , m_filenameno{from.m_filenameno}
+        , m_waive{from.m_waive}
+        , m_contentLineno{from.m_contentLineno}
+        , m_firstLineno{from.m_firstLineno}
+        , m_firstColumn{from.m_firstColumn}
+        , m_lastLineno{from.m_lastLineno}
+        , m_lastColumn{from.m_lastColumn}
+        , m_contentp{from.m_contentp}
+        , m_parent{from.m_parent} {
+        if (m_contentp) m_contentp->refInc();
+    }
     explicit FileLine(FileLine* fromp)
         : m_msgEnIdx{fromp->m_msgEnIdx}
         , m_filenameno{fromp->m_filenameno}
```

This is my first go at reading the verilator source code, I might be wrong!

Thanks for all your work developing verilator.

wsnyder, 2025-02-09:
Makes sense, thanks for debugging.  Might you change this into a pull request?

Ideally we'd want a test to show the problem but I can appreciate that might be difficult and will accept without.

dsargeantbrcm, 2025-02-10:
Pull request submitted. First one, so I hope it looks ok. Unfortunately I've not found a standalone example that shows the issue yet and I had to resort to fsanitize and high debugi levels to see the potential for corruption, rather than it actually occurring.

wsnyder, 2025-02-10:
Thanks for debugging and converting to a pull.

closed 2025-02-10 by wsnyder