verilator/verilator #5775: Wrong simulation result after a multiplexer

flaviens, 2025-02-12:
Hi there!

It seems that Verilator simulates wrongly this small design.

OS:
```
Ubuntu 24.04.2 LTS
```
Verilator version:
```
Verilator 5.033 devel rev v5.032-42-g0507fb465
```
Verilator command:
```
verilator --trace --trace-underscore --cc --exe --build -Wall -Wno-WIDTHTRUNC -Wno-CASEOVERLAP -Wno-UNUSEDSIGNAL -Iobj_dir -Idv -build-jobs 8 top.v dv/toplevel.cc
```

The verilog code is generated but the issue is easy to follow.

`_01_` is 1 as expected, but the assignment to `_28_` is mistaken: it should  be `2` but it is `0`. I compared with Icarus, which gives the correct response.

[image]

`top.v`
```
module top(clk_i, a_i, b_i, s_i, y_o, a_i_t0, b_i_t0, s_i_t0, y_o_t0);
  wire [3:0] _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  wire [3:0] _04_;
  wire _05_;
  wire [3:0] _06_;
  wire [3:0] _07_;
  wire _08_;
  wire _09_;
  wire _10_;
  wire _11_;
  wire _12_;
  wire _13_;
  wire _14_;
  wire _15_;
  wire _16_;
  wire _17_;
  wire _18_;
  wire _19_;
  wire [11:0] _20_;
  wire [11:0] _21_;
  wire [11:0] _22_;
  wire [11:0] _23_;
  wire [11:0] _24_;
  wire [11:0] _25_;
  wire [11:0] _26_;
  wire [11:0] _27_;
  wire [11:0] _28_;
  wire [11:0] _29_;
  input [11:0] a_i;
  wire [11:0] a_i;
  input [11:0] a_i_t0;
  wire [11:0] a_i_t0;
  input [47:0] b_i;
  wire [47:0] b_i;
  input [47:0] b_i_t0;
  wire [47:0] b_i_t0;
  input clk_i;
  wire clk_i;
  input [3:0] s_i;
  wire [3:0] s_i;
  input [3:0] s_i_t0;
  wire [3:0] s_i_t0;
  output [11:0] y_o;
  wire [11:0] y_o;
  output [11:0] y_o_t0;
  wire [11:0] y_o_t0;
  function [11:0] _30_;
    input [11:0] a;
    input [47:0] b;
    input [3:0] s;
    casez (s) // synopsys parallel_case
      4'b???1:
        _30_ = b[11:0];
      4'b??1?:
        _30_ = b[23:12];
      4'b?1??:
        _30_ = b[35:24];
      4'b1???:
        _30_ = b[47:36];
      default:
        _30_ = a;
    endcase
  endfunction
  assign y_o = _30_(a_i, b_i, s_i);
  assign _06_ = _04_ | s_i;
  assign _20_ = _04_[0] ? b_i[11:0] : y_o;
  assign _22_ = _04_[1] ? b_i[23:12] : y_o;
  assign _24_ = _04_[2] ? b_i[35:24] : y_o;
  assign _26_ = _04_[3] ? b_i[47:36] : y_o;
  assign _21_ = _06_[0] ? b_i_t0[11:0] : 12'h000;
  assign _23_ = _06_[1] ? b_i_t0[23:12] : 12'h000;
  assign _25_ = _06_[2] ? b_i_t0[35:24] : 12'h000;
  assign _27_ = _06_[3] ? b_i_t0[47:36] : 12'h000;
  assign _01_ = | _07_;
  assign _28_ = _01_ ? a_i : y_o;
  assign _29_ = _01_ ? a_i_t0 : y_o;
  assign _08_ = { _26_[0], _24_[0], _22_[0], _20_[0] } != { _20_[0], _20_[0], _20_[0], _20_[0] };
  assign _09_ = { _26_[1], _24_[1], _22_[1], _20_[1] } != { _20_[1], _20_[1], _20_[1], _20_[1] };
  assign _10_ = { _26_[2], _24_[2], _22_[2], _20_[2] } != { _20_[2], _20_[2], _20_[2], _20_[2] };
  assign _11_ = { _26_[3], _24_[3], _22_[3], _20_[3] } != { _20_[3], _20_[3], _20_[3], _20_[3] };
  assign _12_ = { _26_[4], _24_[4], _22_[4], _20_[4] } != { _20_[4], _20_[4], _20_[4], _20_[4] };
  assign _13_ = { _26_[5], _24_[5], _22_[5], _20_[5] } != { _20_[5], _20_[5], _20_[5], _20_[5] };
  assign _14_ = { _26_[6], _24_[6], _22_[6], _20_[6] } != { _20_[6], _20_[6], _20_[6], _20_[6] };
  assign _15_ = { _26_[7], _24_[7], _22_[7], _20_[7] } != { _20_[7], _20_[7], _20_[7], _20_[7] };
  assign _16_ = { _26_[8], _24_[8], _22_[8], _20_[8] } != { _20_[8], _20_[8], _20_[8], _20_[8] };
  assign _17_ = { _26_[9], _24_[9], _22_[9], _20_[9] } != { _20_[9], _20_[9], _20_[9], _20_[9] };
  assign _18_ = { _26_[10], _24_[10], _22_[10], _20_[10] } != { _20_[10], _20_[10], _20_[10], _20_[10] };
  assign _19_ = { _26_[11], _24_[11], _22_[11], _20_[11] } != { _20_[11], _20_[11], _20_[11], _20_[11] };
  assign y_o_t0 = { _19_, _18_, _17_, _16_, _15_, _14_, _13_, _12_, _11_, _10_, _09_, _08_ } | _21_;
  assign _00_ = ~ s_i;
  assign _07_ = _00_ | s_i_t0;
  assign _03_ = _02_ & _07_[3];
  assign _05_ = 1'h1 & _07_[1];
  assign _02_ = _05_ & _07_[2];
  assign _04_ = s_i_t0 & { _03_, _02_, _05_, 1'h1 };
endmodule
```

The testbench is basically just attributing the correct values to the signals + some boilerplate code.
`testbench.h`
```
#include "Vtop.h"
#include "verilated.h"

#if VM_TRACE
#if VM_TRACE_FST
#include <verilated_fst_c.h>
#else
#include <verilated_vcd_c.h>
#endif // VM_TRACE_FST
#endif // VM_TRACE

#include <iostream>
#include <stdlib.h>

#ifndef TESTBENCH_H
#define TESTBENCH_H

const int kResetLength = 5;
#if VM_TRACE
// Depth of the trace.
const int kTraceLevel = 6;
#endif // VM_TRACE

typedef Vtop Module;

// This class implements elementary interaction with the design under test.
class Testbench {
 public:
  Testbench(const std::string &trace_filename = "trace.vcd")
      : module_(new Module), tick_count_(0l) {
#if VM_TRACE
#if VM_TRACE_FST
    trace_ = new VerilatedFstC;
#else // VM_TRACE_FST
    trace_ = new VerilatedVcdC;
#endif // VM_TRACE_FST
    module_->trace(trace_, kTraceLevel);
    trace_->open(trace_filename.c_str());
#endif // VM_TRACE
  }

  ~Testbench(void) { close_trace(); }

  void close_trace(void) {
#if VM_TRACE  
    trace_->close();
#endif // VM_TRACE
  }

  /**
   * Performs one or multiple clock cycles.
   *
   * @param num_ticks the number of ticks to perform.
   */
   // @return false.
  bool tick(int num_ticks = 1, bool false_tick = false) {
    for (size_t i = 0; i < num_ticks || num_ticks == -1; i++) {
      tick_count_++;

      module_->clk_i = 0;
      module_->eval();

#if VM_TRACE
      trace_->dump(5 * tick_count_ - 1);
#endif // VM_TRACE

      module_->clk_i = !false_tick;
      module_->eval();

#if VM_TRACE
      trace_->dump(5 * tick_count_);
#endif // VM_TRACE

      module_->clk_i = 0;
      module_->eval();

#if VM_TRACE
      trace_->dump(5 * tick_count_ + 2);
      trace_->flush();
#endif // VM_TRACE
    }
    return false;
  }

  std::unique_ptr<Module> module_;
 private:
  vluint32_t tick_count_;

#if VM_TRACE
#if VM_TRACE_FST
  VerilatedFstC *trace_;
#else // VM_TRACE_FST
  VerilatedVcdC *trace_;
#endif // VM_TRACE_FST
#endif // VM_TRACE

  // Masks that contain ones in the corresponding fields.
  uint32_t id_mask_;
  uint32_t content_mask_;
};

#endif // TESTBENCH_H
```

`toplevel.cc`
```
#if VM_TRACE
#endif // VM_TRACE
#include "testbench.h"

static unsigned long run_test(Testbench *tben) {
    tben->module_->a_i = 2;
    tben->module_->a_i_t0 = 0;
    tben->module_->b_i = 0;
    tben->module_->b_i_t0 = 0;
    tben->module_->s_i = 1;
    tben->module_->s_i_t0 = 1;
    tben->tick(1);
    printf("y_o    = %d\n", tben->module_->y_o);
    printf("y_o_t0 = %d\n", tben->module_->y_o_t0);

    return 0;
}

int main(int argc, char **argv, char **env) {

  Verilated::commandArgs(argc, argv);
  Verilated::traceEverOn(VM_TRACE);

  Testbench *tb = new Testbench();

  unsigned int duration = run_test(tb);

  ////////
  // Display the results.
  ////////
#if !VM_TRACE
  std::cout << "Testbench complete!" << std::endl;
  std::cout << "Elapsed time: " << std::dec << duration << "." << std::endl;
#else // VM_TRACE
  std::cout << "Testbench with traces complete!" << std::endl;
  std::cout << "Elapsed time (traces enabled): " << std::dec << duration << "." << std::endl;
#endif // VM_TRACE

  delete tb;
  exit(0);
}
```

Thanks!
Flavien

flaviens, 2025-02-12:
Hi @wsnyder ,
Thank you very much as always for looking at it so quickly.
Indeed inside the Verilog it actually runs and displays correctly, it's only the tracing that seems to be incorrect.
Maybe is it due to the fact that `_28_` is not used afterward?
I let this issue open in case someone would like to investigate the tracing issue further. Please feel free to close it if this is too low priority.
Thanks again.
Flavien

wsnyder, 2025-02-14:
Can you indicate how tracing is wrong - that is what signal and what is expected?

flaviens, 2025-02-15:
The _28_ that should be 2 and is reported as 0 on the waveform in the above message, if I'm not mistaken.

wsnyder, 2025-02-15:
I get a 2 in the VCD dump (when use --trace-underlines to get it).

flaviens, 2025-02-15:
Hi @wsnyder , somehow I cannot reproduce it anymore by following the exact same steps. I tried several Verilator versions and tried in a Docker container. Strange. Then I'll close this issue. Thank you very much for your support!

closed 2025-02-15 by flaviens