verilator/verilator #5782: split_var internal error

toddstrader, 2025-02-14:
See #5781

wsnyder, 2025-02-14:
Error for reference:

```
%Error: Internal Error: t/t_0.v:22:22: ../V3SplitVar.cpp:1128: not enough split variables
                                     : ... note: In instance 't'
-node: VARREF 0x555557132410 <e1161#> {f22aw} @dt=0x55555712e3c0@(w512)p[31:0]  data__0287__029 [RV] <- VAR 0x555557130c60 <e1157#> {f20ax} @dt=0x55555$
   22 |       data_out = data[7];
      |                      ^
```

Code:

```
module t(/*AUTOARG*/
   // Outputs
   data_out
   );

   output logic [31:0][15:0] data_out;

   // Test loop
   initial begin
      $write("*-* All Finished *-*\n");
      $finish;
   end

   logic [31:0][15:0] data [8] /*verilator split_var*/;
   always begin
      data_out = data[7];
   end
endmodule
```

wsnyder, 2025-02-14:
@yTakatsukasa might you take a look if you get a chance? thanks.

I'm closing #5781, the pull that has the error, simply because it's just a failing test vs a fix so no need to keep around as open.

PR #5785 (2025-02-17): Fix #5782 internal error of split_var
Fixes #5782
The related code used basicp even for packed arrays.
It caused wrong hi(), and lo().

As usual, a test to reproduce the error is pushed, then fix later.

The test is from #5781 . Thank you @toddstrader
closed 2025-02-17 by wsnyder