verilator/verilator #5784: Verilator compile fatal error

jordankrim, 2025-02-14:
I am trying to compile Verilator 4.034 (with SystemC 2.3.3) on Ubuntu 22.04.5 LTS and I am getting this error message:

Compile flags:  g++ -I/usr/local/include -I/usr/local/include -std=gnu++14 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="/mnt/data/Project/work_jordan/systemc-2.3.3/include" -DDEFENV_SYSTEMC_LIBDIR="/mnt/data/Project/work_jordan/systemc-2.3.3/lib-linux64" -DDEFENV_VERILATOR_ROOT="/mnt/data/Project/work_jordan/verilator-4.034"
g++ -I/usr/local/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"/mnt/data/Project/work_jordan/systemc-2.3.3/include\" -DDEFENV_SYSTEMC_LIBDIR=\"/mnt/data/Project/work_jordan/systemc-2.3.3/lib-linux64\" -DDEFENV_VERILATOR_ROOT=\"/mnt/data/Project/work_jordan/verilator-4.034\" -Wno-char-subscripts -Wno-unused -c ../V3ParseGrammar.cpp
In file included from ../V3ParseGrammar.cpp:22:
verilog.c:313:10: fatal error: verilog.h: No such file or directory
compilation terminated.

wsnyder, 2025-02-14:
Please do a `make distclean` then attach here the complete ./configure and make output.

Note we test every commit on Ubuntu 22.04 so likely it's some issue on your system, most likely with bison.

jordankrim, 2025-02-15:
I have attached the 2 log files.

[verilator_4.034_compile.log](https://github.com/user-attachments/files/18806887/verilator_4.034_compile.log)
[verilator_4.034_configure.log](https://github.com/user-attachments/files/18806888/verilator_4.034_configure.log)

wsnyder, 2025-02-15:
I should have noted earlier but 4.034 is over 5 years old, and as such predates Ubuntu 22.04.  Use the latest release or master git branch.

closed 2025-02-15 by wsnyder

jordankrim, 2025-02-15:
I know that you closed this issue, however, I wanted to respond as to why I am utilizing a very old version of Verilator:

(1) The developer of this design said that they used the following tools to verify the design and I wanted to try to replicate that:
- SystemC 2.3.3 built using C++17
- GCC 9.4.0
- Python 3.8.10
- **Verilator 4.034**

(2) Verilator 4.034 was built on Redhat 8 successfully, however, on 2 of the 5 tests it got a segmentation fault and 1 it hung, so I wanted to see if the results were the same on Ubuntu.

(3) When I tried to build Verilator 5.021 on Redhat 8, I could not.  It seemed to have something to do with SystemC.  I never was able to resolve it, so perhaps moving to Ubuntu might help?

So, you said something about bison.   The version of bison on Redhat 8 (that worked) is the same on Ubuntu - (GNU Bison) 3.8.2, so what do you think the issue might be with bison?  Do I need a more updated version?

wsnyder, 2025-02-16:
One problem is that 4.034 of verilator predates that ubuntu, and so likely does not have fixes to support that gcc(/bison?)

You'll need to debug. Usually that error means something broke running bison, it could just be that bison is incompatible with that verilator version, maybe the command line flags changed?  You could try looking at the Verilator git history to see how bison calls and builds have changed from 4.034 forward to mainline, which works.  Or do git bisect to find oldest version that works and compare Makefiles and generated bison files between that and 4.034.  Or debug 4.034 directly.