verilator/verilator #5786: `$time import no units` error on realtime parameter override with unit

colluca, 2025-02-18:
The following code:
```systemverilog
vip_snitch_cluster #(
  .ClkPeriod(1ns)
) vip (.*);
```
Where `ClkPeriod` is a parameter defined as:
```systemverilog
module vip_snitch_cluster #(
  parameter realtime ClkPeriod = 10ns
)
```
Fails on the following assertion:
https://github.com/verilator/verilator/blob/8ff77e9d47351b0a59114929880687839a51840b/src/V3Width.cpp#L1519

We are using Verilator v5.032 in the official Docker container `verilator/verilator:v5.032`. A full log is available at the following link:
https://github.com/pulp-platform/snitch_cluster/actions/runs/13393597517/job/37407291158?pr=204

wsnyder, 2025-02-19:
Thanks for the good test case, easy with that, fixed.

closed 2025-02-19 by wsnyder

colluca, 2025-02-19:
Thanks a lot for the fix :)