verilator/verilator #5787: Compiling Verilated code with 5.033 gives error

jordankrim, 2025-02-18:
Ubuntu 22.04
Verilator 5.033
SystemC 2.3.3

SystemC and Verilator *seem* to compile fine and pass the tests.
When I try to compile the verilated code (on the RTL) I get this:

g++   -I.  -MMD -I/mnt/data/MIDS_WPE/work_jordan/verilator_5.033/include -I/mnt/data/MIDS_WPE/work_jordan/verilator_5.033/include/vltstd -DVM_COVERAGE=0 -DVM_SC=1 -DVM_TIMING=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-subobject-linkage -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -I/mnt/data/MIDS_WPE/work_jordan/systemc-2.3.3/include  -include Vhydra_tile_main__pch.h.slow -c -o Vhydra_tile_main___024root__DepSet_h7d236311__4__Slow.o Vhydra_tile_main___024root__DepSet_h7d236311__4__Slow.cpp
Vhydra_tile_main___024root__DepSet_h7d236311__3__Slow.cpp: In function ?void Vhydra_tile_main___024root___stl_sequent__TOP__12(Vhydra_tile_main___024root*)?:
Vhydra_tile_main___024root__DepSet_h7d236311__3__Slow.cpp:12874:49: error: invalid operands of types ?unsigned int? and ?sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*? to binary ?operator&?
12874 |     vlSelfRef.cnoc_maxi_req[0U] = ((0xfffffffcU & vlSelfRef.cnoc_maxi_req[0U])
      |                                     ~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                     |                                       |
      |                                     unsigned int                            sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*

The same exact RTL compiled fine on Redhat 8, SystemC 2.3.3 and Verilator 4.034, so I am not sure why upgrading to 5.033 now gives errors on Ubuntu.

wsnyder, 2025-02-19:
Can you please make a minimum example that shows the issue?  Run with --runtime-debug to annotate the C code if you are not sure where that code comes from.

jordankrim, 2025-02-19:
this isn't my RTL, so I am not familiar with it at all, so creating a minimum example is not possible.

closed 2025-03-07 by wsnyder