verilator/verilator #5792: error: one or more PCH files were found, but they were invalid

zzhwarrior, 2025-02-20:
> Help please！I tried to include two header files then the error occurred: `cc1plus: error: one or more PCH files were found, but they were invalid
> cc1plus: note: use -Winvalid-pch for more information
> <command-line>: fatal error: VTestHarness__pch.h.fast: No such file or directory`
> `/root/chipyard/tests/vortex/tool/bin/verilator --cc --exe --no-timing /root/chipyard/generators/boom/src/main/resources/vsrc/vortex/VX_gpu_pkg.sv -CFLAGS " -O3 -std=c++17 -I/root/chipyard/.conda-env/riscv-tools/include -I/root/chipyard/tools/DRAMSim2 -I/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/gen-collateral   -DCY_FST_TRACE -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR " 
> --compiler-include/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/chipyard.TestHarness.VORTEXTestConfig.plusArgs 
> --compiler-include/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/gen-collateral/verilator.h 
> -LDFLAGS " -L/root/chipyard/.conda-env/riscv-tools/lib -Wl,-rpath,/root/chipyard/.conda-env/riscv-tools/lib -L/root/chipyard/sims/verilator -L/root/chipyard/tools/DRAMSim2 -lriscv -lfesvr -ldramsim " -I/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/gen-collateral   --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal --timescale 1ns/1ps --max-num-width 1048576 +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --top-module TestHarness --vpi -f /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/sim_files.common.f  -o /root/chipyard/sims/verilator/simulator-chipyard-VORTEXTestConfig -Mdir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.VORTEXTestConfig/chipyard.TestHarness.VORTEXTestConfig` 

 _Originally posted by @zzhwarrior in [#5139](https://github.com/verilator/verilator/issues/5139#issuecomment-2672379689)_

wsnyder, 2025-02-24:
Does the suggested `-CFLAGS -Winvalid-pch` give more information?

What is the compiler command printed right before the error?

Which `gcc --version` and operating system is this?

wsnyder, 2025-03-16:
Closing for now, please reopen if more info.

closed 2025-03-16 by wsnyder