verilator/verilator #5801: Register dumps

andrewvoznytsa, 2025-02-23:
I'd add [this register](https://github.com/verilator/verilator/blob/master/examples/make_tracing_c/sub.v#L15) to my trace file. How could I do that? (I don't want to printf() its value in a clock loop - I want add it to the trace file).

I looked here and there and I'm almost sure that there is no way to do that. If I'm correct after my first quick look it seems like the variables are somehow skipped during trace generation process - i.e like VerilatedTraceSigKind::VARs are never added to trace. Could you please point me to relevant places to tweak and/or potential algorithm to implement to add such feature? If I develop such - would you be interested in merging it, if I submit pull request?

Verilator is mature project and if there is no such feature then probably there is some good reason. Could you please shed some light on that topic so I'll not waste time/re-invent a wheel?

wsnyder, 2025-02-24:
I don't understand your question.  You pointed to `count_c`, but that is already in the trace file created.

andrewvoznytsa, 2025-02-24:
My `count_c` had underscore in front of it. Adding `--trace-underscore`, which I missed to my shame, fixed the issue.

Anyway - thank you for you help! It gave me idea.

closed 2025-02-24 by andrewvoznytsa