verilator/verilator #5808: Simulation returning -11 return code SIGSEGV

jrwagz, 2025-02-26:
I'm spinning up on using verilator with cocotb, and I have a case where the verilator simulation exits with a returncode of `-11` or SIGSEGV even though the actual simulation seems to have passed and all assertions passed appropriately.

I'm unsure what exactly in my source code is causing this, and only a specific test and RTL code is causing this problem (i.e. other code works just as expected).

Is there a way to dump a deeper stacktrace or something similar to identify what is causing this to happen?

Here is the simplified output to help isolate what's going on if that's useful:

```
- V e r i l a t i o n   R e p o r t: Verilator 5.034 2025-02-24 rev v5.034
- Verilator: Built from 0.000 MB sources in 0 modules, into 0.000 MB in 0 C++ files needing 0.000 MB
- Verilator: Walltime 0.001 s (elab=0.000, cvt=0.000, bld=0.000); cpu 0.001 s on 1 threads; alloced 18.680 MB
make: Entering directory 'sim_build'
make: Nothing to be done for 'default'.
make: Leaving directory 'sim_build'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:117  in _embed_init_python              Using Python interpreter at /usr/bin/python3
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:102  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.034 2025-02-24
     0.00ns INFO     cocotb                             Seeding Python random module with 1740606981
     0.00ns INFO     cocotb                             Initialized cocotb v2.0.0.dev0+c91ad618 from /usr/local/lib/python3.10/dist-packages/cocotb
     0.00ns INFO     cocotb                             Running tests
     0.00ns INFO     cocotb.regression                  running test_module_name.testcase_name (1/1)
     0.00ns INFO     test                               Running testcase_name
  2000.00ns INFO     cocotb.regression                  test_module_name.testcase_name passed
  2000.00ns INFO     cocotb.regression                  *****************************************************************************************************
                                                        ** TEST                                         STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        *****************************************************************************************************
                                                        ** test_module_name.testcase_name                PASS        2000.00           0.02      98000.05  **
                                                        *****************************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                              2000.00           0.02      83121.36  **
                                                        *****************************************************************************************************
                                                        
- :0: Verilog $finish
Simulation failed: -11
```

wsnyder, 2025-02-26:
This seems perhaps more a question for cocotb.

But to answer ignoring cocotb, verilate using '--runtime-debug' then rather than running your simulation executable e.g. "Vmodel", run "gdb Vmodel" then on that signal do a backtrace "bt".

Closing as answered but feel free to follow up.

closed 2025-02-26 by wsnyder

jrwagz, 2025-02-26:
I think your assertion that this is more related to cocotb is likely on point.  I modified the testbench to use a systemverilog testbench, instead of a python cocotb testbench, and then verilated the source and ran the simulation with verilator without problems.

I would have tried the gdb debugging you mentioned, but it was more difficult than I was expecting, given the cocotb integration, hence why I tried the route of a system verilog testbench instead of cocotb to isolate cocotb from the equation.

Thanks for the quick feedback!

jrwagz, 2025-02-27:
For the benefit of future internet travelers, I figured out what was causing the problem.  In my cocotb testbench, I had started a custom coroutine to run in parallel which ran forever. However I had not handled the stopping of that coroutine correctly, so when the simulation finished things went sideways since I hadn't properly stopped that coroutine.