verilator/verilator #5815: Combinatorial trees, UNOPTFLAT, and cocotb

alexforencich, 2025-02-28:
I'm trying to figure out how to implement a tree structure in verilator in a way that satisfies the linter.  Unfortunately, I am running in to issues with UNOPTFLAT.  It seems like the issue is stemming from the fact that cocotb sets `public_flat_rw` and this prevents `split_var` from working correctly.

I have been doing some testing with `prim_arbiter_tree.sv` from opentitan as it does something similar to what I'm trying to implement and ostensibly it should work in verilator (https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/rtl/prim_arbiter_tree.sv).  Unfortunately I have not been able to get it working.  It does not work out of the box, reporting lots of UNOPTFLAT warnings  ("Signal unoptimizable: Circular combinational logic").

It seems like annotating with `split_var` should be the solution, but this results in SPLITVAR warnings "'req_tree' has split_var metacomment but will not be split because it is public.".

Is there a path to actually make this work correctly in verilator, or do I simply need to do a lint_off UNOPTFLAT?

alexforencich, 2025-02-28:
I also tried splitting an unpacked array and got a SPLITVAR warning that seems incorrect:

    %Warning-SPLITVAR: ../../../rtl/prim/taxi_penc.sv:42:17: 'stage_valid' has split_var metacomment but will not be split because its bitwidth is 1.
                                                           : ... note: In instance 'taxi_penc'
       42 | logic [W/2-1:0] stage_valid[LEVELS] /*verilator split_var*/ ;

In this case, `LEVELS` is 5 and `W = 2**LEVELS`, so both dimensions should be wider than 1.

gezalore, 2025-02-28:
UNOPTFALT in these cases usually shows up because you have some dependency between bits of packed variables:

```
module a(
  input wire i,
  output wire o
);
  logic [1:0] x;

  assign x[0] = i;
  assign x[1] = x[0];
  assign o = x[1];
endmodule
```

The pattern I use is, instead of packed variable, use a generate block:

```
 module a(
   input wire i,
   output wire o
 );

   for (genvar n = 0 ; n < 2; ++n) begin: gvars
     logic x;
   end

   assign gvars[0].x = i;
   assign gvars[1].x = gvars[0].x;
   assign o = gvars[1].x;

 endmodule
```

Obviously this requires all parts be accessed via constant indices, but that is generally true of these recursive structures you are describing.

Also note UNOPTFLAT is just a performance warning, it will still function correctly. How bit a performance hit this is depends on many things, can be very small, can be very large.

alexforencich, 2025-02-28:
Interesting, I have never looked at using hierarchical references for synthesis, outside of debugging.  Seems like that could have the potential to cause other problems later on, especially if I need to feed something into older tools via sv2v.

gezalore, 2025-02-28:
Beyond what I showed, and preprocessor macros doing effectively the same via token pasting (I know...), I can't think of another way of to not get the UNOPTFLAT without variable splitting. Some structures you could express as recursively instantiated parametrized modules, but that probably have eve more potential to break in other tools. If you don't want to do either of those, then you are probably stuck with lint_off.

alexforencich, 2025-02-28:
The module in question actually used to use recursive instantiation, but IIRC that caused trouble somewhere (can't recall where offhand) so I rewrote it to use loops instead.

Seems like it should be possible to split a 2D array though, I don't know why Verilator doesn't want to do that.  If that turns out to be a bug, then maybe fixing that is the long-term solution and I can disable those linter warnings in the meantime.

gezalore, 2025-02-28:
As you said, `--public-flat-rw` will prevent any splitting as it requires every variable to be present in the verilated output with it's original data type. You could also not add the global `--public-flat-rw` (I know that is the standard with cocotb), but selectively mark the signals you actually need to access in cocotb: https://verilator.org/guide/latest/extensions.html?highlight=public_flat_rw#cmdoption-verilator-public_-flat-flat_rd-flat_rw-_on The performance impact of `--public-flat-rw` on simulation speed is ~5-10x slowdown, so you might want to consider that just for this reason alone.

alexforencich, 2025-02-28:
Seems like splitting unpacked array elements could be a different story.  But that is an interesting thought, in most cases I drive stuff at the top-level, but there are possibly a couple of exceptions to that, which would not be too hard to annotate.  I just really don't like annotating anything unless there is truly no other option.  And naturally public_flat_rw does need to get turned on at some point, then the splitting will stop working.

closed 2025-03-02 by wsnyder