verilator/verilator #5820: Internal error on out of range assignment

gezalore, 2025-03-02:
This will die with a `%Error: Internal Error: ../V3EmitCFunc.cpp:106: Wide Op w/ no temp, perhaps missing op in V3EmitC?`

```verilog
   module top(
     input  wire [ 31:0] foo,
     output reg  [144:0] bar
   );
     assign bar[159:128] = foo;
   endmodule
```

This is due to V3Width rewriting the LHS into this, which looks nonsense to me:

```
SEL @dt=@(G/w32) decl[144:0]]
1: EXTEND @dt=@(G/w160)
1:1: VARREF @dt=@(w145)  bar [LV] => VAR @dt=@(w145)  bar OUTPUT [VSTATIC]  PORT
2: CONST @dt=@(G/sw8)  8'h80
3: CONST @dt=@(G/wu32/6)  ?32?h20
```

There *is* a SELRANGE warning, however, this is a case reduced from a 140k line codebase, which has quite a lot of warnings, so it's quite hared to notice. The same code claims to work with some 4.\* version of Verilator, though I have not verified that.

I'm also aware of #5490, but this is a static select, and the behaviour is specified in the standard.

Remarkably sv-bugpoint gets the original 140k lines down to 56, which can be further simplified into this. Thanks to the authors for that tool.

wsnyder, 2025-03-02:
SELRANGE is one of those warnings where disabling it often causes strange new code problems to turn up :(

I'll take a look (let me know if you already are).

gezalore, 2025-03-02:
> I'll take a look (let me know if you already are).

Please do, I'm still scared of V3Width.

closed 2025-03-08 by wsnyder

kislay536, 2025-06-06:
Hi,
I am using v5.030 and while building I got this same error but it doesn't come up when trying it with v5.036. Is there any patch or way to build it with v5.030?