verilator/verilator #5821: Warning with always_comb and coverage

pinicarus, 2025-03-02:
Hi,

I'm new to both SystemVerilog and Verilator, so I'm trying some basic stuff to get to know both the language and tool better.

I have the following files:

```
// imply.sv:
module imply(input logic p, input logic q, output logic r);
  always_comb begin
    r = p | q;
  end
endmodule
```

```
// imply_tb.sv:
module imply_tb();
  logic p;
  logic q;
  logic r;

  imply dut(.p(p), .q(q), .r(r));

  initial begin
    p = 1;
    q = 0;
   $strobe("@%t: %d, %d, %d", $time, p, q, r);
  #1 $finish;
endmodule
```

```
// top.sv:
module top;
  imply_tb tb;
endmodule
```

I'm running Verilator 5.028.
Simulation works as expected when compiled with this command: `verilator +1800-2023ext+sv --assert --assert-case --binary -j 0 -O3 -Wall -Wpedantic --trace --timing top.sv`

But when I try to activate coverage by adding these flags: `--coverage --trace-coverage`, I get two weird warnings:

```
%Warning-ALWCOMBORDER: dut.sv:6:2: Always_comb variable driven after use: 'vlCoverageLineTrace_imply__6_block'
                                   : ... note: In instance 'top.tb.dut'
    6 |  always_comb begin
      |  ^~~~~~~~~~~
                       imply_tb.sv:15:1: ... note: In file included from 'imply_tb.sv'
                       top.sv:20:1: ... note: In file included from 'top.sv'
                       ... For warning description see https://verilator.org/warn/ALWCOMBORDER?v=5.028
                       ... Use "/* verilator lint_off ALWCOMBORDER */" and lint_on around source to disable this message.
%Warning-UNOPTFLAT: imply.sv:6:2: Signal unoptimizable: Circular combinational logic: 'top.tb.dut.vlCoverageLineTrace_imply__6_block'
    6 |  always_comb begin
      |  ^~~~~~~~~~~
                    imply_tb.sv:15:1: ... note: In file included from 'imply_tb.sv'
                    top.sv:20:1: ... note: In file included from 'top.sv'
                    imply.sv:6:2:      Example path: top.tb.dut.vlCoverageLineTrace_imply__6_block
                    imply.sv:6:2:      Example path: ALWAYS
                    imply.sv:6:2:      Example path: top.tb.dut.vlCoverageLineTrace_imply__6_block
%Error: Exiting due to 2 warning(s)
```

I don't get which variable is "driven after use" or leads to a "circular combinational logic".
Any help?

Best regards.

wsnyder, 2025-03-02:
I'll take a look.  Note `--coverage-trace` isn't used much, you might want to avoid for now.

pinicarus, 2025-03-02:
> I'll take a look. Note `--coverage-trace` isn't used much, you might want to avoid for now.

Hi,

It does indeed work fine without this flag.
I'll avoid it for now.

Thanks for your help.

closed 2025-03-03 by wsnyder