verilator/verilator #5830: Support bit queue streaming

wsnyder, 2025-03-06:
For future UVM version, support streaming to/from bit queue.

```
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);
`define checks(gotv,expv) do if ((gotv) != (expv)) begin $write("%%Error: %s:%0d:  got='%s' exp='%s'\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

typedef bit bit_q_t[$];

module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   integer cyc = 0;
   reg [63:0] crc;
   reg [63:0] sum;

   // Take CRC data and apply to testblock inputs
   wire [31:0] in = crc[31:0];

   /*AUTOWIRE*/
   // Beginning of automatic wires (for undeclared instantiated-module outputs)
   wire [31:0]          out;                    // From test of Test.v
   // End of automatics

   Test test(/*AUTOINST*/
             // Outputs
             .out                       (out[31:0]),
             // Inputs
             .clk                       (clk),
             .in                        (in[31:0]));

   // Aggregate outputs into a single result vector
   wire [63:0] result = {32'h0, out};

   initial begin
      byte unsigned p[$];
      byte unsigned po[$];
      bit bits[$];
      string s;
      p = {8'h84, 8'haa};
      bits = {<< 8 {bit_q_t'({<< {p}}) }};
      bits.push_front(1'b0);  // Shift
      po = {<< 8 {bit_q_t'({<< {bits}}) }};
      s = $sformatf("p=%p", p);
      `checks(s, "p='{'h84, 'haa} ");
      s = $sformatf("bits=%p", bits);
      `checks(s, "bits='{'h0, 'h0, 'h0, 'h1, 'h0, 'h0, 'h0, 'h0, 'h1, 'h0, 'h1, 'h0, 'h1, 'h0, 'h1, 'h0, 'h1} ");
      s = $sformatf("po=%p", po);
      `checks(s, "po='{'h8, 'h55, 'h80} ");
   end

   // Test loop
   always @ (posedge clk) begin
`ifdef TEST_VERBOSE
      $write("[%0t] cyc==%0d crc=%x result=%x\n", $time, cyc, crc, result);
`endif
      cyc <= cyc + 1;
      crc <= {crc[62:0], crc[63] ^ crc[2] ^ crc[0]};
      sum <= result ^ {sum[62:0], sum[63] ^ sum[2] ^ sum[0]};
      if (cyc == 0) begin
         // Setup
         crc <= 64'h5aef0c8d_d70a4497;
         sum <= '0;
      end
      else if (cyc < 10) begin
         sum <= '0;
      end
      else if (cyc == 99) begin
         $write("[%0t] cyc==%0d crc=%x sum=%x\n", $time, cyc, crc, sum);
         if (crc !== 64'hc77bb9b3784ea091) $stop;
         // What checksum will we end up with (above print should match)
`define EXPECTED_SUM 64'h9721d4e989defb24
         `checkh(sum, `EXPECTED_SUM);
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule

module Test(/*AUTOARG*/
   // Outputs
   out,
   // Inputs
   clk, in
   );

   input clk;
   input [31:0] in;
   output reg [31:0] out;

   always_ff @(posedge clk) begin
      byte unsigned p[$];
      byte unsigned po[$];
      bit bits[$];
      p = {in[31:24], in[23:16], in[15:8], in[7:0]};
      // Convert 'p' byte stream into byte-opposite queue bit-stream
      bits = {<< 8 {bit_q_t'({<< {p}}) }};
      bits.push_front(1'b0);  // Shift
      po = {<< 8 {bit_q_t'({<< {bits}}) }};
      out = {po[3], po[2], po[1], po[0]};
   end
endmodule
```

wsnyder, 2025-04-04:
@RRozak maybe you could take a look after your other streaming patch?

RRozak, 2025-04-08:
Unfortunately, right now I need to focus on other tasks, but in the future I think I will work on extending the support for streaming expressions.

paul-demo, 2025-06-27:
Full solution now seems to work. See the updated PR (I just force-push to my own issue_5830 branch). Commit e2ea42a426a1643db68008518106cd3bf27e9ae4 is the one for consideration (currently the latest). Sorry I had to push several times to clean up stuff like files accidentally formatted with "make format".

paul-demo, 2025-07-02:
*Now* seems to work. PR comments resulted in some missed zero-padding cases which I have now fixed.

wsnyder, 2025-07-26:
Need to fix regression, see #6103.