verilator/verilator #5832: Support getting range from unpacked array handles

ktbarrett, 2025-03-06:
Support was added for finding unpacked arrays via the VPI, however this did not include support for getting the ranges of the arrays. This lack of functionality is causing cocotb runs to fail.

Per IEEE 1800-2017 37.11:

> vpi_iterate(vpiRange, instance_array_handle) shall return the set of instance array ranges beginning with the
leftmost range of the array declaration and iterating through the rightmost range. Using the vpiLeftRange/
vpiRightRange properties returns the bounds of the leftmost dimension of a multidimensional array.

For example, to get the range of the outermost dimension of an unpacked array the following code would be correct.
```c
vpiHandle iter = vpi_iterate(vpiRange, hdl);
assert(iter != NULL);
vpiHandle rangeHdl = vpi_scan(iter);
assert(rangeHdl != NULL);
int leftBound = vpi_get(vpiLeftRange, rangeHdl);
int rightBound = vpi_get(vpiRightRange, rangeHdl);
vpi_release_handle(rangeHdl);  // optional, but shouldn't error
vpi_release_handle(iter);  // ends iterator, *required*
```

wsnyder, 2025-03-06:
Makes sense. A pull would be appreciated, as I'm not sure when will get resolved otherwise.

ktbarrett, 2025-03-15:
Well looks like it's already implemented, just incorrectly. There's a confusing and incorrect mixing of 1-based and 0-based indexing for some reason.

ktbarrett, 2025-03-15:
Actually it looks like this is working as intended, it's just cocotb's code being written against 10+ year old incorrect commercial simulators.

closed 2025-03-15 by ktbarrett