verilator/verilator #5836: Variable splitting fails silently

gezalore, 2025-03-07:
Verilating this will not split up `v`, and there is no SPLITVAR warning to tell the user either:

```systemverilog
module bit_reverse(input wire clk);

  logic [2:0] v /* verilator split_var */ = 3'd0;

  always @(posedge clk) begin
     v <= v + 3'd1;
     $display("%1b%1b%1b", v[0], v[1], v[2]);
  end

endmodule
```

With `--debugi-V3SplitVar 9` I can see:

```
V3SplitVar.cpp:1165:In module bit_reverse var 'v' which has 2 lhs refs and 6 rhs refs will be split.
```

This suggests it should be split up, but then `splitPlan` returns a single full width result variable, so we hit the 2nd continue and end up not splitting it:

https://github.com/verilator/verilator/blob/96bffd49aa5640a527a24a026119160dff828308/src/V3SplitVar.cpp#L1165-L1169

What is the intention here? Is this a bug in `splitPlan`? If so, what is `splitPlan` supposed to return for this case?

wsnyder, 2025-03-07:
@yTakatsukasa

yTakatsukasa, 2025-03-09:
I think the current behavior is intended by guessing me in years ago.
The primary purpose of SplitVar is to resolve UNOPTFLAT.
SplitVar splits a variable according to how the variable is written, in other words, how it appears in the LHS.
Variable `v` in https://github.com/verilator/verilator/issues/5836#issue-2903224337 is assigned twice.
1.  `v <= v + 3'd1;` in `always`
2.  `v = 0` in the implicit initialization

In each assignment, the entire bit of `v` is written.
This is why `splitPlan` returns in that way.

Perhaps it's user friendly to emit a warning of "No need to split" in such case.

gezalore, 2025-03-09:
Thanks, it was not obvious to me from the algorithm that we are trying to split by LHS usage, but it makes very good sense.

closed 2025-03-09 by gezalore