verilator/verilator #5837: Error during build

jeras, 2025-03-07:
Git master version of Verilator compiling on Ubuntu 24.04.
I will just mention I sometimes experienced errors while compiling with `-j8` I usually just recompiled without.
Compiling (after a `git clean -xdf; autoconf; ./configure;`) returns the following error:

```
/usr/bin/ld: V3Const__gen.o: in function `AstNodeDType::skipRefp()':
/home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79:(.text._ZNK6AstVar13dtypeSkipRefpEv[_ZNK6AstVar13dtypeSkipRefpEv]+0x1f): undefined reference to `AstNodeDType::skipRefIterp(bool, bool) const'
/usr/bin/ld: /home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79:(.text._ZNK7AstNode18isClassHandleValueEv[_ZNK7AstNode18isClassHandleValueEv]+0x25): undefined reference to `AstNodeDType::skipRefIterp(bool, bool) const'
/usr/bin/ld: /home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79:(.text._ZN15SimulateVisitor12prettyNumberB5cxx11EPK8V3NumberP12AstNodeDType[_ZN15SimulateVisitor12prettyNumberB5cxx11EPK8V3NumberP12AstNodeDType]+0x51): undefined reference to `AstNodeDType::skipRefIterp(bool, bool) const'
/usr/bin/ld: /home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79:(.text._ZN12ConstVisitor5visitEP11AstArraySel[_ZN12ConstVisitor5visitEP11AstArraySel]+0xad): undefined reference to `AstNodeDType::skipRefIterp(bool, bool) const'
/usr/bin/ld: /home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79:(.text._ZN12ConstVisitor5visitEP11AstArraySel[_ZN12ConstVisitor5visitEP11AstArraySel]+0x182): undefined reference to `AstNodeDType::skipRefIterp(bool, bool) const'
/usr/bin/ld: V3Const__gen.o:/home/izi/VLSI/verilator/src/obj_dbg/../V3AstNodeDType.h:79: more undefined references to `AstNodeDType::skipRefIterp(bool, bool) const' follow
collect2: error: ld returned 1 exit status
make[2]: *** [../Makefile_obj:359: ../../bin/verilator_bin_dbg] Error 1
make[2]: Leaving directory '/home/izi/VLSI/verilator/src/obj_dbg'
make[1]: *** [Makefile:85: ../bin/verilator_bin_dbg] Error 2
make[1]: *** Waiting for unfinished jobs....
      Linking ../../bin/verilator_bin...
g++ -static-libgcc -static-libstdc++ -Xlinker -gc-sections -o ../../bin/verilator_bin V3Const__gen.o V3Error.o V3FileLine.o V3Graph.o V3GraphAcyc.o V3GraphAlg.o V3GraphPathChecker.o V3GraphTest.o V3Hash.o V3OptionParser.o V3Os.o V3ParseGrammar.o V3ParseImp.o V3ParseLex.o V3PreProc.o V3PreShell.o V3String.o V3ThreadPool.o V3Waiver.o Verilator.o  V3Ast.o V3AstNodes.o V3Broken.o V3Config.o V3EmitCBase.o V3EmitCConstPool.o V3EmitCFunc.o V3EmitCHeaders.o V3EmitCImp.o V3EmitCInlines.o V3EmitCPch.o V3EmitV.o V3File.o V3FuncOpt.o V3Global.o V3Hasher.o V3Number.o V3Options.o V3Stats.o V3StatsReport.o V3VariableOrder.o  V3Active.o V3ActiveTop.o V3Assert.o V3AssertPre.o V3Begin.o V3Branch.o V3CCtors.o V3CUse.o V3Case.o V3Cast.o V3Class.o V3Clean.o V3Clock.o V3Combine.o V3Common.o V3Coverage.o V3CoverageJoin.o V3Dead.o V3Delayed.o V3Depth.o V3DepthBlock.o V3Descope.o V3Dfg.o V3DfgAstToDfg.o V3DfgCache.o V3DfgDecomposition.o V3DfgDfgToAst.o V3DfgOptimizer.o V3DfgPasses.o V3DfgPeephole.o V3DfgRegularize.o V3DupFinder.o V3EmitCMain.o V3EmitCMake.o V3EmitCModel.o V3EmitCSyms.o V3EmitMk.o V3EmitXml.o V3ExecGraph.o V3Expand.o V3Force.o V3Fork.o V3Gate.o V3HierBlock.o V3Inline.o V3Inst.o V3InstrCount.o V3Interface.o V3Life.o V3LifePost.o V3LinkCells.o V3LinkDot.o V3LinkInc.o V3LinkJump.o V3LinkLValue.o V3LinkLevel.o V3LinkParse.o V3LinkResolve.o V3Localize.o V3MergeCond.o V3Name.o V3Order.o V3OrderGraphBuilder.o V3OrderMoveGraph.o V3OrderParallel.o V3OrderProcessDomains.o V3OrderSerial.o V3Param.o V3Premit.o V3ProtectLib.o V3Randomize.o V3Reloop.o V3Sampled.o V3Sched.o V3SchedAcyclic.o V3SchedPartition.o V3SchedReplicate.o V3SchedTiming.o V3SchedVirtIface.o V3Scope.o V3Scoreboard.o V3Slice.o V3Split.o V3SplitAs.o V3SplitVar.o V3StackCount.o V3Subst.o V3TSP.o V3Table.o V3Task.o V3Timing.o V3Trace.o V3TraceDecl.o V3Tristate.o V3Undriven.o V3Unknown.o V3Unroll.o V3Width.o V3WidthCommit.o V3WidthSel.o   -Wl,--whole-archive -l:libtcmalloc_minimal.a -Wl,--no-whole-archive   -lpthread -latomic -lm
make[2]: Leaving directory '/home/izi/VLSI/verilator/src/obj_opt'
make[1]: Leaving directory '/home/izi/VLSI/verilator/src'
make: *** [Makefile:149: verilator_exe] Error 2
```

wsnyder, 2025-03-07:
The Github actions builds here completely consistently.  Also note that the second link (of the optimized version) went fine with identical sources on your machine.  Perhaps your disks are mounted on NFS and the file server is slow?  Try building on local disks, or adding a `sleep 10` to before the links in the src/Makefile

Closing, as likely unrelated to verilator itself.

jeras, 2025-03-07:
Thanks, I will look into it myself, something similar happened before, but this is the first time it is repeatable. All storage is local SSD, so this is not it.

wsnyder, 2025-03-07:
Maybe clear ccache? `ccache --clear`

jeras, 2025-03-08:
Thanks, this did the trick.

closed 2025-03-08 by jeras