verilator/verilator #5838: Support for in-memory storage of signal traces

alikates, 2025-03-08:
Hello,

For context, I'm working on a library that interfaces Verilator with a [CXXRTL](https://cxxrtl.org/protocol.html)-compatible client. Those clients can request signal traces through a network socket in a dynamic way, asking for different sets of signals and ranges of time.

I've thought that for achieving this, a nice way would be to have in-memory storage of the signal traces and expose an API to query and control which ranges of time are captured, which signals, etc. 

For testing my application I've written a naive proof of concept that allows me to trace all the signals and stores them in a time-indexed hash-map, but I wanted to know if this would fit in the project before continuing development or opening a PR.

Thanks!

alikates, 2025-03-08:
I think it would also be interesting to have this together with #5813, having a disk-backed trace and an in-memory subset which allows fast queries from user code.

wsnyder, 2025-03-08:
Note Verilator must determine all dumped nets at compile time.  Of course you could runtime-filter what's compiled in for dumping.

I can see your use for this, but my concern would be if this will really be used enough to make the support worthwhile.  For now, I'd suggest hacking the dumpers as you need and get it all well usable before considering a patch.

alikates, 2025-03-08:
Thanks for the response!

Indeed, I had thought about some way of defining 'dump groups' or something similar to not have per-signal enables and avoid disabling all optimizations because of marking every signal as public. This is way out of my scope for now though.

The only 'issue' I have with not having this integrated in Verilator itself is that for adding new dumpers, i need to make modifications to V3EmitMk/CMake and V3Options. I don't really want to keep updating a fork with my changes, so maybe a compromise would be to allow user-defined dumpers?

wsnyder, 2025-03-08:
Could you just make functions that are compatible with the existing FST writer library?

wsnyder, 2025-03-08:
Closing as don't think we'll change anything until your API is settled, but please continue to discuss.

closed 2025-03-08 by wsnyder