verilator/verilator #5844: Inline randomization

WajahatRiaz, 2025-03-09:
When I'm trying to do inline randomization, some example below: 
1.  _ok = randomize(data) with {data inside {  [8'h21:8'h7E]  }; };_ 
2.  _ok = randomize(data) with { data dist {  [8'h41:8'h5a]:/80, [8'h61:8'h7a]:/20  };  };_

It generates an erro message saying: 
Calling implicit class method 'randomize' without being under class.

The systemverilog specification have a whole subsection on inline randomization. I've been part of the semiconductor industry for over 2 years now. Would love if verilator can start putting an effort to add features in regards, to layered testbenches. If there's an example out there then please do provide it. 

Verilator command: 
```
verilator --binary -j 8 --skip-identical \
		--timing -Wno-fatal -Wno-INSECURE -Wno-style -Wno-lint -quiet \
		${TBENCH} --top top \
		--autoflush --trace --coverage
```
My solver verilator: z3
My verilator verison: Verilator 5.035 devel rev v5.034-27-g258becd17. It's a git master verison.
My OS and distrbution details: Ubuntu 24.04.2 LTS

LRM: IEEE Standard for SystemVerilog—Unified Hardware Design, Specification, and Verification Language
Topic:  _18.7 In-line constraints—randomize() with_

wsnyder, 2025-03-09:
Is the randomize under a class?

If it is under a class, and you're getting the error, please post a complete standalone example.

If it's not, (and that's my reading of your message), see the examples in Verilator's test_regress for what's supported.

As to more support, I understand it's covered in IEEE but so are many other useful unsupported things (e.g. covergroups).  We're driven entirely by code contributions, so please consider making a pull to add what is important to you.

WajahatRiaz, 2025-03-12:
Nope, randomize is not under the class. This is a super important feature to implement though.

WajahatRiaz, 2025-03-12:
Btw do you have a layered testbench example available?

wsnyder, 2025-03-12:
>Btw do you have a layered testbench example available?

Not AFAIK.

closed 2025-03-15 by WajahatRiaz