verilator/verilator #5846: Expected an interface scope name to have at least one dot

Kreijstal, 2025-03-10:
I don't understand what the problem is but it is the same as https://github.com/verilator/verilator/issues/5237

```
%Error: Internal Error: /home/kreijstal/git/risc-v-lab/src/fv/system_tb.sv:76:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
                                                                                : ... note: In instance 'jtag_master::__VDynScope_76'
   76 |   jtag_master jtag ();
      |               ^~~~
```

```
kreijstal@kreijstalnuc:~/git/risc-v-lab$ verilator --version
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "en_GB.UTF-8",
	LC_NUMERIC = (unset),
	LC_COLLATE = (unset),
	LC_TIME = (unset),
	LC_MESSAGES = (unset),
	LC_MONETARY = (unset),
	LC_ADDRESS = (unset),
	LC_IDENTIFICATION = (unset),
	LC_MEASUREMENT = (unset),
	LC_PAPER = (unset),
	LC_TELEPHONE = (unset),
	LC_NAME = (unset),
	LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").
Verilator 5.034 2025-02-24 rev v5.034
kreijstal@kreijstalnuc:~/git/risc-v-lab$ uname -a
Linux kreijstalnuc 6.8.0-53-generic #55-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 17 15:37:52 UTC 2025 x86_64 GNU/Linux
``` 

I can give an even bigger example if that is helpful but questasim and vivado seem to have no problem with this.

wsnyder, 2025-03-10:
#5237 was fixed, so that isn't exactly it.

Please attach an openly licensed, completely self-contained test case so we can directly run what you provide.  Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

Kreijstal, 2025-03-10:
> [#5237](https://github.com/verilator/verilator/issues/5237) was fixed, so that isn't exactly it.
>
> Please attach an openly licensed, completely self-contained test case so we can directly run what [...]

the first error, was fixed, the second error wasn't, the source code is the same.

<details><summary>Details</summary>
<p>

```
kreijstal@kreijstalnuc:~/git/example$ verilator tb_hello_world.sv -main --timing --binary --trace-fst  dm_pkg.sv --debug
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "en_GB.UTF-8",
	LC_NUMERIC = (unset),
	LC_COLLATE = (unset),
	LC_TIME = (unset),
	LC_MESSAGES = (unset),
	LC_MONETARY = (unset),
	LC_ADDRESS = (unset),
	LC_IDENTIFICATION = (unset),
	LC_MEASUREMENT = (unset),
	LC_PAPER = (unset),
	LC_TELEPHONE = (unset),
	LC_NAME = (unset),
	LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").
export VERILATOR_ROOT='/usr/share/verilator'
Starting Verilator 5.034 2025-02-24 rev v5.034
- Verilator.cpp:646:  Option --verilate: Start Verilation
- V3File.cpp:218:        --check-times failed: no input obj_dir/Vtb_hello_world__verFiles.dat
- V3ParseImp.cpp:311: parseFile: verilated_std_waiver
- V3PreShell.cpp:90:  Preprocessing /usr/share/verilator/include/verilated_std_waiver.vlt
- V3PreShell.cpp:143:     Reading /usr/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:393: Lexing /usr/share/verilator/include/verilated_std_waiver.vlt
- V3ParseImp.cpp:311: parseFile: verilated_std
- V3PreShell.cpp:90:  Preprocessing /usr/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /usr/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:393: Lexing /usr/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:311: parseFile: tb_hello_world
- V3PreShell.cpp:90:  Preprocessing tb_hello_world.sv
- V3PreShell.cpp:143:     Reading tb_hello_world.sv
- V3ParseImp.cpp:379: Writing all preprocessed output to obj_dir/Vtb_hello_world__inputs.vpp
- V3ParseImp.cpp:393: Lexing tb_hello_world.sv
- V3ParseImp.cpp:311: parseFile: dm_pkg
- V3PreShell.cpp:90:  Preprocessing dm_pkg.sv
- V3PreShell.cpp:143:     Reading dm_pkg.sv
- V3ParseImp.cpp:393: Lexing dm_pkg.sv
- V3ParseImp.cpp:311: parseFile: jtag_master [LIB]
- V3PreShell.cpp:90:  Preprocessing jtag_master
- V3PreShell.cpp:143:     Reading jtag_master.sv
- V3ParseImp.cpp:393: Lexing jtag_master
- V3ParseImp.cpp:311: parseFile: rvlab_tests [LIB]
- V3PreShell.cpp:90:  Preprocessing rvlab_tests
- V3PreShell.cpp:143:     Reading rvlab_tests.sv
- V3ParseImp.cpp:393: Lexing rvlab_tests
- V3LinkLevel.cpp:42: modSortByLevel()
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_002_cellsort.tree
- V3LinkDot.cpp:4382: linkDotPrimary: 
- V3LinkJump.cpp:369: linkJump: 
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_007_linkjump.tree
- V3LinkInc.cpp:370:  linkIncrements: 
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_008_linkinc.tree
- V3Param.cpp:1534:   param: 
- V3LinkDot.cpp:4388: linkDotParamed: 
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_010_linkdotparam.tree
- V3Dead.cpp:556:     deadifyModules: 
[... 90 lines cut ...]
                     tb_hello_world.sv:7:1: ... note: In file included from 'tb_hello_world.sv'
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_054_active.tree
- V3Split.cpp:1009:   splitAlwaysAll: 
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_055_split.tree
- V3SplitAs.cpp:192:  splitAsAll: 
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_056_splitas.tree
- V3TraceDecl.cpp:715:traceDeclAll: 
%Error: Internal Error: tb_hello_world.sv:5:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
                                              : ... note: In instance 'jtag_master::__VDynScope_2'
-node: SCOPE 0x5555563d7170 <e46754> {e5ap}  TOP.jtag_master__03a__03a__VDynScope_2 [abovep=0x5555563bd8a0] [cellp=0x55555612ea60] [modp=0x55555630b500]
    5 |   jtag_master jtag();
      |               ^~~~
- V3Ast.cpp:1341:     Dumping obj_dir/Vtb_hello_world_990_final.tree
- V3StatsReport.cpp:231:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=/usr/share/verilator
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/bin/verilator_bin_dbg tb_hello_world.sv -main --timing --binary --trace-fst dm_pkg.sv --debug
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize /usr/bin/verilator_bin_dbg tb_hello_world.sv -main --timing --binary --trace-fst dm_pkg.sv --debug

``` 

</p>
</details> 

I would love to minimize the code further but im not sure what triggers the bug.

Kreijstal, 2025-03-10:
trying to remove some code makes the error not reproduce anymore

<details><summary>diff</summary>
<p>

```diff
diff --git a/dm_pkg.sv b/dm_pkg.sv
index af03d73..22b8e39 100644
--- a/dm_pkg.sv
+++ b/dm_pkg.sv
@@ -17,17 +17,11 @@
  */
 
 package dm;
-
   typedef struct packed {
     logic [6:0]  address;
     logic [31:0] data;
     logic [1:0]  op;
   } dmi_t;
-
-  typedef enum logic [1:0] {
-    DMINoError = 2'h0, DMIReservedError = 2'h1,
-    DMIOPFailed = 2'h2, DMIBusy = 2'h3
-  } dmi_error_e;
   
   localparam logic [3:0] DbgVersion013 = 4'h2;
   // size of program buffer in junks of 32-bit words
diff --git a/jtag_master.sv b/jtag_master.sv
index f20afcf..d58e664 100644
--- a/jtag_master.sv
+++ b/jtag_master.sv
@@ -1,6 +1,3 @@
-// Copyright Tobias Kaiser
-// Originally for PTC1.
-
 interface jtag_master;
 
   logic tck;
@@ -9,11 +6,7 @@ interface jtag_master;
   logic tdo;
   logic trst_n;
 
-  localparam verbose = '0;
-
-  // 50 MHz TCK
-  // -----------
-
+  // Clock generation
   always begin
     tck = 1'b0;
     #10000;
@@ -21,163 +14,26 @@ interface jtag_master;
     #10000;
   end
 
-  // JTAG tasks
-  // ----------
-
-  // All signals going to the DUT are written on falling edge and read by the
-  // DUT on the rising edge; all signals from the DUT are written on the
-  // falling edge by the DUT and read by this jtag_master on the rising edge.
-
   logic [4:0] ir_stored;
   logic ir_stored_valid;
 
   initial begin
-    ir_stored_valid <= '0;
-    ir_stored       <= '0;
+    ir_stored_valid = '0;
[... 463 lines cut ...]
 
 endmodule
diff --git a/tb_hello_world.sv b/tb_hello_world.sv
index d940ea5..eae9bac 100644
--- a/tb_hello_world.sv
+++ b/tb_hello_world.sv
@@ -16,13 +16,9 @@ module tb_hello_world;
     .clk_i(clk)
   );
 
-  // Testbench
-  always begin
+  initial begin
     $display("Hello, World!");
-
-    // Run the test
     uut.test_idcode();
-
     $finish;
   end

```

</p>
</details>

wsnyder, 2025-03-10:
Please try sv-bugpoint as described in the link above.

Kreijstal, 2025-03-10:
Thank you for suggesting sv-bugpoint, unfortunately I do not understand, the minized outputs do not reproduce the bug, maybe my check script is not good enough, anyway here goes my work:

<details><summary>Details</summary>
<p>
bugpoint-input.sv
```systemverilog
module tb_hello_world;
  logic clk;
  jtag_master jtag();
  initial begin
    clk = 1'b0;
    forever #5 clk = ~clk;
  end
  rvlab_tests uut (
    .jtag(jtag),
    .clk_i(clk)
  );
  always begin
    $display("Hello, World!");
    uut.test_idcode();
    $finish;
  end
endmodule
package dm;
  typedef struct packed {
    logic [6:0]  address;
    logic [31:0] data;
    logic [1:0]  op;
  } dmi_t;
  typedef enum logic [1:0] {
    DMINoError = 2'h0, DMIReservedError = 2'h1,
    DMIOPFailed = 2'h2, DMIBusy = 2'h3
  } dmi_error_e;
  localparam logic [3:0] DbgVersion013 = 4'h2;
  localparam logic [4:0] ProgBufSize   = 5'h8;
  localparam logic [3:0] DataCount     = 4'h2;
  localparam logic [63:0] HaltAddress = 64'h800;
  localparam logic [63:0] ResumeAddress = HaltAddress + 4;
  localparam logic [63:0] ExceptionAddress = HaltAddress + 8;
  localparam logic [11:0] DataAddr = 12'h380;  
  typedef enum logic [7:0] {
    Data0        = 8'h04,
    Data1        = 8'h05,
    Data2        = 8'h06,
    Data3        = 8'h07,
    Data4        = 8'h08,
    Data5        = 8'h09,
    Data6        = 8'h0A,
    Data7        = 8'h0B,
    Data8        = 8'h0C,
    Data9        = 8'h0D,
    Data10       = 8'h0E,
    Data11       = 8'h0F,
    DMControl    = 8'h10,
    DMStatus     = 8'h11,  
    Hartinfo     = 8'h12,
    HaltSum1     = 8'h13,
    HAWindowSel  = 8'h14,
    HAWindow     = 8'h15,
    AbstractCS   = 8'h16,
    Command      = 8'h17,
    AbstractAuto = 8'h18,
    DevTreeAddr0 = 8'h19,
    DevTreeAddr1 = 8'h1A,
    DevTreeAddr2 = 8'h1B,
    DevTreeAddr3 = 8'h1C,
    NextDM       = 8'h1D,
    ProgBuf0     = 8'h20,
    ProgBuf8     = 8'h28,  
    ProgBuf15    = 8'h2F,
[... 288 lines cut ...]
```bash
#!/bin/bash
set -euo pipefail

# Check if input file was provided
if [ $# -eq 0 ]; then
    echo "Usage: $0 <input.sv>"
    exit 1
fi
# Run verilator and capture both stdout and stderr, while allowing non-zero exit
set +e
output=$(verilator "$1" -main --timing --binary --trace-fst 2>&1)
exit_code=$?
set -e
# Check for the specific error message
if ! echo "$output" | grep -q "Expected an interface scope name to have at least one dot"; then
    exit 1
fi

# If we got here, the check passed
exit 0
``` 

</p>
</details>

Kreijstal, 2025-03-10:
ah I understand the problem it seems verilator seems to take the current path as directory...

```
kreijstal@kreijstalnuc:~/git/example$ verilator minimized/sv-bugpoint-minimized.sv -main --timing --binary --trace-fst 
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "en_GB.UTF-8",
	LC_NUMERIC = (unset),
	LC_COLLATE = (unset),
	LC_TIME = (unset),
	LC_MESSAGES = (unset),
	LC_MONETARY = (unset),
	LC_ADDRESS = (unset),
	LC_IDENTIFICATION = (unset),
	LC_MEASUREMENT = (unset),
	LC_PAPER = (unset),
	LC_TELEPHONE = (unset),
	LC_NAME = (unset),
	LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").
%Warning-INITIALDLY: jtag_master.sv:35:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'tb_hello_world.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   35 |     ir_stored_valid <= '0;
      |                     ^~
                     minimized/sv-bugpoint-minimized.sv:4:1: ... note: In file included from 'sv-bugpoint-minimized.sv'
                     ... For warning description see https://verilator.org/warn/INITIALDLY?v=5.034
                     ... Use "/* verilator lint_off INITIALDLY */" and lint_on around source to disable this message.
%Warning-INITIALDLY: jtag_master.sv:36:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'tb_hello_world.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   36 |     ir_stored       <= '0;
      |                     ^~
                     minimized/sv-bugpoint-minimized.sv:4:1: ... note: In file included from 'sv-bugpoint-minimized.sv'
%Error: Internal Error: minimized/sv-bugpoint-minimized.sv:2:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
                                                               : ... note: In instance 'jtag_master::__VDynScope_2'
    2 |   jtag_master jtag();
      |               ^~~~
kreijstal@kreijstalnuc:~/git/example$ cd minimized/ verilator minimized/sv-bugpoint-minimized.sv -main --timing --binary --trace-fst 
sh: cd: Zu viele Argumente.
kreijstal@kreijstalnuc:~/git/example$ cd minimized/
kreijstal@kreijstalnuc:~/git/example/minimized$ verilator sv-bugpoint-minimized.sv -main --timing --binary --trace-fst 
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "en_GB.UTF-8",
	LC_NUMERIC = (unset),
	LC_COLLATE = (unset),
	LC_TIME = (unset),
	LC_MESSAGES = (unset),
	LC_MONETARY = (unset),
	LC_ADDRESS = (unset),
	LC_IDENTIFICATION = (unset),
	LC_MEASUREMENT = (unset),
	LC_PAPER = (unset),
	LC_TELEPHONE = (unset),
	LC_NAME = (unset),
	LANG = "de_DE.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("de_DE.UTF-8").
%Error: sv-bugpoint-minimized.sv:2:3: Cannot find file containing module: 'jtag_master'
    2 |   jtag_master jtag();
      |   ^~~~~~~~~~~
%Error: sv-bugpoint-minimized.sv:2:3: This may be because there's no search path specified with -I<dir>.
    2 |   jtag_master jtag();
      |   ^~~~~~~~~~~
        ... Looked in:
             jtag_master
             jtag_master.v
             jtag_master.sv
             obj_dir/jtag_master
             obj_dir/jtag_master.v
             obj_dir/jtag_master.sv
%Error: Exiting due to 2 error(s)
kreijstal@kreijstalnuc:~/git/example/minimized$ 
``` 
therefore sv minizes wrongly.

wsnyder, 2025-03-30:
Were you able to reproduce this?

>Edit: Hmmm, it seems the concantenated input can't reproduce the bug for bugpoint:

I don't see how that's possible for this error. Make sure you used the obj/*__inputs* from `--debug` on the original failing test to reproduce it with sv-bugpoint.

Kreijstal, 2025-03-30:
I managed to try again in a VM I could do it.

You might like this command list I wrote, it is like a manual for lazy people it might help someone minimizing it later.

```bash
mkdir repro
cd repro/
sudo pacman -S cmake
git clone https://github.com/verilator/verilator.git
cd verilator
cmake -B build
cmake --build build
sudo cmake --build build --target install
cd ..
git clone https://github.com/antmicro/sv-bugpoint.git
cd sv-bugpoint
cmake -B build
cmake --build build
sudo cmake --build build --target install
cd ..
git clone  https://gist.github.com/4d0fed1e4e7818d9e14e73355bcb3210.git example && cd example
verilator tb_hello_world.sv -main --timing --binary --trace-fst  dm_pkg.sv #check bug exists
#%Error: Internal Error: tb_hello_world.sv:5:15: /home/arch/repro/verilator/src/V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
#                                              : ... note: In instance 'jtag_master::__VDynScope_2'
#    5 |   jtag_master jtag();
#      |               ^~~~
# bug exists!
ls
#dm_pkg.sv  jtag_master.sv  rvlab_tests.sv  tb_hello_world.sv
mkdir combined
verilator -E tb_hello_world.sv dm_pkg.sv jtag_master.sv rvlab_tests.sv > combined/combined.sv
# verilator combined.sv -main --timing --binary --trace-fst #exact same bug verified
cat > check_bug.sh <<'EOF'
#!/bin/bash

# Script to check if the specific Verilator internal error occurs.
# Takes the SystemVerilog file path as the first argument ($1).
# Exits 0 if the specific error IS found, non-zero otherwise.

# --- Configuration ---
# The exact Verilator command arguments needed to trigger the bug
VERILATOR_ARGS="-main --timing --binary --trace-fst"
# The specific error message string to look for in stderr
# Use grep -F for fixed string matching
EXPECTED_ERROR_MSG="Expected an interface scope name to have at least one dot"

# --- Execution ---
# Run Verilator, redirecting stderr to stdout so we can grep it.
# Use "$@" if verilator needs multiple files, but sv-bugpoint usually passes one.
# We use "$1" as sv-bugpoint passes the temp file path as the first arg.
output=$(verilator "$1" ${VERILATOR_ARGS} 2>&1)
verilator_exit_code=$?

# --- Check ---
# Check if Verilator exited with an error AND the specific message is present
# Using grep -Fq for Fixed string, Quiet search.
if [[ $verilator_exit_code -ne 0 ]] && echo "$output" | grep -Fq -- "$EXPECTED_ERROR_MSG"; then
  # The specific bug IS present. Exit 0 to tell sv-bugpoint to keep the change.
  exit 0
else
  # The specific bug is GONE, or a DIFFERENT error occurred, or it succeeded.
  # Exit 1 to tell sv-bugpoint to REJECT the change.
  exit 1
fi
EOF
chmod +x check_bug.sh
mkdir sv_out
sv-bugpoint sv_out ./check_bug.sh combined.sv
```

And now finally I managed to minimize it..

```systemverilog
`line 1 "tb_hello_world.sv" 1
module tb_hello_world;
  jtag_master jtag();

`line 29 "tb_hello_world.sv" 0
endmodule

`line 429 "dm_pkg.sv" 0
`line 1 "jtag_master.sv" 1

`line 4 "jtag_master.sv" 0
interface jtag_master;

`line 6 "jtag_master.sv" 0
  logic tck;
  logic tdo;

`line 121 "jtag_master.sv" 0
  task cycle_dr_dmi(output int data_o, input int data_i);
    @(posedge tck) data_o[$size(data_i)-1] <= tdo;
  endtask

`line 183 "jtag_master.sv" 0
endinterface
```

wsnyder, 2025-03-30:
Thanks for reducing the test case, straight forward with that.

Fixed.  Delays inside functions like that make classes, and classes inside interfaces had never been debugged.

I wouldn't be surprised if you find additional bugs in this area, but please file a new issue as it's unlikely the same cause, thanks.

closed 2025-03-30 by wsnyder