verilator/verilator #5847: Wrong c++ code genetaed => wrong simulation

dev-pts, 2025-03-10:
Hello!
I found a strange behaviour and, after a lot of digging, found that verilator generates unexpected c++ code.
I prepared a very short example (note, this is not a hand-written code, this is generated code from some HDL).

Verilator version (git stable branch):
```
Verilator 5.034 2025-02-24 rev v5.034-47-gac3f30ed6
```
OS: Ubuntu 22.04

SoC.v:
```
module SRAMAXI4(
	clk,
	reset,
	axi__0__r__valid,
	axi__0__aw__valid,
	axi__0__w__valid
);
	input wire clk;
	input wire reset;
	output reg axi__0__r__valid;
	input wire axi__0__aw__valid;
	input wire axi__0__w__valid;
	reg [31:0] addr [1:0];
	reg [7:0] len [1:0];
	always @(posedge clk) begin
		if (reset) begin
			axi__0__r__valid <= 0;
		end
		else begin
			if (axi__0__r__valid) begin
				addr[0] <= 32'h11111111;
				len[0] <= len[0] - 1;
			end
			if (axi__0__w__valid) begin
				addr[0] <= 32'h22222222;
			end
			if (axi__0__aw__valid) begin
				addr[0] <= 32'h33333333;
				len[0] <= 8'hff;
				if (axi__0__w__valid)
					addr[0] <= 32'h44444444;
			end
		end
	end
endmodule
```

tb.cpp:
```
#define TOP VSRAMAXI4

#include "VSRAMAXI4.h"
#include "verilated.h"
#include "verilated_vcd_c.h"
#include <iostream>

bool is_number(const std::string& s)
{
	std::string::const_iterator it = s.begin();
	while (it != s.end() && std::isdigit(*it)) ++it;
	return !s.empty() && it == s.end();
}

int main(int argc, char** argv, char** env)
{
	VerilatedContext* contextp = new VerilatedContext;
	VerilatedVcdC* tfp = new VerilatedVcdC;
	std::string cmd, name, value;

	contextp->commandArgs(argc, argv);
	contextp->traceEverOn(true);

	TOP* top = new TOP { contextp };

	top->trace(tfp, 99);
	tfp->open("simx.vcd");

#define SIG(signal) if (name == #signal) { top->signal = strtoul(value.data(), NULL, 0); }

	while (!contextp->gotFinish()) {
		std::getline(std::cin, cmd);

		if (cmd == "exit") {
			break;
		}

		if (cmd[0] == '#' || cmd.size() == 0) {
			continue;
		} else if (is_number(cmd)) {
			unsigned long cnt = strtoul(cmd.data(), NULL, 0);

			for (int i = 0; i < cnt; i++) {
				top->eval();
				tfp->dump(contextp->time());
				contextp->timeInc(1);
				top->clk ^= 1;
			}

			top->eval();
		} else {
			name = cmd.substr(0, cmd.find(' '));
			cmd.erase(0, cmd.find(' ') + 1);
			value = cmd;

			SIG(clk);
			SIG(reset);
			SIG(axi___05F0___05Fr___05Fvalid);
			SIG(axi___05F0___05Faw___05Fvalid);
			SIG(axi___05F0___05Fw___05Fvalid);
		}
	}

	tfp->close();

	delete top;
	delete contextp;
	return 0;
}
```

tb:
```
clk 0
reset 1
3
reset 0

2
axi___05F0___05Faw___05Fvalid 1
axi___05F0___05Fw___05Fvalid 1

2
axi___05F0___05Faw___05Fvalid 0
axi___05F0___05Fw___05Fvalid 1

2
axi___05F0___05Fw___05Fvalid 0

1000
exit
```

compile:
```
verilator --cc --trace --exe --build --top SRAMAXI4 tb.cpp SoC.v
```

run:
```
./obj_dir/VSRAMAXI4 < tb
```

And examine simx.vcd. I was expecting that addr[0][31:0] is would be 00000000, then 44444444, and then 22222222. But verilator generates wrong c++ code:
```
    if (__VdlySet__SRAMAXI4__DOT__addr__v1) {
        vlSelfRef.SRAMAXI4__DOT__addr[0U] = 0x22222222U;
    }
    if (__VdlySet__SRAMAXI4__DOT__addr__v3) {
        vlSelfRef.SRAMAXI4__DOT__addr[0U] = 0x44444444U;
    }
    if (__VdlySet__SRAMAXI4__DOT__len__v0) {
        vlSelfRef.SRAMAXI4__DOT__len[0U] = __VdlyVal__SRAMAXI4__DOT__len__v0;
        vlSelfRef.SRAMAXI4__DOT__addr[0U] = 0x11111111U;
    }
    if (__VdlySet__SRAMAXI4__DOT__len__v1) {
        vlSelfRef.SRAMAXI4__DOT__len[0U] = 0xffU;
        vlSelfRef.SRAMAXI4__DOT__addr[0U] = 0x33333333U;
    }
```
and I get 00000000, then 33333333, and then 22222222.

PS. Icarus verilog works correctly with this SoC.v file.

wsnyder, 2025-03-11:
@gezalore seems to have broken with V3Delayed rewrite, thanks.

dev-pts, 2025-03-11:
> [@gezalore](https://github.com/gezalore) seems to have broken with V3Delayed rewrite, thanks.

I bisected and found that the things get broken after this commit https://github.com/verilator/verilator/commit/5acced1e338619ee2a6fa285fd5ddd04bae7a6c1

gezalore, 2025-03-11:
Thanks, this should be fixed by #5848, which now generates identical code as before, but you can try that PR if you want to do a bit more testing. Nice test case, I appreciate your work on reporting this.

commit 6ecdd14f (2025-03-11): Fix NBA shared flag reuse (#5848)

Fixes #5847
closed 2025-03-11 by wsnyder