verilator/verilator #5849: Coverage breaks expressions with `this`

kbieganski, 2025-03-11:
Some expressions that contain `this` are broken with `--coverage` enabled since 34ced254c0b1ebcf996dd922266faabd26da1869 (@toddstrader).

For example, this:

```systemverilog
class Foo;
    function logic test;
        return !this;
    endfunction
endclass
```

Results in the following error:

```
Error: test.sv:3:17: Internal: Unknown node type reached emitter: THISREF
    3 |         return !this;
      |                 ^~~~
```

And this:

```systemverilog
class Foo;
    function logic test;
        return this && this;
    endfunction
endclass
```

Results in:

```
%Error: test.sv:3:16: Internal: Unknown node type reached emitter: THISREF
    3 |         return this && this;
      |                ^~~~
%Error: test.sv:3:24: Internal: Unknown node type reached emitter: THISREF
    3 |         return this && this;
      |                        ^~~~
```

It doesn't manifest for all expressions, e.g. a sole `this`, `this.x`, `f(this)` all work. Even `this + 1` is accepted (it breaks on C++ compilation :smile:). Seems like logic expressions are the problem.
 
Test case for completeness:

```systemverilog
class Foo;
    function logic test;
        return this && !this;
    endfunction
endclass

module t;
    initial begin
        Foo f = new;
        void'(f.test);
        $write("*-* All Finished *-*\n");
        $finish;
    end
endmodule
```

toddstrader, 2025-03-11:
Ah, sorry about that.  I'm guessing you don't need me to tell you, but in case others run into this as well, you can work around the problem by enabling all the non-expression coverage features like so:
` --coverage-line --coverage-toggle --coverage-user`

Expression coverage requires V3EmitV support for any expression it describes.  `this` is definitely outside my repertoire so I'm not surprised it doesn't work.  I can take a stab at fixing this next week.

wsnyder, 2025-03-11:
I suspect in V3EmitV just visit(AstThis) and print "this".

wsnyder, 2025-03-11:
I pushed the trivial V3EmitV change, but I think more is needed, when @toddstrader gets a chance.

toddstrader, 2025-03-17:
Confirmed.  I ran `test_regress` with `--coverage-expr` everywhere and found some additional problems.  Mostly stuff coverage expression shouldn't be looking at but also things V3EmitV doesn't handle.  I'll focus on the former when that's sufficient to solve things.  PRs as I've got them.

wsnyder, 2025-03-31:
@toddstrader this example passes with `--coverage`,

wsnyder, 2025-04-14:
I'd suggest excluding all out-of-bounds from coverage, that is the bounds check is ANDed with all coverage increment equations.

Ideally we'd also have an "out-of-bounds" bin.  (NOT out-of-bounds equation).

To do this you might need a new variant of CondBound that V3Unknown replaces with the appropriate equation.  Note later V3Cond will often replace bounds checks with 1'b1, so be sure to test cases where there's a e.g. 2 bit index into a 4-entry array where that happens.

closed 2025-09-06 by wsnyder