verilator/verilator #5850: Internal Error with $clog2 and types

uge, 2025-03-11:
Internal assertion failure during parsing a statement:

`    typedef logic [$clog2(NoSlvPorts)-1:0] select_t;`

`%Error: Internal Error: xxx/axi_lite_mux.sv:139:38: ../V3Ast.cpp:1111: Comparison of a node with dtypep() with a node without dtypep()`

Code can be replaced with 

`    typedef logic [(NoSlvPorts)-1:0] select_t;`

or 

```
  function integer clogb2;
      input [31:0] value;
      integer 	i;
      begin
          clogb2 = 0;
          for(i = 0; 2**i < value; i = i + 1)
      clogb2 = i + 1;
      end
  endfunction

typedef logic [$clogb2(NoSlvPorts)-1:0] select_t;
```

which allows parsing. Failing parse tree from running with --debug here:

https://drive.google.com/file/d/1gt4BtojjPrYzBtX95Reeqnf2N7iuKTgY/view?usp=sharing

----

**What 'verilator' command line do we use to run your example?**

What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.035 devel rev v5.034-45-g8157f21c3 but also failes with 5.022 (fedora-5.022)

What OS and distribution are you using?

AlmaLinux 9.4

wsnyder, 2025-03-11:
Can you expand your one-liner to a self-contained complete .v file, thanks.

uge, 2025-03-11:
> Can you expand your one-liner to a self-contained complete .v file, thanks.

I tried to isolate the failure from the large project but haven't been able to yet. I included the debug AST dump of the top level hoping that would be helpful. It's the usual problem of a build with 1000 source files. I'll keep trying.

wsnyder, 2025-03-11:
Please try sv-bugpoint it does well minimizing examples, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

uge, 2025-03-11:
Thank you so much for pointing me to sv-bugpoint. I wish I could send that tool through a time machine - it would have helped many past problems. 

Here is the output from the run and I think the root cause is a misconfigured build which causes some parameterized types to get screwed up but having the parser Error rather than crash may be a low priority improvement.

[sv-bugpoint-minimized.txt](https://github.com/user-attachments/files/19197795/sv-bugpoint-minimized.txt)

uge, 2025-03-12:
The .txt file above is actually a .sv file. For some reason github wouldn't accept a .sv attachment to the issue.

wsnyder, 2025-03-12:
I get a lot of other errors before the internal error.  Our general policy is only to fix internals if there's no prior error.  I suspect this might be a product of your sv-bugpoint reduction - can you run again please with a regexp that looks for no other errors happening?

buhraian, 2025-03-18:
I was running into similar issues with https://github.com/pulp-platform/axi/blob/69b2dc5b28379855415f4b007b624648a02431bc/src/axi_lite_mux.sv#L126, and I believe it could be that $clog2 could potentially evaluate to 0 creating the potential for a negative bitwidth? The following code snippet change seems to at least allow Verilator to progress rather than throw an Internal Error.

```
localparam SelectWidth = $clog2(NoSlvPorts) > 0 ? $clog2(NoSlvPorts) : 1;
typedef logic [SelectWidth-1:0] select_t;
```

wsnyder, 2025-03-18:
Can you attach a minimal example without that fix showing the issue please?

uge, 2025-03-18:
I tried variations to make a smaller test case but without luck yet. My observations are similar to https://github.com/verilator/verilator/issues/5850#issuecomment-2731231631 in that it seems like a missing parameter or incorrect instantiation leads to a default parameter value that creates a negative width value to clog2 but it also seems to need to be nested under conditional generation (speculation on my part).

buhraian, 2025-03-19:
Attaching a minimal example for the axi_lite_mux. I have the "fix" commented out on lines 83-84 as well. This reduction was also partly done by hand since sv-bugpoint (super cool tool) was spinning for ~8 hours without terminating, so I manually removed a lot of the macros that were not needed. 

Verilator command was: ```verilator --timing -sv --Wno-fatal --top axi_lite_mux_intf --lint-only manual-combined_axi_lite_mux.sv```

[manual-combined_axi_lite_mux.txt](https://github.com/user-attachments/files/19351160/manual-combined_axi_lite_mux.txt)

wsnyder, 2025-03-19:
Thanks, the typedef was unresolved, should be fixed now.