verilator/verilator #5858: Cannot find file on windows

tobias1012, 2025-03-15:
Running verilator through yosysHQ
for a Chisel Project

Verilator is not able to find file present in directory, error is:

```
%Error: ..\primary-sources\verification\assert\layers-MyModule-Verification-Assert.sv:2:10: Cannot find include file: 'verification\layers-MyModule-Verification.sv'
[info]     2 | `include "verification\layers-MyModule-Verification.sv"
[info]       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[info]         ... Looked in:
[info]              C\:\Users\...\AppData\Local\Temp\chiselsim-2025-03-15T11-56-30.144328200249306794953338194\primary-sources\verification\layers-MyModule-Verification.sv
[info]              C\:\Users\...\AppData\Local\Temp\chiselsim-2025-03-15T11-56-30.144328200249306794953338194\primary-sources\verification\layers-MyModule-Verification.sv.v
[info]              C\:\Users\...\AppData\Local\Temp\chiselsim-2025-03-15T11-56-30.144328200249306794953338194\primary-sources\verification\layers-MyModule-Verification.sv.sv
[info]              ..\verification\layers-MyModule-Verification.sv
[info]              ..\verification\layers-MyModule-Verification.sv.v
[info]              ..\verification\layers-MyModule-Verification.sv.sv
[info]              verification\layers-MyModule-Verification.sv
[info]              verification\layers-MyModule-Verification.sv.v
[info]              verification\layers-MyModule-Verification.sv.sv
[info]              verilated-sources\verification\layers-MyModule-Verification.sv
[info]              verilated-sources\verification\layers-MyModule-Verification.sv.v
[info]              verilated-sources\verification\layers-MyModule-Verification.sv.sv
```
I think this happens because of verilator adding a backslash in front of the colon in the windows path

**What 'verilator' command line do we use to run your example?**
```
C:\Users\...\Downloads\oss-cad-suite\bin\verilator.exe \
[info] 	'--cc' \
[info] 	'--exe' \
[info] 	'--build' \
[info] 	'-j' \
[info] 	'0' \
[info] 	'-o' \
[info] 	'../simulation' \
[info] 	'--top-module' \
[info] 	'svsimTestbench' \
[info] 	'--Mdir' \
[info] 	'verilated-sources' \
[info] 	'--assert' \
[info] 	'--debug' \
[info] 	'+incdir+C:\Users\...\AppData\Local\Temp\chiselsim-2025-03-15T11-56-30.144328200249306794953338194\primary-sources' \
[info] 	'-CFLAGS' \
[info] 	'-std=c++14 -I.. -DSVSIM_ENABLE_VERILATOR_SUPPORT' \
[info] 	'..\primary-sources\MyModule.sv' '..\primary-sources\verification\assert\layers-MyModule-Verification-Assert.sv' '..\primary-sources\verification\assume\layers-MyModule-Verification-Assume.sv' '..\primary-sources\verification\cover\layers-MyModule-Verification-Cover.sv' '..\primary-sources\verification\layers-MyModule-Verification.sv' '..\generated-sources\c-dpi-bridge.cpp' '..\generated-sources\simulation-driver.cpp' '..\generated-sources\testbench.sv'
```
**What 'verilator --version' are you using?  Did you try it with the git master version?**
Verilator 5.035 devel rev v5.034-62-g59cb53cfb
But have run multiple versions with the same error

**What OS and distribution are you using?**
Windows 10

**May we assist you in trying to fix this in Verilator yourself?**
Yes

wsnyder, 2025-03-15:
Although we do provide a CMakeFile for Windows, this isn't something we support formally as we lack someone willing to be a maintainer.  So, we're unlikely to debug/fix this, but would welcome a pull request to fix it (or a maintainer).  Please see also #5163.

tobias1012, 2025-03-15:
I'll take a look, I am unsure if it is wise to reenable windows testing for now.

tobias1012, 2025-03-15:
https://github.com/verilator/verilator/blob/59cb53cfbc18b8317e06f07b13d0f9a9821b79ba/bin/verilator#L254

error is in this function, perl escapes : which creates weird behaviour on windows. is it important for colon to be escaped?

tobias1012, 2025-03-15:
Tests pass

closed 2025-03-15 by wsnyder