verilator/verilator #5859: Missing edge event at time 0

gezalore, 2025-03-15:
The following will see 1 negedge on `reset`, instead of the expected 2. Specifically the one at time 0 is missing. I believe this is well defined and should just work, there are a lot of sloppy testbenches out there that rely on this (or worse, a x->0 negedge, but that's another matter).

```systemverilog
module top;
 
  logic reset = 1'b1;

  initial begin
    reset = 1'b0;  // No negedge here
    #10;
    reset = 1'b1;
    #10;
    reset = 1'b0;  // Negedge here
    $finish;
  end

  always @(negedge reset) $display("Got negedge at %0t", $time);

endmodule
```

wsnyder, 2025-03-15:
See `-x-initial-edge`

gezalore, 2025-03-15:
I know about that, but in the example above there is no x involved, static initializer is 1, initial block assigns 0, there is an edge. This should just work. I will look into it.

wsnyder, 2025-03-16:
I think this is a duplicate of #5499, please check and if so close as dup, thanks.

closed 2025-03-18 by gezalore