verilator/verilator #5862: Internal fault on macOS

Dragon-Git, 2025-03-16:
Can you attach an example that shows the issue?  

[process_pool_test.zip](https://github.com/user-attachments/files/19270956/process_pool_test.zip)

What 'verilator' command line do we use to run your example?

```bash
verilator --binary --top top process_pool.sv test_process_pool.sv -Wno-lint -Wno-style -Wno-SYMRSVDWORD                                
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed /opt/homebrew/Cellar/verilator/5.034/bin/verilator_bin --binary --top top process_pool.sv test_process_pool.sv -Wno-lint -Wno-style -Wno-SYMRSVDWORD
```
After add `--debug` option, the compile is success.

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.034 2025-02-24 rev UNKNOWN.REV`

What OS and distribution are you using?

macOS sequoia 15.3.2

May we assist you in trying to fix this in Verilator yourself?

lldb trace message is
```
lldb -- /opt/homebrew/Cellar/verilator/5.034/bin/verilator_bin --binary --top top process_pool.sv test_process_pool.sv -Wno-lint -Wno-style -Wno-SYMRSVDWORD
(lldb) target create "/opt/homebrew/Cellar/verilator/5.034/bin/verilator_bin"
Current executable set to '/opt/homebrew/Cellar/verilator/5.034/bin/verilator_bin' (arm64).
(lldb) settings set -- target.run-args  "--binary" "--top" "top" "process_pool.sv" "test_process_pool.sv" "-Wno-lint" "-Wno-style" "-Wno-SYMRSVDWORD"
(lldb) run
Process 43761 launched: '/opt/homebrew/Cellar/verilator/5.034/bin/verilator_bin' (arm64)
Process 43761 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2)
    frame #0: 0x0000000100049fcc verilator_bin`FileLine::ascii() const + 32
verilator_bin`FileLine::ascii:
->  0x100049fcc <+32>: ldrh   w1, [x20, #0x2]
    0x100049fd0 <+36>: add    x8, sp, #0x48
    0x100049fd4 <+40>: bl     0x1003b3a40    ; symbol stub for: FileLineSingleton::numberToName(unsigned short) const
    0x100049fd8 <+44>: adrp   x1, 1162
Target 0: (verilator_bin) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2)
  * frame #0: 0x0000000100049fcc verilator_bin`FileLine::ascii() const + 32
    frame #1: 0x00000001001507e8 verilator_bin`EmitCFunc::visit(AstComment*) + 84
    frame #2: 0x00000001000ce608 verilator_bin`AstNode::iterateAndNextConst(VNVisitorConst&) + 52
    frame #3: 0x000000010014fab4 verilator_bin`EmitCFunc::visit(AstCFunc*) + 2048
    frame #4: 0x000000010015c1b8 verilator_bin`EmitCImp::emitCFuncImp(AstNodeModule const*) + 504
    frame #5: 0x000000010015bca4 verilator_bin`EmitCImp::EmitCImp(AstNodeModule const*, bool, std::__1::deque<AstCFile*, std::__1::allocator<AstCFile*>>&) + 140
    frame #6: 0x00000001001616c4 verilator_bin`std::__1::__function::__func<V3EmitC::emitcImp()::$_1, std::__1::allocator<V3EmitC::emitcImp()::$_1>, void ()>::operator()() + 32
    frame #7: 0x000000010014dd34 verilator_bin`V3EmitC::emitcImp() + 344
    frame #8: 0x00000001000c93a4 verilator_bin`verilate(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 3876
    frame #9: 0x00000001000c7d38 verilator_bin`main + 348
    frame #10: 0x0000000190630274 dyld`start + 2840
```

wsnyder, 2025-03-16:
Unfortunately this test works fine on Linux even under the address sanitizer and valgrind.  So, we'll need you to debug this and provide a pull to fix this yourself, thanks.

I'd start by going into the debugger and seeing what the node structure is for frame 1 - it looks like a AstComment that has some unlinked FileLine, see where in the Verilog that is coming from and where it last existed before (presumably) got deleted and became dangling.

It's surprising as running with --debug (I assumed you tried given the makefile) should have asserted for dangling pointers well before the code got here.

wsnyder, 2025-03-16:
Do you have anyone at your school/work that can help?  This will need to be relatively interactive, below is getting started but it isn't realistic to debug this via github messages.

The basic idea is run with "--gdb"

Then try
"bt" -- should see backtrace like above
"p this"
"p *this"  (might core dump if so run again and skip this)
"u 1" -- up scope
"p *nodep" -- see what the node pointer is

wsnyder, 2025-03-21:
Can you attach the final output (largest numbered file) generated with `--dump-tree`.  Also the log output as above (as pointer numbers may change).

Dragon-Git, 2025-03-22:
The logs and build artifacts generated with `--dump-tree` are shown in the attachment

[log_with_ast_tree.zip](https://github.com/user-attachments/files/19401962/log_with_ast_tree.zip)

wsnyder, 2025-03-22:
I don't think it will work but please try the current master version.

If fails, can you see if you can get failure with `--debug --no-debug-leak`?

Also get a stack backtrace?

Dragon-Git, 2025-03-22:
I have tried it in the latest master version and it still failed, and I have tried `--debug --no-debug-leak` options in the above release version and in the latest master, and they both failed.If stack traceback information is needed, I'll provide it later.

Dragon-Git, 2025-03-23:
This example works correctly in the latest master version. Thank you!

closed 2025-03-23 by wsnyder