verilator/verilator #5868: Errors in Verilating RTL or wrong SystemC version?

jordankrim, 2025-03-19:
I need to find out how to correlate an error I am getting when Verilating the (RTL) code so that I can locate itto attempt to fix it or determine if there is a problem due to, for example, the wrong SystemC being used with Verilator.  I am using Verilator 5.033 and SystemC 2.3.3.

 A few of the errors are:

Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp: In function â€˜void Vhydra_tile_main___024root___nba_sequent__TOP__16(Vhydra_tile_main___024root*)â€™:
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8088:49: error: invalid operands of types â€˜unsigned intâ€™ and â€˜sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*â€™ to binary â€˜operator&â€™
     vlSelfRef.cnoc_maxi_req[0U] = ((0xfffffffcU & vlSelfRef.cnoc_maxi_req[0U])
                                     ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8142:52: error: invalid operands of types â€˜unsigned intâ€™ and â€˜sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*â€™ to binary â€˜operator&â€™
     vlSelfRef.cnoc_maxi_req[0U] = (0x74000000U | ((3U
                                                    ~~
                                                    & vlSelfRef.cnoc_maxi_req[0U])
                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8196:43: error: lvalue required as left operand of assignment
                                     << 2U));
                                           ^
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8225:60: error: lvalue required as left operand of assignment
                                                     >> 1U)));
                                                            ^
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8226:48: error: invalid operands of types â€˜unsigned intâ€™ and â€˜sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*â€™ to binary â€˜operator&â€™
     vlSelfRef.cnoc_maxi_req[3U] = (0xffffff80U & vlSelfRef.cnoc_maxi_req[3U]);
                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8227:49: error: invalid operands of types â€˜unsigned intâ€™ and â€˜sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*â€™ to binary â€˜operator&â€™
     vlSelfRef.cnoc_maxi_req[3U] = ((0xfe00007fU & vlSelfRef.cnoc_maxi_req[3U])
                                     ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8241:48: error: invalid operands of types â€˜unsigned intâ€™ and â€˜sc_core::sc_signal_inout_if<sc_dt::sc_bv<1388> >*â€™ to binary â€˜operator&â€™
     vlSelfRef.cnoc_maxi_req[3U] = ((0x1ffffffU & vlSelfRef.cnoc_maxi_req[3U])
                                     ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vhydra_tile_main___024root__DepSet_h7d236311__10.cpp:8246:56: error: lvalue required as left operand of assignment
                                               << 0x19U));
                                                        ^

wsnyder, 2025-03-19:
Use --runtime-debug and Verilator will annotate with what source lines are creating that code, so you know where to look in your sources.  Make sure that the code is correct - Verilator might not be reporting an e.g. type mismatch if you are trying to connect an array with a non-array.

jordankrim, 2025-03-19:
I assume that needs to be added to the invocation of the verilator binary?   I cannot seem to find any invocation of the verilator binary to create the verilated binary.  Is there another way to create the verilated binary other than calling the verilator binary and if so, how can I add the --runtime-debug argument?

wsnyder, 2025-03-19:
This is an argument to `verilator`.  Find whatever file is calling Verilator and add that argument please.

wsnyder, 2025-03-30:
@jordankrim were you able to make an example for this or should we close it?

jordankrim, 2025-04-01:
I cannot make an example, so close it.

closed 2025-04-01 by wsnyder