verilator/verilator #5872: Error undefined reference to `VlThreadPool::VlThreadPool(VerilatedContext*

jordankrim, 2025-03-21:
I looked at the other issues in regard to this problem, but I don't understand those possible solutions.   I do not see verilated_threads.cpp or verilated_threads.o anywhere referenced in my compile log.  In the below command I had to add -lpthread because I was getting another error (I don't recall what that was) but now I have this error:

g++ -o run_hydra_tile /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/system.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/system_driver.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/snoc_maxi_driver.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/snoc_saxi_driver.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/cnoc_driver.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/jtag_driver.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/SimJTAG.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/remote_bitbang.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/axi_master_64b.o /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../obj/axi_master_1024b.o \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_main/Vhydra_tile_main__ALL.a \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/Vhydra_tile_secondary__ALL.a \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/verilated.o \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/verilated_vcd_c.o \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/verilated_vcd_sc.o \
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_dp_ram1024/Vdp_ram1024__ALL.a /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_dp_ram64/Vdp_ram64__ALL.a /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_cxbar/Vcxbar__ALL.a /mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../sub/hydra_su/verif/tb/build/obj_dir/Vaxi_cmem__ALL.a -I//mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../include -I/mnt/data/projects/MIDS_WPE/work_jordan/systemc-2.3.3/include -L/mnt/data/projects/MIDS_WPE/work_jordan/systemc-2.3.3/lib-linux64 --std=c++14 -g -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_main -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../../model/common/include -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_dp_ram1024 -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_dp_ram64 -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_cxbar -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../sub/hydra_su/verif/tb/build/obj_dir -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_axi_master_model -I/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../sub/common/vip/systemc-axi/include -I/mnt/data/projects/MIDS_WPE/work_jordan/verilator_5.021/include -I/mnt/data/projects/MIDS_WPE/work_jordan/verilator_5.021/include/vltstd -DVCD -lm -lsystemc -lstdc++fs **-lpthread**
 -L/mnt/data/projects/MIDS_WPE/work_jordan/systemc-2.3.3/lib-linux64 
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/verilated.o: In function `VerilatedContext::threadPoolp()':
verilated.cpp:(.text+0x2084): undefined reference to `VlThreadPool::VlThreadPool(VerilatedContext*, unsigned int)'
/mnt/data/projects/MIDS_WPE/work_jordan/epu_100_test_verilator_5.021/verif/build/../../build/vobj_hydra_tile/vobj_hydra_tile_secondary/verilated.o: In function `VerilatedContext::threadPoolpOnClone()':
verilated.cpp:(.text+0x91f2): undefined reference to `VlThreadPool::VlThreadPool(VerilatedContext*, unsigned int)'

closed 2025-03-21 by wsnyder

jordankrim, 2025-03-21:
I was able to get my verilated binary to compile by adding verilated_threads.cpp and verilated_threads.o, but when I run it, I get this a strange message that seems to point to a SystemC issue and although this may not be a Verilator issue or I may not have changed the *.mk Makefiles correctly, I am reporting it here just in case:

SystemC 2.3.3-Accellera --- Jan 24 2025 08:30:53
        Copyright (c) 1996-2018 by all Contributors,
        ALL RIGHTS RESERVED
%Error: Vmain.cpp:369: Vmain::trace() is called before sc_core::sc_start(). Run sc_core::sc_start(sc_core::SC_ZERO_TIME) before trace() to complete elaboration.
Aborting...

wsnyder, 2025-03-21:
That indicates a problem in the `sc_main()` code.  As Verilator doesn't create sc_main, it's a bug in either your code or whoever wrote that code.

jordankrim, 2025-03-21:
Maybe no help, but compiling the exact same (RTL) code with Verilator 4.034/SystemC 2.3.3 works perfectly.  This reported problem occurs with Verilator 5.021/SsytemC 2.3.3.

wsnyder, 2025-03-21:
The "called before" issue or the compile error?   Assuming the former, that still likely indicates that the sc_main was always wrong.

wsnyder, 2025-03-21:
Verilator 5 has a different tracing implementation.  I could easily be wrong, welcome you to debug it and submit a pull request to fix it if I am!