verilator/verilator #5873: Bad VCD output for single bit newval.

Rki009, 2025-03-21:
https://github.com/verilator/verilator/blob/8c287134d7b9a3b8f9d913ccbfb5b4498b5ab7a4/include/verilated_vcd_c.cpp#L603

 *wp++ = '0' | static_cast<char>(newval);

Somewhere newval is set to to something other than 0x00 or 0x01 so the resultant character VCD is 0xFE in the simulation I was doing. The VCD file is trashed and can not be read by gtkwave. Initially I wrote a python script to patch the VCD file by replacing the 0xFE with '0' and all works.

The above code should be updated to:

*wp++ = '0' | (static_cast<char>(newval)&0x01);

Only '0' and '1' are allowed for single bit value. The real problem is probably further up the code. The newval should always be 0x00 or 0x01.

wsnyder, 2025-03-21:
That upper bits are set means that something upstream is corrupted (as you suspected), and there's a high risk of other problems like math errors.  Can you please attach an openly licensed, completely self-contained test case so we can directly run what you provide.  Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs and especially use of sv-bugpoint, you can have it search for the bad VCD 0xFE.

wsnyder, 2025-03-29:
P.S. the most common cause is if you're writing your own main() and passing inputs with upper bits set.  Compile with VL_DEBUG and run with VerilatedContext::debug() set to check.

closed 2025-04-03 by wsnyder