verilator/verilator #5874: Error during "make test": /usr/bin/ld: cannot find -lstdc++

jordankrim, 2025-03-21:
After compiilng Verilator, I attempted to do a "make test" and got this:

-- COMPILE entire design --------------------------------------
make -j 4 -C obj_dir -f Vtop.mk
make[2]: Entering directory '$VERILATOR_ROOT/examples/make_protect_lib/obj_dir'
g++    sim_main.o verilated.o verilated_dpi.o verilated_vcd_c.o Vtop__ALL.a   -L../obj_dir_secret -lverilated_secret -static  -o Vtop  
/usr/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
make[2]: *** [Vtop.mk:64: Vtop] Error 1
make[2]: Leaving directory '$VERILATOR_ROOT/examples/make_protect_lib/obj_dir'
make[1]: *** [Makefile:73: run] Error 2
make[1]: Leaving directory '$VERILATOR_ROOT/examples/make_protect_lib'
make: *** [Makefile:252: examples] Error 10

This system does not have ccache on it.  Is that needed or is it something else?

wsnyder, 2025-03-21:
Attach the log from ./configure.  I don't see how that could have passed with that problem.  Also it detects ccache, so perhaps you didn't ./configure on the same system?

wsnyder, 2025-03-21:
Closing as it's something wrong at your system, but feel free to follow up here.

closed 2025-03-21 by wsnyder

jordankrim, 2025-03-21:
Here is the config.log attached.

[config.log.txt](https://github.com/user-attachments/files/19398050/config.log.txt)

wsnyder, 2025-03-21:
I don't see anything obviously wrong, but note that C++ program did compile, which based on your message above it should have not.  You'll need to debug yourself, especially since you are using a 5-year old Verilator on a 10-year old OS.