verilator/verilator #5875: Assignment to the multidriven input

L3odr0id, 2025-03-22:
Compilation of the following module results in an error: `%Error-ASSIGNIN: top.sv:2:12: Assigning to input/const variable: 'i1'`
```SystemVerilog
module a(input wire i1);
    assign i1 = 'b1;
endmodule: a
```

Options: `--cc --exe testbench.cpp --timing -Wno-fatal +1800-2023ext+sv`

### Expected Behavior
SystemVerilog allows assigns to multidriven types, so this assign should be valid.

This issue occurs in `Verilator 5.030 2024-10-27 rev v5.030-45-g2cb1a8de7`

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

gezalore, 2025-03-22:
This can be turned off with `-Wno-ASSIGNIN` and works largely as expected.

L3odr0id, 2025-03-22:
Oh, thanks!

wsnyder, 2025-03-22:
Also, in most all cases such code represents a mistake, so despite it being technically legal, I believe it should remain an error to guide common good practices.  So nfix.

closed 2025-03-22 by wsnyder