verilator/verilator #5877: Fails to connect unpacked real arrays with opposite index ranges

L3odr0id, 2025-03-23:
Compilation of the following module results in an error: `%Error: Input port connection 'i1' expected non-complex non-double pin connection in width check`
```SystemVerilog
module a(input real i1 [0:4]);
endmodule: a

module b(input real i2 [4:0]);
  a a_inst(.i1(i2));
endmodule: b
```

The module compiles successfully when both arrays have identical index ranges (e.g., [4:0] in both cases), but fails when one has [0:4] and the other [4:0].

Options: `--cc --exe testbench.cpp --timing -Wno-fatal +1800-2023ext+sv`

### Expected Behavior
`i1` and `i2` are both arrays with the same number of elements, so their connection should be valid regardless of index order

This issue occurs in `Verilator 5.030 2024-10-27 rev v5.030-45-g2cb1a8de7`

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

wsnyder, 2025-03-23:
Verilator doesn't support this sort of slice conversion properly, as it requires swapping every index to accomplish the endian conversion.   If you'd like this fixed in a short time please consider a pull request, thanks.

BTW if you're running some sort of test suite to find these, we always welcome a pull that reports an issue and provides a fix.  We'd prefer however not filing a bunch of issues on each little test suite failure, however, as prefer not to water down those issues that are blocking real user designs, thanks.

paul-demo, 2025-06-11:
Connecting [4:0] unpacked to [0:4] unpacked is highly discouraged because lots of simulators and every engineer would has issues with getting this right.

This is one of the reasons that most/every coding guideline require that packed indexing be declared [hi:low] and unpacked be declared [lo:hi] or just [N] which is implicitly [lo:hi]. I distinctly recall listening to a presentation by some big silicon company (maybe Marvell?) about costly bugs related to mixing [lo:hi] and [hi:lo] in unpacked dimensions.