verilator/verilator #5879: force/release for the hier signal

panicmarvin, 2025-03-24:
Can you attach an example that shows the issue? 
If the COND_0 macro is defined, after release, inst1.i cannot be released;   however, in another branch, releasing signal_0 works as expected. I have already used assign to drive signal_0.
```
`timescale 1ns/1ps
`define COND_0

module inv(
    output reg o,
    input i);
    assign o = ~i;
endmodule

module dff(
    output reg o,
    input i, clk);
    always_ff@(posedge clk)begin
	    o <= i;
    end
endmodule

module top;
wire signal_1 = 1'b1;
wire signal_0;

reg clk;
wire out;

inv inst0(.i(signal_1), .o(signal_0));
dff inst1(.clk(clk), .i(signal_0), .o(out));

initial begin
`ifdef COND_0
    #1;
    force inst1.i = 1;
    #15;
    release inst1.i;

`else

    #1;
    force signal_0 = 1;
    #15;
    release signal_0;
`endif

    #50;
    $finish();
end

initial begin
    #0;
    clk = 1;
    forever #5 clk = ~clk;
end

endmodule
```

What 'verilator' command line do we use to run your example?
verilator --flatten --cc --top top --timing top.sv

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.035 devel master(20250324)

What OS and distribution are you using?
rockylinux
May we assist you in trying to fix this in Verilator yourself?
no

panicmarvin, 2025-03-24:
[image]

panicmarvin, 2025-03-24:
Based on the VCS simulation results, after performing a force and then a release on the instantiated port connections (e.g., top-level wires) and the input ports of the next-level modules (hierarchical wires), the values are not affected by the previous assignments defined within the module (continuous assignment, active assign in procedural continuous assignment, or procedural assignment). Instead, they directly revert to their driven values.

wsnyder, 2025-03-24:
Can you please make the test self-checking, that is check values and $stop on COND_0 and $finish otherwise?

panicmarvin, 2025-03-24:
```
`timescale 1ns/1ps

module inv(
    output o,
    input i);
    assign o = ~i;
endmodule

module dff(
    output reg o,
    input i, clk);
    always_ff@(posedge clk)begin
	    o <= i;
    end
endmodule

module top;
wire signal_1 = 1'b1;
wire signal_0;

reg clk;
wire out;

inv inst0(.i(signal_1), .o(signal_0));
dff inst1(.clk(clk), .i(signal_0), .o(out));

initial begin
`ifdef COND_0
    #1;
    force inst1.i = 1;
    $display("[%s:%0d] After force: inst1.i=%b", `__FILE__, `__LINE__, inst1.i);
    
    #15;
    release inst1.i;
    
    // Immediately assert the value is 0 after release
    if (inst1.i !== 1'b0) begin
        $error("[%s:%0d] ASSERTION FAILED: inst1.i value after release is %b, expected 0", `__FILE__, `__LINE__, inst1.i);
        $stop;  // Stop simulation
    end else begin
        $display("[%s:%0d] ASSERTION PASSED: inst1.i value after release is 0", `__FILE__, `__LINE__);
    end

`else
    
    #1;
    force signal_0 = 1;
    
    #15;
    release signal_0;
    
    // Immediately assert the value is 0 after release
    if (signal_0 !== 1'b0) begin
        $error("[%s:%0d] ASSERTION FAILED: signal_0 value after release is %b, expected 0", `__FILE__, `__LINE__, signal_0);
        $stop;  // Stop simulation
    end else begin
        $display("[%s:%0d] ASSERTION PASSED: signal_0 value after release is 0", `__FILE__, `__LINE__);
    end
`endif

    #50;
    $finish();
end

initial begin
    #0;
    clk = 1;
    forever #5 clk = ~clk;
end

endmodule
```
Command line:
fail:  verilator --flatten --cc --top top --binary --timing --trace top.sv -DCOND_0
pass:  verilator --flatten --cc --top top --binary --timing --trace top.sv 

Meanwhile, vcs always passed:
vcs -sverilog -full64 top.sv -R +define+COND_0
vcs -sverilog -full64 top.sv -R

wsnyder, 2025-03-24:
@b-chmiel is this something you might be able to look into?  Maybe it's related to the way we optimize some direct-connected instances (pinReconnectSimple)?

ArturBieniek4, 2026-09-02:
It works now, fixed in https://github.com/verilator/verilator/pull/7907.

closed 2026-09-02 by ArturBieniek4