verilator/verilator #5886: Error attempting to Verilate code

jordankrim, 2025-03-26:
I have some code I am attempting to Verilate using Verilator 4.228.  I get no error message and the same code verilates fine with 4.034.  Here is the error message from the 4.228 attempt:

%Error: Internal Error: /project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../include/../include/xyz_pkg.sv:8:16: ../V3Ast.h:1976: AstNode is not of expected type, but instead has type 'VAR'
-node: VAR 0xf86340 <e2591> {n8aq} u1=0x8cb74a0 @dt=0@  QQD8DBiN5S [VSTATIC]  LPARAM
    8 |     localparam QQD8DBiN5S  = 8;
      |                ^~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1120:     Dumping /project/work_jordan/abc_100/build/../build/vobj_xyz_def/vobj_xyz_def_main/Vxyz_def_main_990_final.tree
- V3Ast.cpp:1120:     Dumping /project/work_jordan/abc_100/build/../build/vobj_xyz_def/vobj_xyz_def_main/Vxyz_def_main_992_linkdot-preerr.tree
- V3StatsReport.cpp:223:statsReport: 
%Error: Internal Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
0x00007ffff710e52f in raise () from /lib64/libc.so.6
#0  0x00007ffff710e52f in raise () from /lib64/libc.so.6
#1  0x00007ffff70e1e65 in abort () from /lib64/libc.so.6
#2  0x00000000006803a1 in V3Error::vlAbort () at ../V3Error.cpp:172
#3  0x0000000000680a8a in V3Error::vlAbortOrExit () at ../V3Error.cpp:164
#4  0x0000000000682ad4 in V3Error::v3errorEnd (sstr=..., extra="") at ../V3Error.cpp:290
#5  0x000000000069f84f in FileLine::v3errorEnd (this=this@entry=0xf85f40, sstr=..., extra="") at /usr/include/c++/8/bits/basic_string.h:1368
#6  0x0000000000432344 in AstNode::v3errorEnd (this=this@entry=0xf86340, str=...) at /usr/include/c++/8/bits/basic_string.h:256
#7  0x0000000000432451 in AstNode::v3errorEndFatal (this=this@entry=0xf86340, str=...) at ../V3Ast.cpp:1145
#8  0x000000000074d855 in AstNode::privateAs<AstEnumItem, AstNode*> (nodep=0xf86340) at ../V3Error.h:313
#9  LinkDotFindVisitor::visit (this=0x7fffffffc1f0, nodep=0x2100d40) at ../V3LinkDot.cpp:1210
#10 0x0000000000779e7c in AstEnumItem::accept (this=<optimized out>, v=...) at ../V3AstNodeDType.h:273
#11 0x00000000004372d1 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:864
#12 0x0000000000437386 in AstNode::iterateChildren (this=0x2101a50, v=...) at ../V3Ast.cpp:828
#13 0x000000000073647e in VNVisitor::iterateChildren (nodep=<optimized out>, this=<optimized out>) at ../V3Ast.h:2393
#14 LinkDotFindVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ../V3LinkDot.cpp:1376
#15 0x000000000042dc6a in VNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:237
#16 0x000000000042d3df in VNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:114
#17 0x00000000009dfb14 in AstEnumDType::accept (this=<optimized out>, v=...) at ../V3AstNodeDType.h:715
#18 0x00000000004372d1 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:864
#19 0x0000000000437375 in AstNode::iterateChildren (this=this@entry=0x20fa340, v=...) at ../V3Ast.cpp:827
#20 0x000000000074d226 in VNVisitor::iterateChildren (nodep=0x20fa340, this=0x7fffffffc1f0) at ../V3Ast.h:2393
#21 LinkDotFindVisitor::visit (this=0x7fffffffc1f0, nodep=0x20fa340) at ../V3LinkDot.cpp:1178
#22 0x00000000004ad1ce in AstTypedef::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:1891
#23 0x00000000004372d1 in AstNode::iterateAndNext (this=<optimized out>, v=...) at ../V3Ast.cpp:864
#24 0x0000000000437386 in AstNode::iterateChildren (this=this@entry=0x20fbb60, v=...) at ../V3Ast.cpp:828
#25 0x0000000000754e45 in VNVisitor::iterateChildren (nodep=0x20fbb60, this=0x7fffffffc1f0) at ../V3Ast.h:2393
#26 LinkDotFindVisitor::visit (this=0x7fffffffc1f0, nodep=0x20fbb60) at ../V3LinkDot.cpp:821
#27 0x000000000042de99 in VNVisitor::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:269
#28 0x00000000004ad322 in AstPackage::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:2549
#29 0x000000000042ed7f in AstNode::iterateListBackwards (this=<optimized out>, v=...) at ../V3Ast.cpp:884
#30 0x000000000042edaa in AstNode::iterateChildrenBackwards (this=this@entry=0xeff3a0, v=...) at ../V3Ast.cpp:894
#31 0x000000000075b145 in VNVisitor::iterateChildrenBackwards (nodep=0xeff3a0, this=0x7fffffffc1f0) at ../V3Ast.h:2395
#32 LinkDotFindVisitor::visit (this=0x7fffffffc1f0, nodep=0xeff3a0) at ../V3LinkDot.cpp:756
#33 0x00000000004ad070 in AstNetlist::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:1417
#34 0x0000000000734f0e in VNVisitor::iterate (nodep=0xeff3a0, this=0x7fffffffc1f0) at ../V3Ast.h:2389
#35 LinkDotFindVisitor::LinkDotFindVisitor (statep=0x7fffffffc2a0, rootp=0xeff3a0, this=0x7fffffffc1f0) at ../V3LinkDot.cpp:1384
#36 V3LinkDot::linkDotGuts (rootp=rootp@entry=0xeff3a0, step=step@entry=LDS_PRIMARY) at ../V3LinkDot.cpp:3175
#37 0x0000000000735d40 in V3LinkDot::linkDotPrimary (nodep=0xeff3a0) at ../V3LinkDot.cpp:3207
#38 0x0000000000406bc5 in process () at ../V3Global.h:127
#39 0x0000000000407e8c in verilate (argString="vconfig_def.vlt --output-split 80000 --output-split-cfuncs 20000 --debug --gdbbt --trace --trace-depth 99 --trace-structs -sc -I/project/work_jordan/abc_100/verif/build_with_verila"...) at ../Verilator.cpp:608
#40 0x000000000040a68a in main (argc=40, argv=<optimized out>) at ../Verilator.cpp:730
A debugging session is active.

	Inferior 1 [process 139551] will be killed.

Quit anyway? (y or n) [answered Y; input not from terminal]
%Error: export VERILATOR_ROOT=/project/work_jordan/verilator-4.228
%Error: gdb /project/work_jordan/verilator-4.228/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run vconfig_def.vlt --output-split 80000 --output-split-cfuncs 20000 --debug --gdbbt --trace --trace-depth 99 --trace-structs -sc -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../include -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../sub/axi/include -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../sub/xyz_su/include --top-module xyz_def_main --Mdir /project/work_jordan/abc_100/build/../build/vobj_xyz_def/vobj_xyz_def_main -f flists/flist_common.rtl -f flists/flist_spe.rtl -f flists/flist_vperm.rtl -f flists/flist_fpu.rtl -f flists/flist_su.rtl -f flists/flist_tmem.rtl -f flists/flist_l2_cache.rtl -f flists/flist_behav.wrappers -f flists/flist_behav.tech -f flists/flist_def.rtl" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed gdb /project/work_jordan/verilator-4.228/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run vconfig_def.vlt --output-split 80000 --output-split-cfuncs 20000 --debug --gdbbt --trace --trace-depth 99 --trace-structs -sc -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../include -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../sub/axi/include -I/project/work_jordan/abc_100/verif/build_with_verilator_4.228/../../sub/xyz_su/include --top-module xyz_def_main --Mdir /project/work_jordan/abc_100/build/../build/vobj_xyz_def/vobj_xyz_def_main -f flists/flist_common.rtl -f flists/flist_spe.rtl -f flists/flist_vperm.rtl -f flists/flist_fpu.rtl -f flists/flist_su.rtl -f flists/flist_tmem.rtl -f flists/flist_l2_cache.rtl -f flists/flist_behav.wrappers -f flists/flist_behav.tech -f flists/flist_def.rtl" -ex 'set width 0' -ex 'bt' -ex 'quit'
make[3]: *** [Makefile_def:87: build_v_main] Error 255
make[3]: Leaving directory '/project/work_jordan/abc_100/build'
make[2]: *** [Makefile:10: def] Error 2
make[2]: Leaving directory '/project/work_jordan/abc_100/build'

wsnyder, 2025-03-26:
Looks like something related to that parameter being used in an enum.

Please attach an openly licensed, completely self-contained test case so we can directly run what you provide.  Ideally use test_regress format, see [Reporting Verilator Bugs](https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs) and especially section on use of sv-bugpoint.

wsnyder, 2025-05-18:
Closing - reopen if provide example.

closed 2025-05-18 by wsnyder