verilator/verilator #5890: Problem with parametrized parameter sizes

alexforencich, 2025-03-27:
With verilator 5.0.34, I am getting an error with the following test code:

    module test #(
        parameter LEN = 4,
        parameter LST[LEN] = '{LEN{0}}
    )();

    endmodule

    module top();

    parameter LEN = 8;

    test #(
        .LEN(LEN),
        .LST('{LEN{0}})
    )
    inst();

    endmodule

The error is:

%Error: bug.sv:15:10: Assignment pattern with too many elements
                    : ... note: In instance 'top'
   15 |     .LST('{LEN{0}})
      |          ^~
%Error: Exiting due to 1 error(s)

The error does not appear if I do not change the value of the LEN parameter.

wsnyder, 2025-03-28:
It does seem to think the pattern is an array[4] while it's making 8 pattern elements.  Probably related to other bugs with the order of module expansion, but I'm not sure yet.  If you'd like to dig in to what's specifically happening it would be appreciated.

godblesszhouzhou, 2025-04-14:
I have done some debug about that. The problem should be in V3Param, when the 'deepCloneModule' is done. The inst parameter for the range of the array is not considered(still by default).

jeras, 2025-04-21:
I would just like to mention a workaround. I never noticed this issue, since I have not used _array literals_ with replication yet, instead I would write `LST[LEN] = '{default: 0}`.

godblesszhouzhou, 2025-04-28:
I have done some debug further, a simplified example should be

```
module test #(
    parameter LEN = 4,
    parameter LST[LEN] = '{LEN{0}}
)();
endmodule

module top();
parameter LEN = 8;
test #(
    .LEN(LEN),
    .LST('{LEN{0}})
)
inst1();
endmodule
```

First, in V3LinkDot, the VarRef LEN (both in AstPin LEN and AstPin LST) will be linked to variable LEN = 8 in the top module.
Then, in V3Param:
1. Run nodeDeparam() and use constifyParamsEdit() to constify the parameter, in order to constify the parameter
   1) Use V3Width to constify the AstBracketArrayDType of (LST[LEN]), the problem is here.  As the AstPin LST corresponds to the Variable LST (there is not a new parameter variable LST ) in the original module test, the value of LEN is 4, then the corresponding generated AstUnpackArrayDType with the range [0:3]. 
   2) Run visitPatMemberRep(), AstPattern under AstPin LST is constified, the number of AstPattern is 8 (as the VarRef LEN is linked to variable LEN = 8 in the top module), this is right. But as the AstPin LST corresponds to the Variable LST in the original module test with the range[0:3], there will be an error.
In summary, the constifying of the parameter is right, the problem is the AstPin LST links the original variable LST[LEN] with LEN = 4. (In fact the AstPin LEN also links the original varible LEN, it is ok. But for AstPin LST, the constifying of parameter need the AstUnpackArrayDType of LST to be updated to the new variable LEN = 8.).

One solution may be in the V3LinkDot, first we need to find the AstPin like this (there is not a new parameter variable, but there is a new parameter for the AstBracketArrayDType), then we build a new parameter variable for this AstPin. In this way, for the case above, it will be transferred to:

```
module top();
parameter LEN = 8;
**parameter LST[LEN] = '{LEN{0}}; // The new parameter variable.**
test #(
    .LEN(LEN),
    **.LST(LST)**
)
inst1();
endmodule
```
 
Then, the following compilation will be ok.

wsnyder, 2025-04-28:
Thanks for the analysis, this then is basically #4920; a non-parameter module needs to be kept around to avoid LEN being miscomputed.

godblesszhouzhou, 2025-04-29:
I think this case is diffenrent from [4920](https://github.com/verilator/verilator/issues/4920).  For the function nodeDeparam():
```
    void nodeDeparam(AstNode* nodep, AstNodeModule*& srcModpr, AstNodeModule* modp,
                     const string& someInstanceName) {
        m_modp = modp;
        **V3Const::constifyParamsEdit(nodep);**
        // Set name for warnings for when we param propagate the module
        const string instanceName = someInstanceName + "." + nodep->name();
        srcModpr->someInstanceName(instanceName);

        if (auto* cellp = VN_CAST(nodep, Cell)) {
            **cellDeparam(cellp, srcModpr);**
        } else if (auto* ifaceRefDTypep = VN_CAST(nodep, IfaceRefDType)) {
            ifaceRefDeparam(ifaceRefDTypep, srcModpr);
        } else if (auto* classRefp = VN_CAST(nodep, ClassRefDType)) {
            classRefDeparam(classRefp, srcModpr);
        } else if (auto* classRefp = VN_CAST(nodep, ClassOrPackageRef)) {
            classRefDeparam(classRefp, srcModpr);
        } else {
            nodep->v3fatalSrc("Expected module parameterization");
        }
    }
```

For this case, the problem is in **V3Const::constifyParamsEdit(nodep);** Constifying parameter LST has a problem as showed above.  In fact, in cellDeparam, deepCloneModule will be called and a new module with wrong parameter will be built for cell  inst1. Namely, step1 constify the cell parameter, step2 clone a new module for the cell, step3 update the parameter Var of the new module according to the constified cell parameter. As there is not a new constify parameter variable in top module, LST is constified in step1.
While for  [4920](https://github.com/verilator/verilator/issues/4920), the problem is in **cellDeparam(deepCloneModule)**, for two cell bc, two corresponding module add2 should be cloned, then update the constified parameter.

wsnyder, 2025-04-29:
Point taken. The port connection depends on a correct data type of the child module, and that data type incorrectly (or more accurately not yet correctly) has a length. 

I don't think making a temporary local parameter will help, as to do that we still need a correct data type to make it, which has to come from the child, and so requires fixing the issue anyways plus additional work.  Plus problems if the type requires localparams not even visible to the parent.

 It still seems the right thing to do is related to fixing #4920, namely clone first, then determine a pin parameter setting order, and set parameters in the clone one-by-one. Then when the pin resolves with the pattern the dtype will be correct.

wsnyder, 2025-04-29:
BTW a complication with what I said above is we'd need to clone a module, determine parameters, and only then would know that the deparam'ed module is a duplicate. At that point we'd need to link up pins again.

godblesszhouzhou, 2025-04-30:
Agree.  I will try to do some work further to resolve this issue.

alexforencich, 2025-11-09:
This is still a problem with 5.0.40.

godblesszhouzhou, 2025-12-19:
I have done some debug further and try another method to resolve this issue. Since, the AstPin LST corresponds to the Variable LST (there is not a new parameter variable LST ) in the original module test, a simple method maybe after transferring BracketArrayDType  to PackedArrayDType, clone the PackedArrayDType and  repalce the ref of Variable LST in the module test with the new parameter variable LST ref, then set the type range. By do some simple tests, the issues can be resolved and also some more complicated cases can also be compiled correctly like:
```
module test #(
    parameter LEN = 4,
    parameter LST[LEN] = '{LEN{0}}
)();
endmodule

module top();
parameter LEN1 = 8;
test #(
    .LEN(LEN1),
    .LST('{LEN1{0}})
)
inst1();
parameter LEN2 = 8;
test #(
    .LEN(LEN2*2),
    .LST('{2*LEN2{0}})
)
inst2();
parameter LEN3 = 3;
parameter LEN4 = 5;
test #(
    .LEN(LEN3 + LEN4),
    .LST('{{LEN2 + Len4}{0}})
)
inst2();
endmodule
```
I will open a PR in the next few days.

taylor-bsg, 2026-10-10:
This works on current master (c265bd444). The original example now builds, and both `inst1` and `inst2` get an 8-entry `LST`.

It was fixed by @em2machine's #8562, merged on Oct 6: just before it, the same example still fails with "Assignment pattern with too many elements", and with it, the example works. So I think this issue can be closed.

closed 2026-10-10 by wsnyder