verilator/verilator #5891: verilating with mixed Verilated / SV

dpetrisko, 2025-03-27:
Hello!

This is not a problem with Verilator per se. But I'm trying to work around a fairly common use-case supported by other tools.  I'm running into many different problems. I'm hoping someone can point me to the right way to do this and then I can debug offline. 

- I have a testbench written in SystemVerilog with several submodules
- I have a dut also written in SystemVerilog with several submodules
- some of these submodules are shared between them. 
- however, the submodules may be versioned differently between the dut and testbench; different `defines, applied patches, etc.

With other simulators, I can specify which library to pick from using config.v files such as:

        config config_vlog;
            design testbench;
            instance testbench.dut liblist vloglib;
        endconfig

        config config_syn;
            design testbench;
            instance testbench.dut liblist pdklib synlib;
            // Use version with nonsynth delay
            cell bsg_rp_clk_gen_osc_v3_row use pdklib.bsg_rp_clk_gen_osc_v3_row;
        endconfig
 

I understand that Verilator does not / will not support config.v files:
https://github.com/verilator/verilator/blob/master/docs/internals.rst#never-features

> IEEE 1800-2023 33 Config
> Little industry use.

I don't need super fine-grained support, so my workaround here is to try to do multiple verilations and then glue them together. For example's sake:

        [testbench.sv]
        /---------------------------------------------------\
        | testbench                                         |
        |   [driver.sv]                 [dut.sv]            |
        |      /---------------\          /---------------\ |
        |      | driver        |          | dut           | |
        |      |   [sub_v1.sv] |          |  [sub_v2.sv]  | |
        |      |     /-------\ |  <---->  |     /-------\ | |
        |      |     |  sub  | |          |     |  sub  | | |
        |      |     \-------/ |          |     \-------/ | |
        |      \---------------/          \---------------/ |
        \---------------------------------------------------/

I believe this could be supported by --lib-create / hierarchical Verilation. However, I would prefer to avoid this as based on (https://veripool.org/guide/latest/exe_verilator.html)
1) --timing is not supported
2) Signals in the block cannot be traced

I would like to do something like (pseudo-commands)

        verilate dut.sv sub_v2.sv -M obj_dir-dut
        verilate driver.sv sub_v1.sv -M obj_dir-driver
        verilate testbench.sv obj_dir-dut/*.a obj_dir-driver/*.a

What I've tried:
- this approach
  - can't get the linking to work properly during verilation, compilation errors
- verilating dut first, then verilating testbench.sv + driver.sv
  - links sub_v1 instead
- blackboxing sub_v2 and then separately verilating testbench.sv
  - could not get verilation to work for this either
- using --pipe-filter to prefix all modules in DUT
  - Mostly works but is not ideal for readability and debug. If all else fails, I'll probably pursue this more

So main questions:
- Has anyone solved this in a different way? A Verilator-specific compilation flow is completely fine but I would prefer to avoid any SV/C++ modifications incompatible with commercial tools.
- Is there anything missing in Verilator that fundamentally makes some version of libraries hard? I'm willing to work on a small port of this as it will be a fairly substantial remaining gap after other, harder, GLS support (UDP, SDF annotation) is merged.
- Is there a smaller subset of Verilator support (like individual module->file mapping) that would make this easier to implement?
- Any chance to reconsider basic implementation of IEEE 1800-2023.33 (a man can dream ;))

P.S. Is there a better place  (slack/element etc) to post long-winded questions? The Verilator "forum" in the README.md linked me here

Thank you!

wsnyder, 2025-03-27:
There's no current clean way to do this.

One perhaps easy but ugly hack is to write a pipe-filter that when parsing "driver" replaces "sub" with "sub_1", and when parsing "dut" replaces "sub" with "sub_2".  If "dut" is completely a unique name, this can be only a few lines of e.g. python to do a search and replace.

As to supporting config, when you say "reconsider" perhaps that is because this was once on the 
never" list.  Note it no longer is, as config is making a comeback for chiplets etc, so we're receptive to a pull.  I have config maybe 10% started in a very stale branch not being worked on, if you or someone wants to work on I can pass that across.  I'd certainly accept a pull to do even a portion of it, but it's obviously a non-trivial time investment for you/someone to go down.

dpetrisko, 2025-03-27:
>As to supporting config, when you say "reconsider" perhaps that is because this was once on the
never" list. Note it no longer is,

That's awesome! Yes I will see if I can break off a small chunk to get started with that

wsnyder, 2025-03-27:
Attached is the patch to get started.  This applies on the current master 8965401d10e3b78b8b0b823a35304cf690b01b7e

[config.patch.txt](https://github.com/user-attachments/files/19495769/config.patch.txt)

I wouldn't call this even 10%

What would first be appreciated is a pull that simply has test cases that work against other simulators (started in this patch).  On Verilator it would just test for all of the unsupported errors.

The next pull could implement the parsing, also started in this patch, but just throw unsupported errors on each individual config construct, similar to how we do with other items like many of the complicated assertion constructs.

Then later pulls can implement the individual features.

wsnyder, 2025-11-20:
It's not using config at all, it simply assumes whatever library this module was in (based on --work setting the libname()) is where to look now.

dpetrisko, 2025-11-20:
```
config cfg1;
    design t;
    default liblist liba libb ;
endconfig

...

    3: CONFIG 0xbaac649a0 <e966#> {h14ab}  libb.cfg1 configname=cfg1 libname=libb
    3:1: CONFIGCELL 0x1010b6340 <e960#> {h15am}  libb.t
    3:2: CONFIGRULE 0xbaac61bc0 <e964#> {h16af}
    3:2:2: PARSEREF 0xbaac5e580 <e961#> {h16av} @dt=0x0@  liba
    3:2:2: PARSEREF 0xbaac5e640 <e963#> {h16ba} @dt=0x0@  libb

...

%Error: Internal Error: t/t_config_default.v:16:21: ../V3LinkDot.cpp:3417: nullptr lookup symbol table
-node: PARSEREF 0xbaac5e580 <e961> {h16av} u3=0x1 @dt=0x0@  liba
   16 |     default liblist liba libb ;
      |                     ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html?v=5.043 for more assistance.
- V3Ast.cpp:1391:     Dumping obj_vlt/t_config_default/Vt_config_default_990_final.tree
- V3Ast.cpp:1391:     Dumping obj_vlt/t_config_default/Vt_config_default_992_linkdot-preerr.tree
```

Since V3LinkDot has a ParseRef visitor but the liblist clause isn't actually part of the global structure, it seems to be having issues. I think we're only using the "default liblist" rules during V3LinkCells, so we could remove them from the Ast after visiting the config rule, or we could replace them with a different type? What would you recommend?

wsnyder, 2025-11-24:
Currently V3LinkCells makes a graph with vertexes as the module names, and edges representing dependencies.  Edges are of many times and indistinguishable, but you could use a custom weight for the AstCell-made edges, so you know what to follow to walk the graph to find the hierarchy.  You probably need cell names, you could add those to a new custom edge class.  Or you could make a new vertex, so it goes upper_module_vertex -> cell_vertex -> lower_module_vertex, but that will result in more space and changes so probably edges is easiest.

Another choice is to make a symbol-table like in V3LinkDot, but this will be huge if you expand it.  I think using the existing graph is likely sufficient.

dpetrisko, 2025-12-06:
Okay, I think with #6745 and #6764 this should be sufficient support for my use case (gate-level sims). I'm going to context switch to trying that out. Of course, let me know of any fixups needed in the other PRs. Thanks for the help!

closed 2025-12-16 by wsnyder