verilator/verilator #5892: Optimizations generate code that depends on assignment location

dev-pts, 2025-03-27:
Hello!
... and the code either works as in Icarus Verilog, or not.
Perhaps, this is rare case. But I'm using HDL which generates such Verilog code and now I'm blocked by verilator :(

Verilator 5.035 devel rev v5.034-88-g8965401d1

Consider this example:
```
$ cat a.v
module test;
	initial begin
		a = 0;
	end
	reg a;
	always @(*) begin
		a <= 0;
		if (a == 0) begin
			$display("a == 0");
		end
		if (a == 1) begin
			$display("a == 1");
		end
		a <= 1;
	end
endmodule
```

```
verilator -Wno-fatal --cc --top test a.v
```

It seems, verilator sees "a <= 0" and optimizes everything out and generates this code (which gives different result than Icarus Verilog):
```
VL_ATTR_COLD void Vtest___024root___eval_initial__TOP(Vtest___024root* vlSelf) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtest___024root___eval_initial__TOP\n"); );
    Vtest__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Body
    VL_WRITEF_NX("a == 0\n",0);
}
```

If I move "a <= 0" after (below) the if's:
```
module test;
	initial begin
		a = 0;
	end
	reg a;
	always @(*) begin
		if (a == 0) begin
			$display("a == 0");
		end
		if (a == 1) begin
			$display("a == 1");
		end
		a <= 0;
		a <= 1;
	end
endmodule
```

then verilator generates another code (which gives the same result as Icarus Verilog):
```
VL_INLINE_OPT void Vtest___024root___act_sequent__TOP__0(Vtest___024root* vlSelf) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtest___024root___act_sequent__TOP__0\n"); );
    Vtest__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    auto& vlSelfRef = std::ref(*vlSelf).get();
    // Body
    if (VL_UNLIKELY(((1U & (~ (IData)(vlSelfRef.test__DOT__a)))))) {
        VL_WRITEF_NX("a == 0\n",0);
    }
    if (VL_UNLIKELY((vlSelfRef.test__DOT__a))) {
        VL_WRITEF_NX("a == 1\n",0);
    }
    vlSelfRef.test__DOT__a = 1U;
}
```

Icarus Verilog gives the same result regardless of the "a <= 0" location, but verilator generates different code with different outcome.
I suppose, Icarus Verilog behaves correctly, but verilator does not.
Will it be difficult to fix this?

gezalore, 2025-03-27:
Your original should have a `COMBDLY` warning, which explicitly tells you that the NBA `<=` inside a combinaional process `alwayw @*` will be executed as if it was a blocking `=` assignment. This is a limitation of the static scheduling scheme used by Verilator. You could rewrite the `always @(*)` listing all sensitivities explicitly which would make it work as you expect but at a cost of potentially severe performance degradation. If you are using an higher level HDL that generated this code, it would be best to feed it back there that they should not be generating combinational always blocks containing NBAs. (Similar to #5819)

wsnyder, 2025-03-27:
As @gezalore described. 

I checked and Verilator does give a `COMDLY` warning.

We aren't likely to fix this in the medium term, and it's a known deficiency, so closing this issue.

closed 2025-03-27 by wsnyder

dev-pts, 2025-03-27:
> This is a limitation of the static scheduling scheme used by Verilator.
> We aren't likely to fix this in the medium term, and it's a known deficiency, so closing this issue.

Ok, I understand.

> You could rewrite the always @(*) listing all sensitivities explicitly which would make it work as you expect but at a cost of potentially severe performance degradation.

Yeah, that works like Icarus Verilog now. Thank you! Will try to figure out what to do with this...