verilator/verilator #5894: How to get source code after parameter instantiation?

pointerliu, 2025-03-28:
I have a SystemVerilog module that instantiates submodules based on a parameter:
```systemverilog
module A #(parameter width = 3) ();  
  if (width >= 2) begin  
    B b();  
  end else begin  
    C c();  
  end  
endmodule  
```
During compilation, the parameter width is determined to be 3, so only submodule B is instantiated.

How can I obtain the simplified source code after parameter expansion, where only the relevant submodule remains?

wsnyder, 2025-03-28:
I assume by source code out you mean Verilog.  Although Verilator has some Verilog output capabilities it makes transformations towards compuling at many points.  Instead this sounds a good job for you to write a script leveraging [slang](https://github.com/MikePopoloski/slang)

closed 2025-03-28 by wsnyder

udif, 2025-03-28:
In fact, slang already has a tool to do it, called slang-hier, written by me exactly for that purpose.
It had several flags for output formatting and output pruning.
https://github.com/MikePopoloski/slang/tree/master/tools/hier

pointerliu, 2025-03-28:
Thanks a lot~ I will check it.

pointerliu, 2025-05-12:
Hi @wsnyder ,

I have checked the tool `slang-hier` that works well for module instantiation.
But now I'm working with parameter-controlled blocks and want to determine which block is active in a given simulation.
For example:

```systemverilog
module t #(
    parameter ENABLE = 1
) (
    input wire a,
    input wire b,
    output reg c
);

    if (ENABLE) begin
        assign c = a;
    end else begin
        assign c = b;
    end
endmodule
```

At a given simulation, only one of the blocks is instantiated based on the parameter `ENABLE`, and the other is discarded. I would like to distinguish which block is actually used in the simulation.

I tried using `--coverage-line`, but Verilator appears to mark all `assign` statements as covered, even though one of them is not present in the elaborated design.

This requirement may not be general for other users, so I want to extend Verilator's line coverage implementation for myself.
I want to add a marker for the lines that are actually instantiated in the simulation. Something like:
```systemverilog
module t #(
    parameter ENABLE = 1
) (
    input wire a,
    input wire b,
    output reg c
);

    if (ENABLE) begin
$      assign c = a;
    end else begin
       assign c = b;
    end
endmodule
```
Could you give some guidance on enhancing the line coverage implementation?

Thanks!

wsnyder, 2025-05-12:
Note you can make parameters public, and they will hit C++.  Or you can make a JSON dump and they should be included - if there's no clean dump point now we could add one just before the (when not public) parameters are deleted.

pointerliu, 2025-05-14:
Thanks a lot, @wsnyder!

I found that there's indeed a stage where parameters are removed. By dumping the AST just after that point, I'm able to extract the information I need.

However, I noticed something interesting regarding conditional statements. For example:

```systemverilog
if (ENABLE)  // AstGenIf
    assign d = a;
else
    assign d = b;

always @(*) begin
    if (ENABLE) // AstIf
        c = a;
    else
        c = b;

    if (a && ENABLE) // AstIf
        c = a;
end
```

The first `if (ENABLE)` is parsed as an `AstGenIf` and is correctly removed by `ParamVisitor::visit(AstGenIf)`. But the second `if` statements inside the `always` block are parsed as `AstIf` nodes, and `ParamVisitor` doesn't seem to touch those, even when the condition is purely parameter-based. The third `if` contains both parameters and variables, so it should be ignored.

Would it make sense for `ParamVisitor` to also consider `AstIf` nodes when their conditions are entirely parameters?

Thanks again for your help and guidance!

wsnyder, 2025-05-14:
Per IEEE, the inner if is not a generate, it's just like doing a "if (_some_constant)" or whatever.  This is required per the standard.