verilator/verilator #5895: The simulation is abnormally ended

Dragon-Git, 2025-03-28:
When there is only the main process. Use the semaphore.get() to get the key, and when the key is insufficient, the simulation ends abnormally. As described in IEEE document 15.3.3
> If the specfied number of keys is not available, the process blocks until the keys become available.

Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contaned so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs
```systemverilog
module sema_test;
    initial begin
        semaphore sem = new(1);
        sem.get(2);
        #100;
        $display("test end");
    end
endmodule

```

What 'verilator' command line do we use to run your example?
```bash
$VERILATOR_ROOT/bin/verilator  --binary -top sema_test sema_test.sv;./obj_dir/Vsema_test
```

What 'verilator --version' are you using?  Did you try it with the git master version?
```
rev v5.034-81-g464e19caa
```
What OS and distribution are you using?

Macos Sequoia 15.3.2

wsnyder, 2025-03-28:
I assume by "abnormally" you mean crashed.

Running this on current master and on g464e19caa , even adding --runtime-debug, exits without printing anything and returns good exit status, like other simulators I tested.  Perhaps this is only exposed on MacOS issue, please try debugging, especially with `--runtime-debug` & be sure to use master as I fixed a queue issue recently that should be unrelated but better safe.

Dragon-Git, 2025-03-29:
Thank you for your analysis. By "abnormally," I do not mean a crash (which would produce an error code or stack trace), but rather an unexpected finish. According to my understanding of the IEEE description, the simulator should wait for keys unless the simulation is manually terminated, but my tests in the VCS will also end the simulation instead of waiting for the key. Looks like I misunderstood IEEE.

closed 2025-03-29 by Dragon-Git