verilator/verilator #5902: Invalid `fork-join_none` handling behaviour

serge0699, 2025-03-30:
> Can you attach an example that shows the issue?  (Must be openly licensed, completely self-contaned so can directly run what you provide. Ideally use test_regress format, see https://veripool.org/guide/latest/contributing.html?highlight=test_regress#reporting-bugs

Example:

```verilog
module fork_join_none;

    initial begin
        $display("Statement 1");
        fork
            $display("Process 1");
        join_none
        $display("Statement 2");
        $display("Statement 3");
        $display("Statement 4");
    end

endmodule
```

Expected ouptut:

```
Statement 1
Statement 2
Statement 3
Statement 4
Process 1
```

According to 1800-2023 LRM, section 9.3.2: _In all cases, processes spawned by a fork-join block shall not start executing until the parent process is blocked or terminates._

Real output:

```
Statement 1
Process 1
Statement 2
Statement 3
Statement 4
```

> What 'verilator' command line do we use to run your example?

```bash
verilator fork_join_none.sv --binary
./obj_dir/Vfork_join_none
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.035 devel rev v5.034-69-g780163b91`

> What OS and distribution are you using?

CentOS Linux release 7.9.2009.

wsnyder, 2025-03-30:
Confirmed on others.

Might you attempt a fix (presumably reorder statements in V3Fork)?  Otherwise @kboronski-ant?

wsnyder, 2025-03-30:
Self checking test.

```
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);

module t;

   bit stmt2 = '0;
   bit proc1 = '0;

   initial begin
      $display("Statement 1");
      fork begin
         $display("Process 1");
         proc1 = '1;
         `checkh(stmt2, 1'b1);
      end
      join_none
      $display("Statement 2");
      stmt2 = '1;
      `checkh(proc1, 1'b0);

      #1;
      `checkh(proc1, 1'b1);

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule

```

wsnyder, 2025-03-31:
CC @kbieganski

kbieganski, 2025-04-01:
Huh, I think this is a new rule in IEEE 1800-2023.

Something like this would work:

```diff
diff --git i/src/V3Begin.cpp w/src/V3Begin.cpp
index 9eed66f71..4d7e73896 100644
--- i/src/V3Begin.cpp
+++ w/src/V3Begin.cpp
@@ -132,6 +132,13 @@ class BeginVisitor final : public VNVisitor {
                 beginp->addStmtsp(stmtp);
                 stmtp = beginp;
             }
+            if (nodep->joinType().joinNone()) {
+                AstBegin* const beginp = VN_AS(stmtp, Begin);
+                if (AstNode* const stmtp = beginp->stmtsp()) {
+                    stmtp->addHereThisAsNext(new AstDelay{
+                        beginp->fileline(), new AstConst{beginp->fileline(), 0}, false});
+                }
+            }
         }
         dotNames(nodep, "__FORK__");
         nodep->name("");
diff --git i/src/V3Timing.cpp w/src/V3Timing.cpp
index c586365af..7cc6dd2d0 100644
--- i/src/V3Timing.cpp
+++ w/src/V3Timing.cpp
@@ -1221,18 +1221,10 @@ class TimingControlVisitor final : public VNVisitor {
         AstNode* stmtp = nodep->stmtsp();
         // Put each statement in a begin
         while (stmtp) {
-            if (!VN_IS(stmtp, Begin)) {
-                auto* const beginp = new AstBegin{stmtp->fileline(), "", nullptr};
-                if (hasFlags(stmtp, T_HAS_PROC)) addFlags(beginp, T_HAS_PROC);
-                stmtp->replaceWith(beginp);
-                beginp->addStmtsp(stmtp);
-                stmtp = beginp;
-            }
             auto* const beginp = VN_AS(stmtp, Begin);
+            if (hasFlags(stmtp, T_HAS_PROC)) addFlags(beginp, T_HAS_PROC);
             stmtp = beginp->nextp();
             iterate(beginp);
-            // Even if we do not find any awaits, we cannot simply inline the process here, as new
-            // awaits could be added later.
             // Name the begin (later the name will be used for a new function)
             beginp->name(nodep->name() + "__" + cvtToStr(idx++));
         }
```

I haven't tested it fully, but it works on @wsnyder's test. If anyone is interested in developing this feature, feel free to grab this patch.

(I deleted the code that wraps statements in begins in `V3Timing` as it seems unnecessary, but again, untested)

wsnyder, 2025-04-01:
I tried the second half, V3Timing alone, and that broke a lot, I assume it isn't necessary part of the tix.

The first one breaks many tests with

>%Warning: t/t_timing_fork_join.v:55: Encountered #0 delay. #0 scheduling support is incomplete and the process will be resumed before combinational logic evaluation.

So more work needed.

kbieganski, 2025-04-02:
> I assume it isn't necessary part of the tix.

That's right, it can be dropped.

As for the `#0` warning, we could have some sort of a special flag for 0-delays that suppresses this warning. Or make a separate queue for forked processes.

An alternative approach would be to move the `fork` to the parent's next suspension point (or the end). More efficient at runtime, but it'd require control flow analysis, so it's not really worth it.

serge0699, 2025-04-02:
> Might you attempt a fix (presumably reorder statements in V3Fork)?

@wsnyder, I can try, but I have no experience with editing Verilator sources, so it can take a long time.

wsnyder, 2025-04-03:
Closer but still causes bunch of test unsupported fallout, beyond the `#0` warnings.  I'm leaving this for a while, if someone wants to pick it up.

```
diff --git a/src/V3Fork.cpp b/src/V3Fork.cpp
index 60b14a268..445f17f68 100644
--- a/src/V3Fork.cpp
+++ b/src/V3Fork.cpp
@@ -379,6 +379,32 @@ class DynScopeVisitor final : public VNVisitor {
         VL_RESTORER(m_forkDepth);
         if (!nodep->joinType().join()) ++m_forkDepth;

+        // If a statement is not a begin, wrap it in a begin. This fixes an issue when the
+        // statement is a task call that gets inlined later (or any other statement that gets
+        // replaced with multiple statements)
+        if (debug()) nodep->dumpTree("-in-FIXME ");
+        for (AstNode* stmtp = nodep->stmtsp(); stmtp; stmtp = stmtp->nextp()) {
+            if (!VN_IS(stmtp, Begin)) {
+                AstBegin* const beginp = new AstBegin{stmtp->fileline(), "", nullptr};
+                stmtp->replaceWith(beginp);
+                beginp->addStmtsp(stmtp);
+                stmtp = beginp;
+            }
+            if (nodep->joinType().joinNone()) {
+                // Add '#0;' to front of each fork...join_none block
+                // as IEEE says the new process must begin at the next blocking statement
+                AstBegin* const beginp = VN_AS(stmtp, Begin);
+                if (!VN_IS(beginp->stmtsp(), Delay)) {  // Don't need to delay then delay
+                    AstNode* const laterStmtsp = beginp->stmtsp();
+                    if (laterStmtsp) laterStmtsp->unlinkFrBackWithNext();
+                    beginp->addStmtsp(new AstDelay{stmtp->fileline(),
+                                                   new AstConst{stmtp->fileline(), 0}, false});
+                    if (laterStmtsp) beginp->addStmtsp(laterStmtsp);
+                }
+            }
+        }
+        if (debug()) nodep->dumpTree("-ou-FIXME ");
+
         const bool oldAfterTimingControl = m_afterTimingControl;

         ForkDynScopeFrame* framep = nullptr;
```