verilator/verilator #5903: Invalid streaming operator packing ordering

serge0699, 2025-03-30:
> Can you attach an example that shows the issue?

Example:

```verilog
module data_packing;

    byte    data [];
    longint word;

    initial begin

        data = new[8]('{8'hfa, 8'hde, 8'hca, 8'hfe,
                        8'hde, 8'had, 8'hbe, 8'hef});

        word = {>>{data}};
        $display("word: %h", word);
        word = {<<8{data}};
        $display("word: %h", word);

    end

endmodule
```

Expected outptut:

```
word: fadecafedeadbeef
word: efbeaddefecadefa
```

According to 1800-2023 LRM, section 11.4.14.2: _>> causes blocks of data to be streamed in left-to-right order, while << causes blocks of data to be streamed in right-to-left order._

Real output:

```
word: efbeaddefecadefa
word: fadecafedeadbeef
```

> What 'verilator' command line do we use to run your example?

```bash
verilator data_packing.sv --binary
./obj_dir/Vdata_packing
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.035 devel rev v5.034-69-g780163b91`

> What OS and distribution are you using?

CentOS Linux release 7.9.2009.

wsnyder, 2025-03-30:
Agreed.  Might you be willing to make a pull to fix this?

Self-checking test:

```
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while($

module t;

   byte data [];
   longint word;

   initial begin
      data = new[8]('{8'hfa, 8'hde, 8'hca, 8'hfe,
                      8'hde, 8'had, 8'hbe, 8'hef});
     `checkh(data[0], 8'hfa);
     `checkh(data[7], 8'hef);

      word = {>>{data}};
     `checkh(word, 64'hfadecafedeadbeef);

      word = {<<8{data}};
     `checkh(word, 64'hefbeaddefecadefa);

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

serge0699, 2025-04-02:
> Might you be willing to make a pull to fix this?

@wsnyder, I can try, but I have no experience with editing Verilator sources, so it can take a long time.

paul-demo, 2025-05-29:
I looked at this to try to get familiar with Verilator internals.

Is the issue that `AstCvtArrayToPacked` on lines 2212 for StreamR and 2300 for StreamL are not interpreting the right hand side with the correct order?

https://github.com/verilator/verilator/blob/5fcd0e52e72e9f5e28d27fbfd763b6a12a65aca9/src/V3Const.cpp#L2211-L2212

https://github.com/verilator/verilator/blob/5fcd0e52e72e9f5e28d27fbfd763b6a12a65aca9/src/V3Const.cpp#L2297-L2302

I don't see where `AstCvtArrayToPacked` code is -- possibly something generated by astgen or else it's so compact that it's only 2 lines here and I just don't understand what they do:

https://github.com/verilator/verilator/blob/5fcd0e52e72e9f5e28d27fbfd763b6a12a65aca9/src/V3AstNodeExpr.h#L1137-L1138

GuangMing34, 2025-06-09:
@serge0699 
I'm also confused with bit order with stream operation.
I found some descriptions in spec.

> `IEEE` Std 1800-2023. 11.4.14.2 Re-ordering of the generic stream
> 
>  Left-to-right streaming using >> shall cause the slice_size to be ignored and no re-ordering performed.
> Right-to-left streaming using << shall reverse the order of blocks in the stream,

and also examples
> { << { 8'b0011_0101 }} // generates stream 'b1010_1100 (bit reverse)

It seems Right-to-left stream operator >> won't change the bit order.

I tested on synopsys VCS & mentor modelsim, these two tools get same result:
for dynamic array(eg: bit var_tmp[])  : use Right-to-left stream operator >> will get a reversed result.
for packed array(eg: bit[3:0] var_tmp): use Left-to-right stream operator << will get a reversed result.

Could you tell me why the stream operation on vcs & modelsim is right?

wsnyder, 2025-06-09:
I believe for arrays, left is before the colon of the range and right is after so `[3:4]' left-to-right means [3] then [4], while '[4:3]' is reverse.  For queues and associative arrays 0:max.  But best to make some examples as I don't think IEEE clearly states this beyond left/right in the VPI section.

GuangMing34, 2025-06-10:
in spec 1800-2023, it mentioned about unpacked array with stream operator.
I think it is clear enough, maybe

>  11.4.14.1 Concatenation of stream_expressions
> 
>  else if the expression is an unpacked array (i.e., a queue, dynamic array, associative array, or fixed-size unpacked array) 
> this procedure shall be applied in turn to each element of the array. An associative array is
>  processed in index-sorted order. Other unpacked arrays are processed in the order in which
>  they would be **traversed by a foreach loop** (see 12.7.3) having exactly one index variable; 
>
>
> 12.7.3 The foreach-loop
>
>
> int A [2][3][4];  bit [3:0][2:1] B [5:1][4];
> foreach( A [ i, j, k ] ) ... 
> foreach( B [ q, r, , s ] ) ...
> The first foreach-loop causes i to iterate from 0 to 1, j from 0 to 2, and k from 0 to 3.
> The second foreach-loop causes q to iterate from 5 to 1, r from 0 to 3, and s from 2 to 1 (iteration over the third index is skipped).

paul-demo, 2025-06-10:
A quick look at git blame suggests it's some minor oversight in this commit by @kozdra 

https://github.com/verilator/verilator/commit/a32b8d80f9ae225563e2fca2632b39283c953087

paul-demo, 2025-06-10:
Here's what I found:

```systemverilog
      i = new[8]('{8'hfa, 8'hde, 8'hca, 8'hfe,
                   8'hde, 8'had, 8'hbe, 8'hef});
      j = {>>{i}};
```

produces:
```c++
    vlSelfRef.t__DOT__unnamedblk1__DOT__i.renew_copy(8U, 
                                                     VlQueue<CData/*7:0*/>::consVC(0xefU,    // element [7] is here
                                                                                VlQueue<CData/*7:0*/>::consVC(0xbeU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xadU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xdeU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xfeU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xcaU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xdeU, 
                                                                                VlQueue<CData/*7:0*/>::consVC(0xfaU,   // element [0] is here
                                                                                VlQueue<CData/*7:0*/>{})))))))));
t__DOT__unnamedblk1__DOT__j = VL_PACK_Q_RI(64, 8, vlSelfRef.t__DOT__unnamedblk1__DOT__i);
```

So one of the two is probably happening:
1) This should not be a VlQueue type, OR
2) The code here in `verilated_funcs.h` needs to use inverse slicing rather than foreach slicing:
https://github.com/verilator/verilator/blob/d9534ec6266b6dbc43cbad17c8bb4fdc2638c8da/include/verilated_funcs.h#L1608-L1624

It's not clear to me if my PR is going to mess up other things but as far as I can tell, VL_PACK_* is only used in this one specific place in `V3EmitCFunc.h` when visiting `AstCvtArrayToPacked` nodes.

https://github.com/verilator/verilator/blob/master/src/V3EmitCFunc.h#L403-L412

I could be wrong about that but hopefully the scope of impact here is pretty small.

closed 2025-06-10 by wsnyder